MarlinFirmware/Marlin #1775: http://marlinfirmware.org/     is down

boelle, 2015-04-01:


thinkyhead, 2015-04-02:
__Sorry! This site is experiencing technical difficulties.__
__Try waiting a few minutes and reloading.__
__(Cannot contact the database server)__

boelle, 2015-04-02:
yep, and waiting an hour or day gives same message

thinkyhead, 2015-04-02:
Ah well, at least http://marlinfirmware.org/configurator/ still works. I'm sure it's something simple.

@scotty1024 ?

boelle, 2015-04-04:
@scotty1024... still among us?

boelle, 2015-04-05:
its the database failing.... but i grow a bit worried that @scotty1024 is no longer among the living...

how long should we give it before seaking alternative domain names etc etc?

thinkyhead, 2015-04-07:
@boelle For the wiki, it might be best to just add a set of sub-pages directly to the RepRap wiki, starting at the top page http://reprap.org/wiki/Marlin – and I can install a redirect from marlinfirmware.org to that wiki page.

boelle, 2015-04-07:
so you have access to the webspace where the domain is?

if that is the case we can maybe get it back on track

thinkyhead, 2015-04-07:
@boelle I don't have the database credentials, which seems to be the missing piece. But I saw a live @scotty1024 lurking around Marlin earlier today.

boelle, 2015-04-07:
:-D well i'm off to work.... i get home at about 14:30 my time

and day off tomorrow.... will prob hang out at the hangout link i put in the redme.md

scotty1024, 2015-04-07:
It wanted it’s latest security updates and a reboot.

My bad, I was distracted with tuning my first dual extrusion printer.

boelle, 2015-04-07:
hehe, thing happens :-D

might be wise to share access with one more person should you not be able to attend server

maybe @thinkyhead ....

thinkyhead, 2015-04-08:
If need be, I have a Hostgator hosting account with a dozen domains on it, which I pay for annually in one big lump. It doesn't cost extra to add new domains. They just charge me for the total CPU and bandwidth. :smiling_imp:  And I already have one MediaWiki instance hosted there, which has been hack-proofed against being edited by sock-puppets.

Drupal 7 with AJAX makes a good platform, as it likes markdown, can have forums added, uses tags, can display code with syntax highlighting, and can double as a database – for example, each GCode could be an individual object (node), which gives it its own "page" (ajax-usable view), makes it cacheable, more usable in views, lists, blocks, searches, etc. Then, each "Machine" (as in example_configurations) can also have a node, with all the same capabilities, the current configuration for each linked to the Marlin repository - also fetchable and cacheable. Heavy use of AJAX makes such sites much better - more like apps (or like Github).

emartinez167, 2015-04-08:
> domains. They just charge me for the total CPU and bandwidth. [image:

I like that idea!!!

scotty1024, 2015-04-08:
> If need be, I have a Hostgator hosting account with a dozen domains on it, which I pay for annually in one big lump. It doesn't cost extra to add new domains. They just charge me for the total CPU and bandwidth.  And I already have one MediaWiki instance hosted there, which has been hack-proofed against being edited by sock-puppets.

Digital Ocean is flat rate and has many different "droplets" that can be spun up in about 55 seconds. Each droplet can be scaled from $5 to $640 per month of resources.

I had seen and dealt with a couple edits but was unaware you were seeing stuff elsewhere. I'm happy to lock it down and roll it back. I will put the polycarbonate down tonight and go flush out the wiki.

We don't need you distracted from working on Marlin.

PIDs in bad shape. Those positive excursions cook Taulman Bridge and jam a RRD Hexagon. :-(

Scotty

scotty1024, 2015-04-09:
I’ve installed Nuke and I’m cleaning it all out which is also deleting the users accounts.

Fortunately I didn’t pop for the $5 droplet so this is going very quickly.

Probably quicker to have just dusted off and nuked it from orbit but I’m having fun mowing the SPAM pages.

scotty1024, 2015-04-09:
Locked down, all accounts deleted but one, main page erased, g-code page saved and no one has changed the main page in an hour. 

I think it’s cleaned out and locked out.

thinkyhead, 2015-04-09:
Cool. If my account there had Edit access, I could restore the front page, etc.

scotty1024, 2015-04-09:
Yes sir, just waiting to let them do their worst to get back in. I never realized there was so much money to be made in warning the world about the dangers of femdom's. :-)

I'm also granting your ssh account access to "sudo service apache2 restart" and "sudo shutdown -r now".

I've run the maintenance scripts to clean up the database after our recent departure of 100,000+ unwelcome users and pages.

Anyone else wants a wiki account please speak up. I'm adding them via command line. The front door now has an account request queue, which I figured would also be abused. So I have it set to require a 100,000 word biography to request an account.

scotty1024, 2015-04-10:
@Thinkyhead I’ve loosened then re-tightened back a quarter turn. Your old account should be active and capable of edits. The 'user' group has some vulnerability in it so if you want to adjust please don’t tweak anything related to ‘user’.

thinkyhead, 2015-04-10:
My "Thinkyhead" account there still can't edit. :cry:

scotty1024, 2015-04-10:
I thought I gave you enough soup, I’ve up’d you some more.

thinkyhead, 2015-04-10:
Power is a beautiful thing…

thinkyhead, 2015-05-13:
Down again!

thinkyhead, 2015-06-06:
Down today too…

boelle, 2015-06-29:
was  up today so closing again

closed 2015-06-29 by boelle

indazoo, 2015-11-15:
Hi today it is down too:
Sorry! This site is experiencing technical difficulties.

Try waiting a few minutes and reloading.

(Cannot contact the database server)