MarlinFirmware/Marlin #1777: Prefered tool for dev chat...???

boelle, 2015-04-01:
We have IRC.... but every time i have been there it just felt like a giant parking lot.

IS IRC the preferred choice among those who active do bug fixing? or are things like Google hangouts more of the choice even thou it requires a G account?

epatel, 2015-04-01:
Slack?

Natealus, 2015-04-01:
Is Netmeeting still around? That was quite useful back in the day but there's many many options like that now. Hangouts seems like a well featured collaboration tool too.

alexborro, 2015-04-01:
I use hangout and skype. They are nice for chatting.

boelle, 2015-04-02:
Me using google anyways so i think hangout are going to be the norm...

For those who hate google... We cant let that hold back development work, you can just use google for this and nothing else. It nothing worse than using a login for IRC. That is if you only use google for hangouts

Natealus, 2015-04-02:
And 15 gigs of free web storage too. Useful for sharing files among colleagues and friends.

boelle, 2015-04-02:
@thinkyhead what is your thoughts?

thinkyhead, 2015-04-02:
My preference would be for Google+ because I get notifications instantly. I don't leave IRC open all the time, or I might prefer that. Being in many time-zones does make real-time collaboration trickier.

boelle, 2015-04-02:
could you create a hangout then for marlin ?

thinkyhead, 2015-04-04:
Well, it's like other chat clients. I would need the gmail addresses of all people who would be invited to the hangout, and we would need to decide on a pre-arranged time. If we were to do a video hangout, then it could be recorded for posterity, and publication, to connect with the wider community.

I have a list of time-zones which we can add to. This one shows the times relative to me, and which times in each of their time-zones are convenient for me, at least for one-on-one chat about topics.

    @thinkyhead      GMT-7                        9am -  1am         9am -  1am
    @clefranc        GMT-4   +  3 hours           6am - 10pm         9am - 10pm
    @fmalpartida     GMT-1   +  6 hours           3am -  7pm         9am -  7pm
    @AnHardt         GMT+1   +  8 hours           1am -  5pm         9am -  5pm
    @emartinez167    GMT+8   + 15 hours = -9      6pm - 10am         6pm -  1am, 9-10am

emartinez167, 2015-04-04:
ernesto.e.martinez@gmail.com 

Regards,

Ernesto

alexborro, 2015-04-04:
alexborro@gmail.com GMT-3

boelle, 2015-04-04:
i cleaned up the thread and added myself... i belive that GMT -1 is where i'm

then again there is summer time.... right now i'm 2 hours ahead of GMT

boelle, 2015-04-04:
added hagnout link to the readme.. i might be there but cam and mic is off for now.... (want to pick my nose in piece)

ntoff, 2015-04-05:
I'd love a forum or something, just to post random ideas instead of clogging up the issues thing in here with my dribble.

thinkyhead, 2015-04-05:
…or your drivel. Your meaningless drivel…

ntoff, 2015-04-05:
all of the above, i'll see myself out

thinkyhead, 2015-04-05:
@ntoff Maybe, if the marlin wiki comes back we could use that. But here's an idea, why not start a topic at the RepRap forums? Just a general "Marlin Firmware" topic with a list of informal agenda items…

ntoff, 2015-04-05:
GMT +10, I'm not a pro-coder though, I have no idea what I'm doing with C++. Took me ages to figure out how to add features to pronterface. Stupid python and it's lack of curly braces.

thinkyhead, 2015-04-06:
@ntoff An interest in RepRap and open source is all that's required to have a worthwhile opinion. And often the best ideas come from a more naive perspective, not so conditioned by the way things are "supposed" to work.

boelle, 2015-04-08:
@thinkyhead i looked at the table... in risk of screwing up i wondered if you could add the timespan where everyone is able to be online? but in each users local time?

Wurstnase, 2015-04-08:
Lovely Python!
GMT +1 btw :)

thinkyhead, 2015-04-09:
@boelle Users will have to contribute that information, when they will be online - I suppose the best unit would be UTC.

boelle, 2015-04-09:
ok

but for me from about 15:00-22:00

weddensday all day

weekends the same

utc +2

Den torsdag den 9. april 2015 skrev Scott Lahteine <notifications@github.com

boelle, 2015-04-09:
added 2 columns...

fmalpartida, 2015-04-11:
Hi folks, 

count me in for a hangout, G+ account fmalpartida(a)gmail.com

boelle, 2015-04-11:
link for hangout in the readme.md (frontpage on github), but here it is

https://plus.google.com/hangouts/_/g2wp5duzb2y6ahikg6tmwao3kua

closed 2015-04-15 by boelle