MarlinFirmware/Marlin #1778: Thermal runaway misbehaving

fotis400, 2015-04-01:
Thermal runaway is not working correctly. It triggers false alarms, especially when I change extruder temperature while printing.

Let's say I send `M104 S180` and wait till temperature is stable. Then I send again `M104 S180` , which triggers the thermal runaway protection almost immediately.

The `THERMAL_RUNAWAY_PROTECTION_PERIOD` timer should reset each time a new temp-related command is send. The way it is working now, the timer starts on the first command, leaving no time for the others to work correctly.

alexborro, 2015-04-01:
It should have one time per heater.

thinkyhead, 2015-04-02:
I've also seen some thermal runaway messages, so I would like to see this fixed ASAP. I'm looking at the code closely and documenting now, and I will post a PR shortly with my tweaks. I don't want to change the code very much without careful consideration, since it has been the same and apparently worked fine for a long time.

So, what is the expected behavior of thermal protection? Looking at the code, it seems kind of wrong, but not exactly in the way described here. I've documented the function and this is the weird logic:

1. If the target temperature is 0 or thermal runaway is triggered, `state=Inactive`, `timer=0`, `return`.
2. Else, if state=Inactive and a target temperature has been set, change state to `FirstHeating`.*
3. If `state==FirstHeating`, wait for the target temperature to be reached, then `state=Stable`.
4. If `state==Stable` and the temperature is _over_ `target - hysteresis` reset the timer.**
5. Else, if the timer has not been reset in 30 seconds (for example) trigger a thermal shutdown.

* Item 2 will always succeed because it can only execute when a temperature is set.
** This means the timer will be reset for any temperature over the target, but why?

So, I thought that the point of thermal protection was to check for a temperature going way over the target. But if that's the case, then the timer should _not be reset_ for temperatures that are too low _or_ too high. But it only seems to care about low temperatures.

As a final note, once thermal runaway has been triggered, the thermal runaway routine goes into an infinite loop, continuously turning off the heaters and updating the LCD. It could shut off the heaters just before the loop, then go into its infinite loop, just updating the LCD. But maybe there's some good reason why it does this. The question is, does the board then need to be reset, or is there some other way this loop is escaped?

nophead, 2015-04-02:
See #1509

thinkyhead, 2015-04-02:
#1787

thinkyhead, 2015-04-02:
@nophead #1509 helps to clarify some things. I recall reading it before, it just left my brain. I think in my own personal case, I must not have set the protection period long enough. I'm not sure what's going on with @fotis400's situation because as far as I can tell if you send `M104` all it does is update the target temperature value. If the target is already at 180 then `M104 S180` _should_ have no effect at all.

thinkyhead, 2015-04-02:
On another note, if what you really want to detect is a detached thermistor, then using a timer doesn't make the most sense. You should look for the temperature changing opposite to the target temperature for some number of degrees. For example, if the target is 180, and the temperature is 150, going down instead of up, then after 10 degrees (for example) you can assume something is wrong. Similarly, if the target is 180 and the temperature is rising, and reaches 190, then you know something is wrong. To keep up with what thermal protection is doing now, when heating first starts (target != 0), then it should check that the temperature is moving in the right direction, and if it doesn't start to do so within some time period, _then_ you know something is wrong. In all cases it should quickly catch a detached thermistor.

nophead, 2015-04-02:
The most likely outcome of not attaching the thermistor properly is it will
partially detach while the machine is printing. That is not trivial to
detect when you are using PID and PWM as there is no on or off state and
the temperature can move in the wrong direction for some time due to
overshoot, undershoot, etc. It would be hard to distinguish between a
sudden increase in power generated by turning a fan or, or extruding much
faster and a partially detached thermistor. You would need a thermal model
that included fan and flow rate inputs and computed a rough power level and
just use P and I for fine adjustment. That you could say if the error term
was bigger than some limit something must be wrong.

As soon as this protection was added we started to see false positives. It
needs quite a lot of complexity to get this right. I hate complexity and
prefer to cement in my thermistors so they never come out. The leads would
break first and that is easy to detect. To get them out I need to use a
drill.

fotis400, 2015-04-02:
@nophead We do not need any more complexity. This concept for me is OK, and will protect some cases. Do not forget that a partially detached thermistor, is still sensing the temperature, and there is still control. The temperature and thus the print quality may not be the desired, but the printer will not burn in flames.

@thinkyhead You haven't understood my problem, I think. The false trigger is when I send a command for __increasing the temperature__ while printing. Follow the scenario:

1. Printer is cold.
2. `M104 S150`. Starts heating.
3. Target temperature reached, and printing has began.
4. `M104 S200`. I decided that I need higher temperature, so I sent the `M104` command.
5. __Immediate thermal runaway error!__

So what I thing will fix that. On your above writing, on step 2:
Else, if `state==Inactive`  and a target temperature has been set __OR target temperature has changed__, set `state=FirstHeating`, `timer=0`.

nophead, 2015-04-02:
A  partially detached thermistor is actually the most dangerous because it looks like it is controlling correctly. The only detectable discrepancy is the power needed to maintain the target is much higher than normal. The temperature could be way higher than normal but read correctly. I had this happen when I used to glue my thermistors on with JB-Weld. It turns to dust after a while. Fortunately my heater maxed out around 300C, see http://hydraraptor.blogspot.co.uk/2008/06/catalogue-of-disasters.html.

Some time later I bought a hot end that used kapton tape to secure the thermistor and another that used silicone. They both failed in a similar way. Then I discovered fire putty and never had a thermistor come out since.

fotis400, 2015-04-02:
Well... I am very lucky... For the last few years that I am on 3D printing, I never had such an issue.

But I think the solution is simple. Redundant thermistors...

nophead, 2015-04-02:
Or fire putty. As I said before, it can't come out unless you drill it out. And the heater needs to not able to detach as well.

But you do need redundant processors both looking at the temperature with each one being able to turn the power supply off.

fotis400, 2015-04-02:
Sorry but fire putty is not the ultimate solution.... I had once a
"cracked" thermistor.... It was in place, reading nonsense....

fotis400, 2015-04-02:
Maybe it is time to stop talking about safety features, and start developing a second "co-board" with redundancy for all our safety needs. Second thermistor, smoke sensor, PSU control, amp sensing etc etc...

nophead, 2015-04-02:
Yes I intend to do all that with a Raspberry PI2 to do all the user
interface, plane equations, motion planning, PID tuning, etc and use a
small fast arm to just do the real time bits, which are bresenham stepper
loop and heater control loop. A tiny piece of simple firmware and all the
rest running on the RPI in a high level language using OO to handle
variation, rather than #if. They can then monitor each other and either be
able turn the PSU off if the temperature control is failing.

I can't see why a thermistor would crack if it is rated for the temperature
and mounted in a thermally compatible material but I think the answer to
detecting all faults is with a model based controller rather than treating
the heater as black box and using PID.

JustAnother1, 2015-04-02:
@nophead  I'm currently working on something very similar as you described. The OO Part on RPI2 is here (https://github.com/JustAnother1/Pacemaker) the client part for AVR(https://github.com/minnow-pmc/Minnow) and ARM (https://github.com/JustAnother1/pmc) 

Another Idea would be to monitor the PID. In the "max out" situation the PID will go into max heating for a long time. So max heating for long time + no change in temperature would be a combination that should trigger the alarm.

nophead, 2015-04-02:
The problem is you don't need anything like max PWM to start a fire with a 40W cartridge heater as somebody asserted recently the auto ignition temperature of PLA is just over 300C. Since you apply full power to warm up you really need to quite accurate in deciding it is warming too slowly, hence why I think it needs an accurate process model as part of the control scheme, which includes the fan, etc. In some situations fan on might need enough extra power to take it over 300C when the fan isn't on.

JustAnother1, 2015-04-02:
You would then also need to know the material used. Other materials might ignite sooner or later. You will also need to calibrate the thermal model. To know how much temperature increase to expect,..

And that would create a complex monster. And complex and safety don't go together. If safety software becomes complex it is better to look for hardware solutions.

The other problem with complex safety solutions is that these are usually only used by those people that will also be very carefully and therefore wont need them anyway. The users that not watch the initial heat up and  do not glue in the temperature sensor will probably also not do the calibration of the Thermal run away feature and whatever else will be needed to make it work. They will just disable it.

thinkyhead, 2015-04-04:
I will see about making sure the state goes back to `FirstHeating` with a temperature increase, and/or resets the timer to give enough time to reach the temperature. From the original post, I got the impression that the state didn't need to change, but clearly it does.

thinkyhead, 2015-04-04:
#1787 should fix this up. It remembers the target temperature that it's working with, and if it changes then it goes back to the Inactive state, causing it to remember the new target temperature and shift immediately into FirstHeating state.

fotis400, 2015-04-04:
Well you got it, but it is still not correct.

The problem now is that firstly the thermal runaway triggers, and after it gets notified about the temperature change. Or at least this is how I get it... Take a look at the printer's responses (presented in the order I receive them in the terminal).

1.Setting the temperature to 150C.
`Setting hotend temperature to 150.000000 degrees Celsius.`

2. Temperature is reached, all is normal.

3. Setting the temperature to 200C.
`Error:Thermal Runaway, system stopped! Heater_ID:
[ERROR] Error:Thermal Runaway, system stopped! Heater_ID: 

0
Setting hotend temperature to 200.000000 degrees Celsius.`

thinkyhead, 2015-04-04:
How about #1813 ? A difference in the storage type could be causing this. I also rearranged the state machine with 2 more states, Reset and (of course) Runaway.

fotis400, 2015-04-04:
Not compiling...

```
temperature.cpp: In function 'void thermal_runaway_protection(TRState*, long unsigned int*, float, float, int, int, int)':
temperature.cpp:1027:31: error: 'heater_index' was not declared in this scope
     if (tr_target_temperature[heater_index] != target_temperature)
                               ^
```

thinkyhead, 2015-04-04:
Whoops, try again…

fotis400, 2015-04-04:
Ok. Just tested it, and it works as expected.

You can merge it. Thank you.

closed 2015-04-04 by fotis400