MarlinFirmware/Marlin #1779: Compiling error on current #1771 build

Natealus, 2015-04-01:
I'm having an odd issue. I tried a couple of things to fix cause it seems to be in a SD card section about SD printing messages, so I reverted the other two merges after the latest bed leveling fixes but it didn't seem to help. Anything here more obvious to others than myself? :)

Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from Marlin.h:16:0,

                 from Marlin_main.cpp:30:

Marlin_main.cpp: In function 'void get_command()':

Marlin_main.cpp:846:35: error: expected ')' before 'MSG_END_HOUR'

         sprintf_P(time, PSTR("%i "MSG_END_HOUR" %i "MSG_END_MINUTE),hours, minutes);

                                   ^

Error compiling.

AnHardt, 2015-04-01:
@Natealus 
Exactly this i tried to fix with #1771.
You have no display defined. Haven't you?
Please try again.

Natealus, 2015-04-01:
Yeah I have my Viki 2 defined. I didn't change that.

AnHardt, 2015-04-01:
@Natealus 
I'll have a look.

Natealus, 2015-04-01:
Given that idea though I tried disabling the LCD system by commenting out VIKI2 and Temp_Stat_LEDS and it did compile sucessfully.

AnHardt, 2015-04-02:
@Natealus 
Greped for VIKI2:
      ..Marlin\Marlin\configurator\config\Configuration.h (1 hit)
	Line 646: //#define VIKI2

that's all! ???

AnHardt, 2015-04-02:
@Natealus 
Sorry have been in an other directory than i supposed.

AnHardt, 2015-04-02:
@Natealus 
Again sorry. Can't reproduce this. For me this compiles as it should. Tried all combinations of VIKI2 and Temp_Stat_LEDS - makes no difference for me.

Natealus, 2015-04-02:
Hmm yeah all I had to do to compile was comment my lcd subsystem out. What board do you have defined? Does it have pins for Viki2?

Natealus, 2015-04-02:
I've also allocated a kill pin which wasn't on the original Viki 2 manual for my board but...I'm not sure that would be related? The only pinsets for the Viki 2 I could put in were ones they had specified. I'm sure it would work on just about any board but I don't know their pinouts and setups well enough to guess at pin sets for other boards.

AnHardt, 2015-04-02:
For compilation i used 'BOARD_AZTEEG_X3_PRO'
But that does not matter as long as the VIKI-Pins are in.
VIKI2 defines ULTRA_LCD in Conditionals.h line 20.
The macro causing the error is defined in ultralcd.h line 11 for ULTRA_LCD - the original place @thinkyhead  put it in.
ultralcd.h is loaded in  time by Marlin_main.cpp.
I can't find the error.

Could you please upload your configuration somewhere?

Natealus, 2015-04-02:
Sure some of the bed leveling stuff is off normal but thats cause I've been testing bug on it.

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(Nate Stokes)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_AZTEEG_X3_PRO
#endif

// Define this to set a custom name for your generic Mendel,
// #define CUSTOM_MENDEL_NAME "Abacus 3D Kraken"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
[... 703 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

boelle, 2015-04-02:
please put long files like that on pastebin.com

Natealus, 2015-04-02:
Hmm the embedding doesn't seem to be working

Natealus, 2015-04-02:
<iframe src="http://pastebin.com/embed_iframe.php?i=VvwTKkfv" style="border:none;width:100%"></iframe>

Natealus, 2015-04-02:
<script src="http://pastebin.com/embed_js.php?i=VvwTKkfv"></script>

thinkyhead, 2015-04-03:
@Natealus I also saw the issue you're referring to - but not there, I got an error because of string concatenation in the language file. And the cause was that one of the strings it was trying to concatenate was undefined, so the compiler saw `"string1"0"string2"`. I'm not sure that's your issue, but check and make sure those defines are really defined.

Natealus, 2015-04-04:
I'm going to do a clean reconfigure off the current merge and see if that solves the LCD issue and leveling and test test test. :)

Natealus, 2015-04-04:
Well now I get a different message. I tried other LCD defines like REPRAP GRAPHICAL and SMART LCD and they give me the same error.

Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

ultralcd.cpp.o: In function `_lcd_move':

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

collect2.exe: error: ld returned 1 exit status

Error compiling.

Natealus, 2015-04-04:
Well this particular issue's error must've been fixed but another occurred but I have a separate issue for that and this one should probably be closed.

closed 2015-04-04 by Natealus