MarlinFirmware/Marlin #1780: Thermal runaway protection during warmup

unlimitedbacon, 2015-04-01:
The current thermal runaway protection is not activated until the hotend reaches operating temperature. This means that there is still the potential for things to go wrong during warmup. Granted, the human should hang around at least until the print starts, but we can't really count on them to be responsible.

My suggestion is to have a timer that shuts off the heater if it is taking too long to warm up (probably during an M109). Another method might be to monitor the rate of change of the temperature and shut off the heater if it is too low.

ntoff, 2015-04-02:
Trouble is, my old printer took 15-20 minutes to heat the bed up and that was quite normal as it had a crap bed (only drew 4A).

However, once heated I don't think I'd want the thermal runaway timer to wait 15 minutes before turning off the printer if the bed went wrong.

thinkyhead, 2015-04-02:
There is a setting called `WATCH_TEMP_PERIOD` that does additional sanity checking on the temperature.

unlimitedbacon, 2015-04-07:
`WATCH_TEMP_PERIOD` looks like pretty much what I was describing.

ntoff: the regular thermal runaway protection would take over after warmup, so you would not have to worry about that.

AnHardt, 2015-04-28:
Is your problem resolved? Can we close this?

closed 2015-05-09 by thinkyhead