MarlinFirmware/Marlin #1783: G28 Z0 homing very very slow

fitch22, 2015-04-02:
Built latest Marlin (as of couple hours ago) for Printrboard on Simple Metal.  When I run G28 with no arguments, homing speeds are as expected.  If I run G29, again speeds are as expected.  If I run G28 Z0, the extruder is raised at a snails pace, then lowered to the stop at expected speed, then raised and lowered to stop again at expected (slower) speed.  Only the first raise is painfully slow.

Is there a setting for this that I don't know about?

Thanks.

Natealus, 2015-04-02:
Check out HOMING_FEEDRATE in configuration.h and in configuration_adv.h

#define X_HOME_RETRACT_MM 7
#define Y_HOME_RETRACT_MM 7
#define Z_HOME_RETRACT_MM 7 // deltas need the same for all three axis
#define HOMING_BUMP_DIVISOR {13, 13, 13}  // Re-Bump Speed Divisor (Divides the Homing Feedrate)

Note I use a Delta but I believe these variables are in all setups. They're just all the same on mine because of the way deltas work. I notice that Z bump divisor is usually twice as much as X and Y on other setup's configs.

fitch22, 2015-04-02:
Not sure if I was clear.  The sequence for homing Z by issuing G28 Z0 is:
painfully slow rise
normal decent until stop hit
normal speed rise
slower (half speed) decent until stop hit again

Everything seems to be at the rate I expect except that initial rise.  This slow rise does not occur if I issue G28 with no arguments, but since it then moves the extruder to the middle of the bed, I cannot issue G29 afterwards.

But I will try changing the HOMING_FEEDRATE to see which of the steps is affected.

Thanks.

Natealus, 2015-04-02:
Hmm the only time that effect happens for me is if I try to move any axis...x, y, z, or an extruder before I home with just a normal G28. I don't have another type of printer besides Delta but from what I hear homing is kind of...opposite on other types I'm not sure I fully understand it yet until I build one. 

When I'm homed and my carriages are all at the top of the towers...I have my manually set home Z of 367.55. Then it travels to Z0 for appropriate nozzle height.

Is your printer supposed to home at Z0 normally?

TechMasterJoe, 2015-04-02:
it is funny because this section is in a else Brackets so in a nut shell it's saying 
if not delta do this
still has another bug hiding 3rd print it started Z down on G29 tell it crashed so more then a few problems to be found so back to me 3-15-15 Dev dump that just works i got 117 parts to print i will do more debugging tomorrow

fitch22, 2015-04-02:
OK.  Thank you TechMasterJoe!  In my snapshot, the offending line was 1953 in Marlin_main.cpp.  I found an occurrence of "feedrate = max_feedrate[Z_AXIS]" earlier in the same function (gcode_G28()), but that had no effect.  Changing line 1953 to "feedrate = homing_feedrate[Z_AXIS]" did the trick.

I finally have working FW that properly compensates for the bed level, and homes within my lilfetime.  

Thanks.

Natealus, 2015-04-02:
Thats awesome. And a 11 foot tall delta? Woot hehe.

c-born, 2015-04-03:
G29 seems to somewhat randomly leave the z-probe down after the last probe.
Using "G29 E" to restore the old behaviour (deploying probe for every test point) gets around it for now.

Wurstnase, 2015-04-03:
max_feedrate[Z_AXIS] * 60 is that what it should be.#1790
feedrate for line_to_destination is feedrate/60. So you take your 4mm/s / 60 which makes it so slow.

AnHardt, 2015-04-03:
@Wurstnase 
Are you sure? This line is unchanged since at least Nov. 2013?

Wurstnase, 2015-04-03:
@AnHardt Sure, sure! feedrate is needed in line_to_destination. And this changed in https://github.com/MarlinFirmware/Marlin/commit/d73b169de3b031779771a14a65cad7fe476d3d98

Take a look here: https://github.com/MarlinFirmware/Marlin/commit/d73b169de3b031779771a14a65cad7fe476d3d98#diff-1cb08de130a6ece2d1b5b9c37bcfef48R1906

alexborro, 2015-04-03:
I have just checked it and @Wurstnase is correct. The newly added function line_to_destination takes the feedrate in mm/min. This mess in speeds is pissing me off.

I propose we get rid of any mm/min functions, converting all to mm/s.. and just convert the gcodes received in mm/min as soon as they are received.

I have just open an issue to discuss that: #1791 

Cheers.

Alex.

AnHardt, 2015-04-03:
Ok. I see.
When this correction is correct in line 1953 https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L1953
why not in line 1820
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L1820?

Wurstnase, 2015-04-03:
already fixed this also ^^ #1790

thinkyhead, 2015-04-04:
Fix has been merged!

closed 2015-04-04 by thinkyhead