MarlinFirmware/Marlin #1785: 15 degrees temp difference on extruder

maukcc, 2015-04-02:
we see a 15 degrees temp difference on extruder when temp is over 200 degrees.
If you set temp to 230 it will go to 215 and hold, if you set it to 270 it will hold on 255.
max. temp is 350 and through display you can go to a max of 335.
we use thermocouples AD597.

nophead, 2015-04-02:
Most likely your PID P and I gains are too low.

maukcc, 2015-04-02:
I might add that this behaviour is not constant. If I heat up from room temp to 250 it usually goes good.
If I heat from room temp to 220 and then set it higher or lower it could result in the 15 degrees lower temp.
P=6
I=0.2
d=42
these values work really well at keeping the temp stable. delta T of about 1 degree.

nophead, 2015-04-02:
I think if you have an error of 15 degrees then the P term will give you a base PWM of 90. If it hangs low the I term should ramp up and max out at 255 * 0.2 = 51. Does the PWM value max out at 141? Does it need more power to get to 250?

There is a PID debugging option you can turn on to see exactly what the numbers are. Even without it you can see the PWM value with M105.

What have you got PID_MAX set to?

maukcc, 2015-04-02:
I can go up to 350 degrees just fine.
The thing is that sometimes the temp gets stuck exactly 15 degrees lower then set. 
For instance: I set the temp to 230, then it goes to 215. if I then set it to 231 it goes to 216. If I set it to 245 it goes to 230. so it is always exactly 15 degrees lower.

nophead, 2015-04-02:
When the error is outside the PID range it uses bang-bang so it will apply full power. Then when it gets close it starts using PID. If the PID gain is too low it will hang low, possibly alternating bang-bang to PID.

To debug this you need to turn on PID debugging and see what the variables are doing.

What temperature did you autotune it at?

There are many values the PID constants can have that will cause it to hang low or else oscillate wildly. It isn't a bug, PID is like that.

maukcc, 2015-04-02:
autotune temp was at 150.
pid functional range is 15 degrees, so that could be worth looking in to.

firmware says extruder max temp is 350, and I can not set the temp higher then 335 through the LCD display.(also 15 degrees)
will that have the same underlying reason?

nophead, 2015-04-02:
Seems to be due to a hard coded - 15: https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L479. That is probably to allow for some overshoot otherwise you would get a MAXTEMP alarm during warmup.

You probably need to autotune at a more representative temperature, but again you need some headroom for overshoot as the PID tuning deliberately invokes oscillation.

maukcc, 2015-04-03:
Thanks Nophead,
I will do some testing.

AnHardt, 2015-04-28:
Is your problem resolved? Can we close this?

maukcc, 2015-05-04:
I have not done some real testing, but I also have had no problems up until now.
So I think it can be closed

closed 2015-05-04 by AnHardt