MarlinFirmware/Marlin #1789: Extruder Temp PID bug?  Can maintain >=186C within +/- 1C  but not antyhing <=185.

DonKahones, 2015-04-02:
Can maintain any temp >=186C within +/- 1C  but when target is set to <=185C temp cycles between 174C and 185C.  Looking at the power to the heater, at 186C the power averages around 50% but as soon as I set it to 185C the power to the heater is cycles between 100% and 0%.  I do not suspect PID settings as it works perfectly for other temps above 185.

Natealus, 2015-04-03:
Often its best to tune PID at the temps you intend to use so that it knows best how to maintain that temperature. That's basically what the auto tuning is doing is going + or - 10C from the temperature variable (I think its S230 for 230C) to measure how many pulses it takes to move the temp up and down. So the settings generated wouldn't be as suited to maintain a temperature 50+ different than you tuned.

Alot of things come into account when trying to maintain a certain temperature. Mostly the characteristics of your hotend at a given temp and general environmental conditions.

DonKahones, 2015-04-03:
That's the thing though.  It never levels out at 185, at 186 it'll level out perfectly.  It just fluctuates, I can see the need to PID tune for my specific range but there is something specific in the algorithm (which I can't locate) that makes it function properly for 186 but not 185.  At 186 it'll use PWM correctly but 185 is either full on or full off, there is no PWM occurring.   I can't see environmental conditions making that much difference for that small of a temp change.

JustAnother1, 2015-04-03:
Did you check the temperature sensor? Could it be that the temperature sensor is misbehaving at <=185°?

Natealus, 2015-04-03:
Well what I meant is usually its auto tuned at 230 or 250C so it's calibrated for that range but yeah it could as well be a sensor reading inconsistency. Its it causing problems cause you're printing at 185?

DonKahones, 2015-04-03:
If no one else is experiencing this issue then it is likely an issue with something in my system. 

 I'll take some measurements tonight on the temp sensor and look for inconsistencies between 185 and 186.

Natealus, 2015-04-03:
Try hooking a multimeter to your temp sensor wire...board-side should work fine I believe. Measure the resistance as the temperature changes. Thermistors work off of that resistance difference. Should be in several Kohms maybe even 100K. See if anything odd jumps around at that temperature.

boelle, 2015-04-04:
please close this one if it turns out to be hardware fault

closed 2015-04-06 by DonKahones