MarlinFirmware/Marlin #1791: Get rid of mm/min speeds.

alexborro, 2015-04-03:
The code is a mess mixing mm/min and mm/s stuffs.

I think mm/min, although a standard in milling machines, makes no sense in 3D Printers due high speeds.

We could handle all the speeds in the code as mm/s and convert the received standard gcodes from mm/min to mm/s before passing them to internal functions. That way we keep compatibility with slicers and other tools and get a clean code.

What do you guys think?

Cheers.

Alex.

Wurstnase, 2015-04-03:
I would go more to teacup and take µm/s.
In config-file mm/s is a good way.

This can makes big advantage because you can handle some numbers in integer than in floats. We can make then some faster math with shifting etc.

ZetaPhoenix, 2015-04-03:
+1

I think the idea of standardizing on unit for speed is appropriate.

whosawhatsis, 2015-04-03:
@Wurstnase with units of µm/s, speeds over 65mm/s would require more than a two-byte integer. The integer math would be significantly easier in mm/s if the precision was adequate.

+1 for standardizing on mm/s

Wurstnase, 2015-04-03:
Long vs. Double? I see no disadvantages. Or which format do you need at 65mm/s?

alexborro, 2015-04-03:
Humm, .. µm/s ... maybe..

There is just one math made with feedrate, in planner:

float inverse_second = feed_rate * inverse_millimeters;

I don't see too much advantage using uint32 than float..(for speeds like
500mm/s - 500.000µm/s - we need a uint32).
Just changing from mm/min to mm/s is matter of removing / adding many "/60"
or "*60" in the code.. changing the type implies critical changes...

Cheers.

Alex.

Wurstnase, 2015-04-03:
Alex you are right. This would be a greater rework with step rates and so on...
+1 for mm/s

alexborro, 2015-04-03:
Let's finish this bugfix phase and then I will implement this change.

Any suggestions are welcome.

Cheers

Alex.

NarimaanV, 2015-04-04:
+1 for mm/s

thinkyhead, 2015-04-04:
This will also end up being a minor documentation issue. But perhaps not too many are dependent on `M203` right now.

Wurstnase, 2015-06-29:
Good thing to push this. We have a lot of /60-division in the code. This will also improve the perfomance.

thinkyhead, 2015-07-03:
Items to be changed:

__Configuration.h__:

    #define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)

__Configuration_adv.h__:

    #define MANUAL_FEEDRATE {50*60, 50*60, 4*60, 60} // Feedrates for manual moves along X, Y, Z, E from panel
    #define AUTO_FILAMENT_CHANGE_FEEDRATE 300   //Extrusion feedrate (mm/min) on automatic extrusion loop

__Marlin_main.cpp__:

    static float feedrate = 1500.0, ...

Other potential changes:

__gcode_M207__:

    if (code_seen('F')) retract_feedrate = code_value() / 60;

__gcode_M208__:

    if (code_seen('F')) retract_recover_feedrate = code_value() / 60;

Wackerbarth, 2015-07-04:
I think that the original purpose in using mm/min was to have numbers which were large enough that they could be measured (and possibly also treated as integers which retained "enough" resolution).

Since everything seems to be going toward floating point, as inefficient as it is, it now becomes a issue of user compatibility.  You're not really saving any space or time if you code the calculations appropriately.

closed 2016-04-23 by jbrazio