MarlinFirmware/Marlin #1792: Secondary home position

d00616, 2015-04-03:
I'm on the way implementing some extensions to use Marlin for printing and milling. For milling i want to set an alternative home position for every axis and enable the capability to move Z axis below zero position.

I have implemented an M211 command which is used to unlock software endstops. http://reprap.org/wiki/G-code#M211:_Disable.2FEnable_software_endstops (PR follows)

Next i wanna implement an command to set an alternative home position. This command is defined as G30, but this is implemented with another functionality in marlin.

At the moment my plan is implement an M command (what M-Code i can choose?) to change the	behavior of G28 command. I wanna switch between move to home location and set current position as home location per axis. So its possible to move the milling tool on top of an part an set what axis i want to zero and mill or drill down.

Any suggenstions about choosing an M command code or better ideas how this can be implemented?

alexborro, 2015-04-03:
Check the G92 command. It does exactly what you need.

Cheers.

Alex.

closed 2015-04-03 by d00616

thinkyhead, 2015-04-04:
@d00606 Note that you can move Z below zero as long as you set your `Z_MIN_POS` less than 0. The `[XYZ]_MIN_POS` settings define the physical limits of movement relative to your HOME position.

d00616, 2015-04-04:
@thinkyhead I know it. I have implemented the M211 comman to make it possible move out of MIN/MAX Position temporarly if needed. http://reprap.org/wiki/G-code#M211:_Disable.2FEnable_software_endstops

thinkyhead, 2015-04-04:
Extending G28 shouldn't be too hard. The variables to alter would be `homeposition` or maybe `base_home_pos`.