MarlinFirmware/Marlin #1793: Marlin runs RUMBA - Problems with the E2 control

Harpye, 2015-04-03:
Trying to use the standard Marlin 1.1 package on a RUMBA board ( Board 80) I recognized the following!
 using X;Y;Z and E0+E1 works fine, but additionally using E2 causes missbehaviour

I built a Gcode skeleton which works fine on the E2 (solo) but after including some movement blocks of the other channels ( which also work well in stand alone usage) in between the existing E2/ T2 blocks, E2 does not work properly anymore . 

Both gcodes work well themselves but merged they cause issues on the E2 behaviour ( the specified number after E __  is ignored and always replaced by a fixed stepcount as also the sign is ignored so 

e.g. E -1 is executed  by 120mm as E1 causes the same result

I put a gcode comparison ( attached below) in here and aligned the corresponding code sequences:

( sorry but I'm brand new to that programming thingy and tired for rather a week now, to understand the Marlin code ! I got stuck...)

Any help would absolutely be appreciated!

Hansjoerg

[image]

thinkyhead, 2015-04-04:
Still looking through the code to see if there's any obvious cause. So far nothing stands out, but I will keep looking. One thing I might need to know, are you using `FWRETRACT`?

Harpye, 2015-04-04:
Hello Thinkyhead! 
I use Marlin to control a model of a 6 axis model of an idustrial handling robot! The misbehaving element is the turntable. The arm behaves well as long as I dont want to use the turntable / T2
the turntable behaves well as I dont want to use the arm! Merging both causes the turntable to missbehave

 I'll pass along my config.h printout as soon as I'm back at my workstation

Thanks in advance for your help!

Hansjoerg

Harpye, 2015-04-05:
As said before - here are the config FIles ... I did not change anything else:

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

// This configurtion file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//User specified version info of this build to display in [Pronterface, etc] terminal window during startup.
//Implementation of an idea by Prof Braino to inform user that any changes made
//to this build by the user have been successfully uploaded into firmware.
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" //Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 250000
//#define BAUDRATE 115200

//// The following define selects which electronics board you have. Please choose the one that matches your setup
// 10 = Gen7 custom (Alfons3 Version) "https://github.com/Alfons3/Generation_7_Electronics"
// 11 = Gen7 v1.1, v1.2 = 11
// 12 = Gen7 v1.3
// 13 = Gen7 v1.4
// 3  = MEGA/RAMPS up to 1.2 = 3
// 33 = RAMPS 1.3 / 1.4 (Power outputs: Extruder, Bed, Fan)
// 34 = RAMPS 1.3 / 1.4 (Power outputs: Extruder0, Extruder1, Bed)
// 4  = Duemilanove w/ ATMega328P pin assignment
// 5  = Gen6
// 51 = Gen6 deluxe
// 6  = Sanguinololu < 1.2
// 62 = Sanguinololu 1.2 and above
// 63 = Melzi
// 64 = STB V1.1
// 7  = Ultimaker
// 71 = Ultimaker (Older electronics. Pre 1.5.4. This is rare)
// 8  = Teensylu
// 80 = Rumba
// 81 = Printrboard (AT90USB1286)
// 82 = Brainwave (AT90USB646)
// 9  = Gen3+
// 70 = Megatronics
// 701= Megatronics v2.0
// 702= Minitronics v1.0
// 90 = Alpha OMCA board
// 91 = Final OMCA board
// 301 = Rambo

#ifndef MOTHERBOARD
#define MOTHERBOARD 80 //80WÄre richtig wenn alles funktioniert
#endif

// This defines the number of extruders
#define EXTRUDERS 3

//// The following define selects which power supply you have. Please choose the one that matches your setup
// 1 = ATX
// 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)

#define POWER_SUPPLY 1

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================
//
[... 378 lines cut ...]

// Support for the BariCUDA Paste Extruder.
//#define BARICUDA

/*********************************************************************\
*
* R/C SERVO support
*
* Sponsored by TrinityLabs, Reworked by codexmas
*
**********************************************************************/

// Number of servos
//
// If you select a configuration below, this will receive a default value and does not need to be set manually
// set it manually if you have more servos than extruders and wish to manually control some
// leaving it undefined or defining as 0 will disable the servo subsystem
// If unsure, leave commented / disabled
//
// #define NUM_SERVOS 3

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H

thinkyhead, 2015-04-06:
I see you are using an older version of Marlin, and we don't fix older versions. You should get the current version of Marlin – the Development branch – and update your configurations for the new version. A lot has changed. I am doing a merge of your `Configuration.h` which I will post shortly. But I am curious about the strange temperature values you have set. What is the meaning of your temperatures all being set to such unusual values?

thinkyhead, 2015-04-07:
An updated `Configuration.h`…

    #ifndef CONFIGURATION_H
    #define CONFIGURATION_H

    #include "boards.h"

    //===========================================================================
    //============================= Getting Started =============================
    //===========================================================================
    /*
    Here are some standard links for getting your machine calibrated:
     * http://reprap.org/wiki/Calibration
     * http://youtu.be/wAL9d7FgInk
     * http://calculator.josefprusa.cz
     * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
     * http://www.thingiverse.com/thing:5573
     * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
     * http://www.thingiverse.com/thing:298812
    */

    // This configuration file contains the basic settings.
    // Advanced settings can be found in Configuration_adv.h
    // BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

    //===========================================================================
    //============================= DELTA Printer ===============================
    //===========================================================================
    // For a Delta printer replace the configuration files with the files in the
    // example_configurations/delta directory.
    //

    //===========================================================================
    //============================= SCARA Printer ===============================
    //===========================================================================
    // For a Scara printer replace the configuration files with the files in the
    // example_configurations/SCARA directory.
    //

    // User-specified version info of this build to display in [Pronterface, etc] terminal window during
    // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
    // build by the user have been successfully uploaded into firmware.
    #define STRING_VERSION "1.0.2"
    #define STRING_URL "reprap.org"
    #define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
    #define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
    #define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
    //#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

    // SERIAL_PORT selects which serial port should be used for communication with the host.
    // This allows the connection of wireless adapters (for instance) to non-default port pins.
    // Serial port 0 is still used by the Arduino bootloader regardless of this setting.
    #define SERIAL_PORT 0

    // This determines the communication speed of the printer
    #define BAUDRATE 250000

    // This enables the serial port associated to the Bluetooth interface
    //#define BTENABLED              // Enable BT interface on AT90USB devices

    // The following define selects which electronics board you have.
    // Please choose the name from boards.h that matches your setup
    #ifndef MOTHERBOARD
      #define MOTHERBOARD BOARD_RUMBA
    #endif

    // Define this to set a custom name for your generic Mendel,
    // #define CUSTOM_MENDEL_NAME "This Mendel"

    // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
[... 668 lines cut ...]
     * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
     * 301 - Rambo  - uses Analog input 3
     * Note may require analog pins to be defined for different motherboards
     **********************************************************************/
    // Uncomment below to enable
    //#define FILAMENT_SENSOR

    #define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
    #define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

    #define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
    #define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
    #define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
    #define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

    //defines used in the code
    #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

    //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
    //#define FILAMENT_LCD_DISPLAY

    #include "Configuration_adv.h"
    #include "thermistortables.h"

    #endif //CONFIGURATION_H

Harpye, 2015-04-07:
Hello Scott!

First  THANK YOU for your kind support on the problem!
I use Marlin - mentioned otherwise- for the 6 Axis control of an model industrial robot  ( perhaps I understand that code one day as far to get the 6 axis interpolated - not just the xyz)
but at the moment I'll try to live with this ! 
I don't need temperature measurement for this job, but clamping off all the measurement causes failure through compilation - so I decided to set the R25° value by using a fix resistor instead of a thermistor and set the window around that value so 
there is no heating period and no wait sequence - the program is just going through all this as everything is fulfilled!

-I'll try your new merge and again THANK YOU VERY MUCH!

BR

Hansjoerg

From: Scott Lahteine 
Sent: Tuesday, April 07, 2015 2:00 AM
To: MarlinFirmware/Marlin 
Cc: Harpye 
Subject: Re: [Marlin] Marlin runs RUMBA - Problems with the E2 control (#1793)

An updated Configuration.h…

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// SERIAL_PORT selects which serial port should be used for communication with the host.
[... 688 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM    0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM            14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY           20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

Harpye, 2015-04-07:
Hello Scott!

After updating the Marlin Version ->1.0.2 
and exchanging the Configuration.h against 
your merge, situation is quite the same - 
the skeleton runs as defined, but with the 
movement blocks it still causes missbehaviour!!
Sorry about the bad news!

 - Hansjoerg

From: Scott Lahteine 
Sent: Tuesday, April 07, 2015 2:00 AM
To: MarlinFirmware/Marlin 
Cc: Harpye 
Subject: Re: [Marlin] Marlin runs RUMBA - Problems with the E2 control (#1793)

An updated Configuration.h…

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0
[... 685 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM    0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM            14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY           20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

clefranc, 2015-04-08:
@Harpye
Don't know if it's related, but the updated config from @thinkyhead have
`#define EXTRUDE_MINTEMP` set to 170
yours was set to 0.

Also, I'll undefine `PREVENT_DANGEROUS_EXTRUDE` as I think you don't need it.

thinkyhead, 2015-04-08:
@Harpye I never expected the problem to go away with the latest code. But, we have to start with the latest code to do anything for your situation. (Not the tagged version 1.0.2, but the latest Development branch code. The configuration I posted is not intended to work with the 1.0.2-tagged Marlin.)

I think the latest code will need to be somewhat re-adapted for your 6-axis setup – I can't imagine you will be able to get it to compile successfully and run without making several adjustments. Most importantly, the `NUM_AXIS` value is no longer set in `Configuration.h` because it's been (historically) just an alias for the number 4. But, you should put it back into yours and remove it from `Marlin.h`. Maybe you did that and that was all it took to compile…?

Harpye, 2015-04-08:
Is 1.0.2 not the  latest frozen revision - I tried to find the Development branch code you mentioned, but was not that successful .
Regarding the "sponsored Configuration .h"  file - I just replaced the code and it immediately was able to be compiled!

Where can I find the recent code - I first found the 1.0.2 version. - do I have to replace EACH single file standing below the //Marlin/Marlin/+ Directory?

From: Scott Lahteine 
Sent: Wednesday, April 08, 2015 5:14 AM
To: MarlinFirmware/Marlin 
Cc: Harpye 
Subject: Re: [Marlin] Marlin runs RUMBA - Problems with the E2 control (#1793)

@Harpye I never expected the problem to go away with the latest code. But, we have to start with the latest code to do anything for your situation. (Not the tagged version 1.0.2, but the latest Development branch code. The configuration I posted is not intended to work with the 1.0.2-tagged Marlin.)

I think the latest code will need to be somewhat re-adapted for your 6-axis setup – I can't imagine you will be able to get it to compile successfully and run without making several adjustments. Most importantly, the NUM_AXIS value is no longer set in Configuration.h because it's been (historically) just an alias for the number 4. But, you should put it back into yours and remove it from Marlin.h. Maybe you did that and that was all it took to compile…?

clefranc, 2015-04-08:
@Harpye 
Go to https://github.com/MarlinFirmware/Marlin
Click the `Download ZIP` button. The button is located at the bottom of the right column.

Open the zip file you just downloaded, you'll see a folder named `Marlin-Development`, this is the latest development branch. Unzip to a location of your choice, delete the zip file, goto the folder `..\Marlin-Development\Marlin` and open `Marlin.ino`.

Do what thinkyhead wrote above.

Harpye, 2015-04-09:
Thank you ! - I did what you said - except the NUM AXIS thingy - which I can't remember that I have it ever seen! - Neither in the Configuration.h nor in Marlin.h
Result of the new Revision is now that it ignores ALL E-Movement ccords and just fulfills the axis movements X Y Z  - is there another way to address them in Gcode now?

thinkyhead, 2015-04-10:
What results do you get if you turn off `PREVENT_DANGEROUS_EXTRUDE`?

Harpye, 2015-04-10:
Now the axis move, but the problem seems to be the same like at the beginning. 
Just only one direction independent from the sign I use and  wrong values .

thinkyhead, 2015-04-10:
Such a mystery. We have to ask, what might possibly be different about E2 in terms of movement? And I assume it would also apply to E3 if there was one. So far nothing in the code is jumping out at me.

Harpye, 2015-04-11:
Another time I tried to understand what Marlin does - but there is so much jumping arround to me - I got no clue until now. Could it be that the movement blocks in between the T2 sequences -someway increment a value which afterwards is used to calculate the E2 steps I specified all axis to be absolute
value adressed - E2 ???  I'd put E2 on an external testing Stepper and as soon as I'll can find the time I'll simplify the Gcode back to skeleton  and a second version  to skeleton + single "desturbing lines" of other single axis and axis combinations - just to see what happens.. I'll come back with the results!

Thank you again for your engagement and help!

Hansjoerg

Harpye, 2015-04-11:
Starting  with the tests I recognized the following:
Simple Gcodes work correctly even if all axis are engaged
More complex Gcodes cause malfunction ->  except I set a G92 just in between the <T2> and the <G1 E2> 
Here I have to use the G1  as an incremetal set since I reset the E2 state every time before I use it even if I set the mode as absolute in header ( <G90>).
I'm not that programming genius  (I'm far... far from :) ) but it appears to me as it could be that the 
variable keeping the position of E2 could be someway affected by the use of other axis . ( use of a local variable instead of a global or sharing of memory ( I'm not the man who should make assumtions of that Level, but if you could not find a reason on the surface and shortly afte a reset the function is correct later on it is not ... something must have changed!)

thinkyhead, 2015-04-12:
There is only a single E position from the GCode point of view. The `e_steps` of individual extruder steppers are tracked, but only with the `ADVANCE` feature, and only within `stepper.cpp`...

But I keep thinking, what if `active_extruder` is changed by a `T` command, but there are still moves in the planner that apply to the previous extruder? Can this happen, or are all moves cleared out whenever a `T` command is issued...?

clefranc, 2015-04-12:
@thinkyhead I think it's the right time (debugging phase) to remove active_extruder in plan_buffer_line, and use exclusively extruder, the argument passed to the function.

thinkyhead, 2015-04-13:
@clefranc Good idea: How does this look? https://github.com/thinkyhead/Marlin/commit/8857b9e921d88e9d8c8bdef3bf9d59e8622b0070

clefranc, 2015-04-13:
@thinkyhead Yes, exactly what I've found.

boelle, 2015-04-15:
should this one not be upgraded to verified? or is the problem gone?

thinkyhead, 2015-04-16:
I made the change suggested by Christian, but don't know if it helped with the E2 problem.

Harpye, 2015-05-01:
Sorry about the delay on the  answere - had some issues with the Notebook - will update the Firmware early next week and will see if it has an effect!

Harpye, 2015-05-01:
oUuups sorry wrong click!

thinkyhead, 2015-05-02:
Still seeing this with the latest code? It's always shifting and changing, so perhaps this got fixed in the process.

Harpye, 2015-05-02:
Yesterday I got the latest developmental Version - and on Monday I will check - with the version of my latest post it was still wrong - but wee will see

Harpye, 2015-05-02:
I did not want to wait, so exchanged the config.h in the current version to that one you sent previously 
(I marked with "1.0.2+ud") and tried the code... unfortunately it seems not to change in behaviour.
Attachd you find the code  - the marked lines (>>>>>>>>>>>>>>>) are executed incorrectly!
Independent from the sign of the G1 entry the motor turns the same direction ( always CW )

;TITLE 'Test Code
; POSTPROCESSOR for Marlin 1.1
; APPLICATION MicroRobot 1.0
G21 ; Set units To millimeters
M17 ; Power all Steppers
G90 ; use absolute coordinates
G4 S3 ;Wait 3s 
T0; Switch Tool
G92 X0 Y0 Z0 E0 ; Zero Axis incl E0
T1; Switch Tool
G92 X0 Y0 Z0 E0 ; Zero Axis incl E1
T2; Switch Tool
G92 X0 Y0 Z0 E0 ; Zero Axis incl E2
T1; Switch Tool
Code Start
G1 X69.7640189871358 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X69.7640189871358 Y163.485302816621 Z235.117990506432 E-55.1179905064321 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-50.0198674325519 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-230.019867432552 F8000
G1 X69.7640189871358 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-218.548216795051 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-214.860192178043 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-34.860192178043 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-38.5482167950509 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-22.0723514774678 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-20.2910087000184 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-200.291008700018 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X69.7640189871357 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-230.019867432552 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-50.0198674325519 F8000
G1 X69.7640189871357 Y163.485302816621 Z235.117990506432 E-55.1179905064321 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-38.5482167950509 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-34.860192178043 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-214.860192178043 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-218.548216795051 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-200.291008700018 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-20.2910087000184 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-22.0723514774678 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X69.7640189871358 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X69.7640189871358 Y163.485302816621 Z235.117990506432 E-55.1179905064321 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-50.0198674325519 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-230.019867432552 F8000
G1 X69.7640189871358 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-218.548216795051 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-214.860192178043 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-34.860192178043 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-38.5482167950509 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-22.0723514774678 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-20.2910087000184 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-200.291008700018 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X69.7640189871357 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-230.019867432552 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-50.0198674325519 F8000
G1 X69.7640189871357 Y163.485302816621 Z235.117990506432 E-55.1179905064321 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-38.5482167950509 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-34.860192178043 F8000
G1 X110.279615643914 Y160.549926791405 Z214.860192178043 E-214.860192178043 F8000
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-218.548216795051 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-200.291008700018 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-20.2910087000184 F8000
[... 94 lines cut ...]
G1 X102.903566409898 Y149.191897890333 Z218.548216795051 E-38.5482167950509 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-22.0723514774678 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-20.2910087000184 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-200.291008700018 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X69.7640189871357 Y163.485302816621 Z235.117990506432 E-235.117990506432 F8000
G1 X79.9602651348961 Y179.011856369478 Z230.019867432552 E-230.019867432552 F8000
[... cut ...]
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-200.291008700018 F8000
G1 X139.417982599963 Y127.536743948087 Z200.291008700018 E-20.2910087000184 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-22.0723514774678 F8000
G1 X135.855297045064 Y116.36916253409 Z202.072351477468 E-202.072351477468 F8000
T2; Switch Tool
>>>>>>>>>>>>>>>>>>>>>>>>>>>>G1 E0 F2000
T1; Switch Tool
G4 S3 ;Wait 3s 

G1 X0 Y0 Z0 E0 ; Back to Origin
;end

Harpye, 2015-05-02:
Used Excel to round all entries to .xx accuracy but did not reach the result I'm searching for!

thinkyhead, 2015-05-02:
Hmm, this is a puzzling behavior. I can only look at the code and hope the bug jumps out at me, and I can supply you with a version of Marlin that prints out values as it's running to try to pinpoint where the logic is getting screwed up. I do hope it's just a logical error and not something weird, like the timing of stepper signals to your third E axis. Is there anything different about that axis motor from the others, by the way?

Harpye, 2015-05-03:
Since my turntable  is limited in angle but until now I do not have a reference switch here I do use a single stepper with no load on this channel - nomodifications - the stepper ist connected directly to E2

As already mentionned short programs without many changes between the axis and Tool - Channels seem to be processed correctly. but as soon as it is getting more complex....

Harpye, 2015-05-03:
Perhaps the observing measures would be the right way???!

Harpye, 2015-05-17:
That is an idea - I never blamed the hardware to be the problem...before - would mean two failures on DIR and Microstep on one Stepstick... strange but I'll go for it - until then I would like to  show you something and thank you again for your kind help!

https://www.youtube.com/watch?v=evt_9DVsy2g&feature=em-upload_owner

closed 2016-04-23 by jbrazio