MarlinFirmware/Marlin #1799: Proposed menu change: Move power options to root menu

ntoff, 2015-04-04:
I know there's bug fixing going on but I wonder if anyone else feels like the "turn power on/off" menu should be in the root menu, instead of in the prepare menu? Do many people even make use of the ability to turn the printer on and off?

Also it would be nice if, while off, the LCD screen wasn't constantly updating but that's a story for another day.

thinkyhead, 2015-04-04:
I don't use the menu item since I switched to a different PSU, but even when I had an ATX I didn't really use the option. At the time I just inserted power on/off commands into the slicer. For an emergency stop, I know I would (even with this option) still reach for the physical power button. But I can understand why others may want this option closer at hand.

TechMasterJoe, 2015-04-04:
My 2 big printers are hard wired to a 15 amp breaker about 10 feet to the left of them no other way to turn them off lol and the big delta I'm building don't even use a power supply it's direct powered by the storage batteries for my solar system at 36v I have over 850Ah on tap so seemed like a cool idea btw delta runs nema 23's 5.18:1 geared using custom drivers with feedback encoders getting plain step and direction info. Really want ask about doing this over spi or i2c but that's another topic

ntoff, 2015-04-04:
It's not really for an emergency thing, it just seems more logical. All power buttons are close at hand, even windows' own shutdown command isn't really buried in another menu (well I guess it depends on the version of windows you use).

Oh well, I've changed it on my printer and like how it feels, just thought maybe others might want it but I guess if no one uses it then leave it alone.

TechMasterJoe, 2015-04-04:
i do agree with you i think it and babystep Z need to be moved to the first level because you need to change them quick.
for me the power up pin is been moved to fan 2 on the rambo and drives my LEDS
power up just turns on and off leds lol

ntoff, 2015-04-04:
well you can change the pins in pins_yourboard.h. I swapped around the bed wires and the fan because I wasn't comfortable with the big fat bed wires sitting right on top of a metal USB plug on the RAMPS/Mega setup.

TechMasterJoe, 2015-04-04:
I did a while ago just need to move baby step z. But Rather help with tracking down bugs And finishing the beast  (11foot tall delta)

thinkyhead, 2015-04-04:
:+1: I'm all for more accessible items, so you can count my vote.

closed 2016-05-04 by jbrazio