MarlinFirmware/Marlin #1804: Compiling error when enabling LCD current merges

Natealus, 2015-04-04:
I receive this compiling error now when I enable my Viki2.

Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

ultralcd.cpp.o: In function `_lcd_move':

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

collect2.exe: error: ld returned 1 exit status

Error compiling.

TechMasterJoe, 2015-04-04:
Need a config file

Natealus, 2015-04-04:
I don't think the configuration.h is it...only a difference between defining Viki2 and commenting out my LCD to compile successfully. Perhaps its something missing in the VIKI2 conditional section?

TechMasterJoe, 2015-04-04:
try defining REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
lets test to see if it's viki only or graphic lcd problem

Natealus, 2015-04-04:
Looks like I get the exact same error defining that one as well. I tried the Reprap Smart Controller without graphics as well. Same error.

Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

ultralcd.cpp.o: In function `_lcd_move':

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

C:\Users\Nate\AppData\Local\Temp\build2590251393999223928.tmp/ultralcd.cpp:673: undefined reference to `refresh_cmd_timeout()'

collect2.exe: error: ld returned 1 exit status

Error compiling.

thinkyhead, 2015-04-04:
Your `Marlin.h` line 230 should read:

    void refresh_cmd_timeout(void);

If it doesn't then you don't have the latest sources.

Natealus, 2015-04-04:
@thinkyhead Yes sirrrie it does so read so.

Natealus, 2015-04-04:
The section it's pointing to in ultralcd.cpp line 673

static void _lcd_move(const char *name, int axis, int min, int max) {
  if (encoderPosition != 0) {
    refresh_cmd_timeout();
    current_position[axis] += float((int)encoderPosition) * move_menu_scale;
    if (min_software_endstops && current_position[axis] < min) current_position[axis] = min;
    if (max_software_endstops && current_position[axis] > max) current_position[axis] = max;
    encoderPosition = 0;

Is that refresh_cmd_timeout(); supposed to be filled with something? I tried putting a void in the () but it didn't work hehe.

thinkyhead, 2015-04-04:
Well, in that case, try changing that line in `Marlin.h` to…

    void refresh_cmd_timeout();

…removing the `void`.

Natealus, 2015-04-04:
WaHa! found it! I was paying attention to the changes you did to line up all those void refresh_cmd_timeouts and noticed that the one in ultralcd.cpp which was last modified since the 31st. It's refresh_cmd_timeouts did Not have void in front of it. Insert and the yaaaay. LCD works again. It threw that in my pull there. :)

TechMasterJoe, 2015-04-04:
congrats

Natealus, 2015-04-04:
Doin a dance...alright time for sleep. I'll stare at code tomorrow while I'm printing something lol

closed 2015-04-04 by thinkyhead