MarlinFirmware/Marlin #1806: When doing PID Autotune, Error:Line Number is not last Line Number+1, Last Line: xxx causes PID autotune to stop

emartinez167, 2015-04-04:
I have been noticing when printing in Cura an ocassional message in the console about Line Number is not last Line Number + 1. I have disregarded this as it generally does not interfer with the print. However, it is a different story when trying to do PID autotune from Repetier. The graph just stopped and the process seems to stop altogether. Perhaps we can see where the real issue is where some process is not returning the proper line number as expected by the host?

thinkyhead, 2015-04-04:
I believe sometimes commands are allowed to go in ahead, if they have a higher priority. But maybe someone who knows more about the command queue can speak on this with more authority…

boelle, 2015-04-15:
Should we keep this one open? or maybe upgrade it?

Wurstnase, 2015-04-16:
Most time this happens when the host didn't wait for the 'ok'. Maybe Cura has some 'security' feature which sends the next command when the pause is to long? Did you tried some other host? Pronterface or Repetier?

emartinez167, 2015-04-16:
Will try with something else and report back.

boelle, 2015-06-29:
@emartinez167 what did you find out?

thinkyhead, 2015-10-12:
This should be fixed in the current RC (RC2). Please check and close this issue if it's fixed.

closed 2015-11-24 by AnHardt