MarlinFirmware/Marlin #1807: Adding support of 84x48 LCDs

Ovaday, 2015-04-04:
Hi there, is it possible to add to Marlin support of LCDs with resolution 84x48?
it's really common and cheap, one of them is from nokia 5110, there is a library for that Adafruit_PCD8544 which fully compatable with arduino
So it can be easelly connected to the Ramps instead of 16x2 so it can show more helpfull information for the user.
I haven't developed for marlin yet, so I don't know how to release this.
Could anyone do this?
It's just a need to configure pins, some settings in menu and include the library to the Marlin.
[image]

thinkyhead, 2015-04-04:
Take a look at the existing displays and how they're put together. The more interesting files…

    ultralcd.cpp - the common display routines
    ultralcd_implementation_hitachi_HD44780.h - a character-based display
    ultralcd_st7920_u8glib_rrd.h - low-level support for ST7920 - see the link below
    dogm_lcd_implementation.h - a graphical display

Here's good info on the ST7920, similar to your display in many ways:
http://www.bajdi.com/cheap-128x64-graphic-lcd-12864zw/

TechMasterJoe, 2015-04-04:
i might be able to write up some attiny84 code to make a I2C bridge for it 
this has the pro of level shifting as i have a few and they burn up at 3.6v+
at lest the ones i have then run it like a viki but lower rez ok i will fire up eagle and plain this out with a board and encoder i can make them crazy cheep i have all the stuff

Ovaday, 2015-04-04:
There are two Nokia 5110 LCD'c - one is blue and another is red. Blue LCD's always burn up at 5v, so it's a need to connect Vin at this LCD to the 3.3V(can be wired from arduino) or using a little scheme with resistors. On the other hand is red LCD's which fully tollerance to 3-5v, so it can be easelly be powered from the +5V at AUX pin.
There are lots of differences with all lcds which is compatable to Marlin now, so adding support of that lcd marlin would support all others with adafruit library

thinkyhead, 2015-04-04:
@Ovaday If you can make a small board that includes an encoder wheel plus this display, the wiring to the RAMPS extension pins is probably straightforward. Two 10-pin ribbon cables... I think you would have a popular add-on, and it would fit in with many compact printer designs.

Ovaday, 2015-04-05:
@thinkyhead Oh yep, I can do that, that isn't too hard. I would do it with encoder wheel, beeper and SD-card as Full Graphic SLCDC, but also I'll do the simple wire connector to the external E-Stop buttom for the people who uses Marlin as firmware for CNC machine.
I think at the end this board would cost about 10-15$, that is much lower than other existing boards.

TechMasterJoe, 2015-04-05:
toss an attiny or pic on board and do it all over I2C

Ovaday, 2015-04-05:
@TechMasterJoe I've planned to do that by AUX-4 and AUX-3 connectors, so I'll do that and then another board with just I2C connector, but that'll be much harder, I think

thinkyhead, 2015-04-05:
I can design the case in OpenSCAD. :smiling_imp:

Ovaday, 2015-04-05:
@thinkyhead I can curve the plexiglas сase and the board from textolite by CNC :)

Ovaday, 2015-04-08:
I've started doing the board, but it's examination time in my University so I can't do it fast.
It's better to set the pins from the LCD in Aux-4:
RST - 17;
SCE - 18;
D/C - 23;
MOSI - 25;
SCLK - 27.
If there are any Ideas you can tell me.

Ovaday, 2015-04-14:
I've done schematics at a half, and it can be cheaper than I've planned:
LCD 84x48 2$
copper board 70x100mm 1$
6 mini push buttons + beeper 0.5$
Engraving the plate 1-2$
Sd card slot 1$
So at all it can cost 5$, and I think it's really cheap.
I'm planning to do 5 mini push buttons because I think it's much better than the joystick, that can be damaget easelly in lasercut machine or CNC. The other one button is for the E-stop, there is also two connectors for external E-stop.
@thinkyhead @TechMasterJoe About I2C interface:
I've googled that one person from my country has tried to make this LCD work throught I2C by Attiny, but finished it with failure, so I think that my results will be the same. I have the alternative LCD that is cheap enought and have support of Marlin by I2C - that's 128X64 OLED LCD:
[image]
So what do you think about all of that?
I can do the board that will support that 2LCDs, so the owner can easelly remove one and connect another into the second slot. But I have to be sure that it'll be support of the Nokia 5110 LCD.

AnHardt, 2015-04-14:
Some thoughts.
84x48 gives about 16 columns x 4 rows. That fits to the menu system we use for the hitachis. Sadly this is a graphical display like the ones we use with the u8glib. The things we do with u8glib at the moment totally rely on 128x64 pixel. So, integrating this display will take a good amount off time.

Buttons? No encoder? I don't like it. Not very comfortable.

128x64 - much better - ideally compatible with u8glib - software spi. On the picture above i miss a pin like CS or Enable - without this we will run into conflicts with SDcards and MAX6675.

Ovaday, 2015-04-14:
@AnHardt 
Ok, I can do the encoder, I thought that buttons more comfortable.
The full name of the LCD:
SSD1306 128x64 OLED
Yep, It's fully compatable with u8glib and this version of LCD can be connected directly to I2C, like in the video below:
http://www.youtube.com/watch?v=kUAczzPTgOQ

And about pins:
"The SSD1306 has two different signal lines for the I2C data signal (one for input and one for output). However the pins of this OLED module are only connected to the input data line of the SSD1306. As a result it can not send the I2C ACK (it will be also invisible to any I2C scanner). This display from HelTec is connected to 5V power supply. Data and clock lines accept 5V signals and do not need a pullup resistor. Support for this OLED is added to v1.16 of U8glib."

So it's no need to connect it throught AUX, It can be wired to I2C, SD card and joystick with beeper to AUX. How about that?

AnHardt, 2015-04-14:
@Ovaday 
That sounds much more promising.

Ovaday, 2015-04-14:
Well, waiting for the other developer's opinion.
And is there a need in such LCD controll board?

AnHardt, 2015-04-14:
Need? Only when considerable cheaper than REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.
So cheap, that you don't want to have the 4 times bigger one.

boelle, 2015-06-29:
will close this one as we have agreed that support for hardware should come from makers of the hardware and not us

closed 2015-06-29 by boelle

thborges, 2020-06-13:
Five years late but if someone came here by google, I took the effort while building my own 3D printer. Check the code [here](https://github.com/thborges/Marlin/tree/stm32bluepill/) and a [brief description](https://thborges.github.io/blog/3d/2020/06/13/interactive-status-screen-marlin-firmware-pcd8544-nokia5110-lcd-controller.html).

[image]

victornpb, 2020-06-14:
@thborges I was just looking for this today and came across this
Nice