MarlinFirmware/Marlin #1815: Raise Z befor moving to Z_SAFE_HOMING

Gummibaer, 2015-04-04:
Please implement a Z raise for G28 Command
Befor Homing Z -> raise Z -> then moving to Z Safe

thinkyhead, 2015-04-04:
@Gummibaer Currently it's enabled only for systems using auto-bed leveling, but I agree it should be available for all setups.

TechMasterJoe, 2015-04-05:
testing that very thing got bored so printed a new control knob and did it at 100mm/s with silc3r 
[image]
[image]
[image]

Gummibaer, 2015-04-05:
@thinkyhead i have auto-bed leveling.
I need to send a G28 before the G29. by G28 crashed the hot end with a Screw.
I use corexy 
my solution in slic3r config
G1 Z10 ; raise Z 10mm
G28; homing all axes
G29; auto-bed leveling

thinkyhead, 2015-04-05:
Have you tested the latest code? There was a fix related to Z raise before homing within the last 20 hours or so. `G29` (and maybe `G29 E`) bed leveling are still wrong, so for the moment auto bed leveling needs to be compensated with a z homing offset, or go back to an earlier Marlin before Z raise was moved into `probe_pt()`… about 2 weeks ago.

Gummibaer, 2015-04-05:
I have last from 04.04.2015 17:32.
this is a feuture request. Raise Z before homing

thinkyhead, 2015-04-05:
What is your current setting for `Z_RAISE_BEFORE_HOMING`?

Gummibaer, 2015-04-05:
#define Z_RAISE_BEFORE_HOMING 10
but it dosent work

Homingdirection
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

thinkyhead, 2015-04-06:
@Gummibaer Using a cartesian printer or a Delta?

Gummibaer, 2015-04-06:
Yes its a cartesian. (the printer name is SparkCube)
Y is home direction to MAX

jbrazio, 2016-03-10:
I believe this feature request was implemented with #3107.

thinkyhead, 2016-04-25:
`MIN_Z_HEIGHT_FOR_HOMING` works for me!

closed 2016-04-25 by thinkyhead