MarlinFirmware/Marlin #1816: Board locks up with beeper frequency duration

ntoff, 2015-04-04:
I have a beeper on pin 32 instead of 33, not sure if that matters but when setting
*#define LCD_FEEDBACK_FREQUENCY_HZ* to 150 and *#define LCD_FEEDBACK_FREQUENCY_DURATION_MS* to 130 or less, the beeper sounds constantly and the LCD panel no longer responds. Setting the frequency to 131 causes the beeper to make no sound at all and setting it to 132 causes it to make a very short chirp. Anything 131 and up doesn't cause any lockups.

RAMPS 1.4 using the latest dev version.
LCD panel is defined in config.h as *#define ULTIMAKERCONTROLLER* and is the LCD panel off a Velleman K8200.
The beeper is this: http://www.freetronics.com.au/products/sound-and-buzzer-module because the LCD panel doesn't have a built in beeper.

TechMasterJoe, 2015-04-04:
This honestly sounds like a 5v power problem if you have a 10uf capacitor handy add it to the 5v supply line and see if that helps. Another idea is to set tones the lock it up then unplug the beeper  and reset see if it locks up with it unplugged

ntoff, 2015-04-04:
It's only plugged in via USB, maybe I'll try it today with it plugged into the main power. Why would it only be certain combinations of frequency and duration though? It happens at 500hz and some of the ms values that worked, stop working.

boelle, 2015-04-07:
did it help with mains power`?

ntoff, 2015-04-11:
Nope, certain combinations of frequency and duration still cause a constant beep to sound even when plugged into the main power supply.

I find it odd that one certain frequency that works, will cease to work at different durations.

I've just finished putting the test printer hardware together so I'll re-download the latest version and try with that since I see there's been some changes.

Lukelectro, 2015-04-12:
Is there any way to set frequency and duration without changing the defines and re-uploading? That would make testing easy-er.

How is that frequency generated? Hardware timer module or software?

ntoff: do you get any error messages when in hangs? (On the LCD or on the serial port)
For example "Something is wrong, please turn of the printer." on serial?

Lukelectro, 2015-04-12:
Beeper did at first not work at all, because M300 gets ifdeffed out if LARGE_FLASH is not defined. Whoever thought of that being useful owes me about 3 hours of my life back... 

Anyway, M300 S1500 P300 gives a beep now, and it does not hang.
M300 S1500 P20 works as well. M300 S440 P30000 gives 30s of 440Hz. M300 S440 P40000 however keeps beeping way over 40s... So does M300 S1500 P40000. After 2 minutes I did reset my printer to stop the beeping. It did not give any error messages on serial.

Maybe a enforced maximum duration is a good idea.

thinkyhead, 2015-04-13:
@Lukelectro I think you have found a bug in the number that can be stored into an integer. The `P` parameter needs to have a larger variable. I will add this patch. I'm not sure why `LARGE_FLASH` is needed, but almost all boards have it defined. Which board are you using? Maybe it should be defined for yours. One problem might be related to this note in the documentation of the `tone()` function:

    Use of the tone() function will interfere with PWM output on pins 3 and 11 (on boards other than the Mega).

But `LARGE_FLASH` is defined for boards other than the Mega, as far as I can see, so that probably isn't the issue…

thinkyhead, 2015-04-13:
I'm also wondering, can I use this code:

        tone(BEEPER, beepS, beepP);

in place of what's in `M300` currently:

        tone(BEEPER, beepS);
        delay(beepP);
        noTone(BEEPER);

or is the `delay` required even if the duration parameter is used for `tone()`?

ntoff, 2015-04-13:
One should never use "delay" anyway since it pauses the loop. Should use the millis thing. Basically delay locks everything until the delay has passed. I wonder if that's what it's doing,  getting confused on the delay so it's just playing the tone, hitting the delay() which is somehow set to some stupidly large number so it's just sitting there playing the tone over and over and over.

No, there's no error messages or any response at all. It's completely locked. The only thing I can do is hit the reset button, or reconnect in pronterface (which I believe is the same as resetting since the FTTDI chip sends a reset signal every time I connect).

@Lukelectro for future reference if you have trouble shooting to do http://arduino.cc/en/Hacking/PinMapping2560 contains a mapping of what the arduino numbers are vs the actual port name / numbers of the mega2560.

>Is there any way to set frequency and duration without changing the defines and re-uploading? That would make testing easy-er. (

I would love to be able to move my config file over to an ini file like how smoothieware works. You upload the firmware once and then a configuration file is read from the SD card. That makes changing settings easy and means they were persist over different firmware uploads.

Lukelectro, 2015-04-13:
I'm not sure the P parameter will need a larger variable, but imho it 
would be a good idea to prevent overflow somehow.

I'm using a Melzi, with a atmega644. I know 1284 is standard now, but 
there is a 644 on mine. If it had the 1284, large_flash would be defined.
The thing is that it says nowhere that some mcodes are ignored when you 
don't have large flash. Also, m300 just replies "OK" when it is ifdeffed 
out. In fact, even M1111 or other non-existent codes get the OK reply... 
A  "Err" reply would be realy usefull to detect unrecognized commands.

Lukelectro, 2015-04-13:
I deliberately tested with large delays to try to cause a WDT timeout, 
but didn't. So either delay() resets the WDT periodically (which kind of 
defeats the purpose of a WDT), or WDT is disabled (which is worse). And 
Yep, somewhere around 30/40 seconds delay it "locks up", without error 
messages. WDT timeout should reset the MCU and send error message.

You can set some settings in EEPROM. Not as convenient as a file on a sd 
card but still better then re-uploading the firmware... (Disclaimer: I 
have not used this feature (yet?).)

I have a Atmega644, not a 2560. So I looked it up in its pins.h file 
somewhere hidden in Arduino. (Arduino: hides everything you actually 
need, for your convenience :) )

I'll probably look into this further. Maybe next week-end.

thinkyhead, 2015-04-13:
___I would love to be able to move my config file over to an ini file like how smoothieware works.___

As a middling step you can copy the output of `M501 S0` and put it into a file on an SD Card - name it `my_config.gcode`.

ntoff, 2015-04-14:
>You can set some settings in EEPROM. Not as convenient as a file on a sd
card but still better then re-uploading the firmware... (Disclaimer: I
have not used this feature (yet?).)

Yeah all the steps / mm and things are stored in EEPROM on my Mega + RAMPS, but for changing things like the beeper, or the actual build dimensions (unless I'm wrong?) you need to change and re-upload the firmware for that.

It'd just be kind of nice to have a config.ini file in the root of the SD card with steps/mm, build volume, acceleration values, thermistor table selection, and basically anything that's a variable between printers and not something likely to remain constant (like the type of board it's running on).

Also while I'm here, what's the easiest way to get new versions of Marlin to test? Should I just keep downloading the zip file every once in a while? Or take the time to properly set up git so it just syncs whatever's up here to my laptop? I don't really plan on making any pull requests.

thinkyhead, 2015-04-14:
@ntoff Best way to interface with Github is to get the Github desktop app. If you don't plan to make changes to code then it's fine to just use the __Clone to Desktop__ button on the main Marlin project page.

AnHardt, 2015-04-22:
#1944

thinkyhead, 2015-10-12:
This should be all fixed in the latest RC (RC2). Please check and close this issue if it's fixed for you.

closed 2015-11-06 by AnHardt