MarlinFirmware/Marlin #1820: dual z drivers fails to compile

ntoff, 2015-04-05:
Latest dev version downloaded a minute ago, test compile without changing anything and it works, uncomment out #defiine Z_DUAL_STEPPER_DRIVERS to use dual drivers for Z axis throws a bunch of errors

http://pastebin.com/Maj5qbT8 the errors

thinkyhead, 2015-04-05:
Are Z2 min and max endstops both required for this feature?

boelle, 2015-04-05:
only one endstep at each side should be required... 

for my own mendel90 i would use 1 max endstop to tell the machine the overall height... and 2 min endstops to level X carriage so its parallel with the bed...

for normal printers 2 min endstops should make do

alexborro, 2015-04-05:
I don't know WHY someone changed that.. my original implementation is:

static bool old_x_min_endstop = false,
            old_x_max_endstop = false,
            old_y_min_endstop = false,
            old_y_max_endstop = false,
            old_z_min_endstop = false,
            #ifndef Z_DUAL_ENDSTOPS
              old_z_max_endstop = false;
            #else
              old_z_max_endstop = false,
              old_z2_min_endstop = false,
              old_z2_max_endstop = false;
            #endif

Once Z_DUAL_ENDSTOPS is defined, both old_z2_min_endstop and
old_z2_max_endstop are declared.
These variables declaration do not depend on one or two endstops, the code
take care of it.

Cheers.

Alex.

ntoff, 2015-04-05:
@thinkyhead no idea, all I know is I used to use dual Z drivers in a previous version because 1 driver can't drive both my Z motors, enabled it in the latest dev version on the printer I'm building (building one just to test dev version of marlin because I love you so much) and it failed. If you comment out the dual Z end stops then it compiles ok, with both drivers still enabled.

commenting out the dual z endstops thing fixes it but isn't that what the whole dual endstops was for? to even out the homing of dual z drivers automatically?

thinkyhead, 2015-04-06:
#1827

ntoff, 2015-04-06:
will test tonight. currently chopping down bushes and avoiding spiders

boelle, 2015-04-07:
how did the test turn out?

thinkyhead, 2015-04-08:
It compiles and optimizes away unused stuff, I presume.

closed 2015-04-08 by thinkyhead

ntoff, 2015-04-08:
>how did the test turn out?

Dunno, was waiting on plugs for the end stop switches. I was going to wire it up without them but now I have all the necessary bits I can continue building the test printer. This way I can upload random firmware willy nilly and not have to worry about whether or not it destroys the printer in some spectacular fashion.

thinkyhead, 2015-04-08:
@ntoff Apologies in advance. Please record disasters in progress on video.

ntoff, 2015-04-09:
well the home made hot end didn't set itself on fire, so that's a good start http://imgur.com/9lo6mz9