MarlinFirmware/Marlin #1824: RRD Full Graphics Displays, Displays marlin splash screen whenever knob is turned instead of moving cursor or changing values

Sniffle, 2015-04-05:
As the description says, on the current dev version and for the past 2-3 weeks of Dev commits, the Rep Rap Discount Full Graphics controller doesnt do anything other than display whats going on.  trying to adjust or manipulate anything through the menus is impossible due to it displaying the marlin 1.0.2 splash screen when the knob is turned.  

you can access the menu by pushing the knob and it will display the first menu, but navigating the menu is impossible. as the splash screen is displayed, you can then push the knob again and it will go back to the main display.  manipulating feedrate on the main screen by turning the knob is also impossible for this same reason.

not that this should have any bearing on this as it hasnt until recent changes, but i do disable the beeper pin in pins.h with a -1 so that it doesnt beep.  just mentioning it for full disclosure as the beeper is on the display.

Sniffle, 2015-04-05:
I should have looked at pull requests first looks like @AnHardt is already on this with #1823.  

I really need to add myself to the list of testers.

AnHardt, 2015-04-05:
Confirmed for when LCD_FEEDBACK_FREQUENCY_DURATION_MS = 0.
#1823 will not fix this -only the splash-screen will not appear.
This is quite the same effect as in #1763, where the call for the beep was missed.

If no one has a better idea i'll implement a little break when beep is not defined or to short.

Sniffle, 2015-04-05:
@AnHardt I think at this point you are pretty much the display guy :-)  I don't think anyone would complain about you fixing it

AnHardt, 2015-04-05:
@Sniffle 
Cant reproduce this anymore. I know i have seen that. Checked out standard Configuration.h. Committed my last changes. Switched to Development. Pulled last Marlin. Configured just display and LCD_FEEDBACK_FREQUENCY_DURATION_MS = 0. Gone. Can't see your error. Switched back to last branch. Configured. Gone. Searched for hours. Gone. I'll give up for today.

Sometimes i believe on an other memory corruption.
Thing like this (from german RepRap Forum): 
[image]
normally can't happen. The algorithm has no chance to produce this symbols.

Sniffle, 2015-04-06:
makes me wonder if it is RAMBO specific, maybe @avluis can see if it is a specific problem to me and shorting the beeper or if it is something else...

but to double check i'll probably flash back to the 1.0.2 tag to make sure that it does still work again on the older firmware

TechMasterJoe, 2015-04-06:
i/n a rambo user with that same display new and old firmware works fine for me granted still working on abl bugs

Sniffle, 2015-04-06:
could be that i edited pins.h to make take the beeper to -1 instead of it's pin...

AnHardt, 2015-04-06:
The corrupted display for 'X steps/mm' has gone with a M502 M500.
Do we need plausibility checks when reading from EEPROM?

Sniffle, 2015-04-06:
i never enable eeprom because it always plays hell with auto leveling if you forget to m500 m502 while tuning... more trouble than its worth, but as a test i could enable then disable to see if it is a corrupt memmory block somewhere.

Sniffle, 2015-04-06:
OK, turned back on the beeper, no dice, enabled and disabled eeprom using m500 and m502 in between, no dice rolled back firmware to 1.0.2, no dice... 

So at this point i am at a loss, gonna double check my cables just in case... but it shouldnt be displaying anything if the cables weren;t right.. something might have gone wrong with the display... sigh time to do super in depth testing

avluis, 2015-04-06:
@Sniffle Trying to reproduce this, no dice just yet.

Sniffle, 2015-04-06:
Yeah i'm beginning to think it's a hardware problem and not a software one... i'm probably going to pull it apart and re-tin all of the pins on it to see if it was maybe a cold solder joint somewhere that broke loose even though I don't have any bolts on the screws to add pressure to the board because the screw holes dont line up well.

thinkyhead, 2015-04-07:
If a small delay would help, consider adding it when the feedback duration is 0. But if the bug is gone, then :grinning:

boelle, 2015-04-15:
@Sniffle is the bug gone in the latest dev?

Sniffle, 2015-04-16:
I think something is up with myhardware, i rolled back to the 1.0.2 tag and
it's still causing issues... Still need to roll back further to tripple
check, but i have a lot to print right now so it is on the back burner for
the moment...

theccm, 2015-05-26:
Hi,
Happy to see that I'm not the only one who have this problem :) I have the same behaviour, I tried many version, 1.01 1.02 and dev (from 1 month ago) but it does not work. It did work a few minutes but I flashed just after and I was not able to got it work again. 

So Do you need any file to investigate? Configuration.h. ?

My LCD was bought on eBay with ramps connector, I did a melzi adapter founded on thingiverse because I have a melzi.  Then just uncomment "Lcf full graphic..." In configuration.h

thinkyhead, 2015-05-27:
It sounds like the knob-turning action is being treated like a button-press action. We should check to make sure the right pin numbers are assigned to these functions.

theccm, 2015-05-27:
Yesterday I tried to plug in the RRD LCD and it worked great although I didn't change anything ! It worked great until I tried to print something, as soon as the Z-axis moved then the screen started to blink again.
And I was not able to make it work again. I tried to change value LCD_FEEDBACK_FREQUENCY_DURATION_MS without any luck.

In my case you don't have to press or turn the button to see the problem. Without doing anything the screen alternates between menu and logo, with a beep. Also on the melzi board you have a LED and the led is blinking at the same time you hear the beep.

AnHardt, 2015-05-27:
@theccm 
If you experience changes in behavior without changing the configuration, look out for solder bridges, cold solderings, broken cables, defect switch, ...

boelle, 2015-06-29:
@Sniffle have you had a chance to check more on this one?

Sniffle, 2015-06-29:
Ill be doublechecking this this week but i think it might have been user
error on my part... I couldnt see it with my enclosure at the time but i
think i had the plugs off by 2 pins.

closed 2015-10-05 by AnHardt