MarlinFirmware/Marlin #1826: Add a hard coded code to wipe EPROM with zeros

TechMasterJoe, 2015-04-05:
This is a  feature request but I think it will help with a lot of bugs I have seen. We need a way to tell marlin to erase all unused storage and dump ram. It be best if in future version it can detect a new ver of marlin has been installed using a self generated key so even if you replace it with the same version the hash key forces a full Memory and EPROM over write
Maybe some type of key gen the calls config values as a seed value.
Well this is just an idea

Natealus, 2015-04-06:
Doesn't Arduino IDE have a sketch to do this? Examples > EEPROM > eeprom_clear?

TechMasterJoe, 2015-04-06:
ya just think it needs to be apart of firmware updating

Natealus, 2015-04-06:
There might be a way to add such an include but for the firmware update? That might be more of the IDE program side of things. Unless we could do it from the "Configurator" which I am scared to use cause I know what variables I'm messing with now...I don't know if I trust a web applet to do it for me...lol. :)

boelle, 2015-04-06:
Ya.... I agree partly with @Natealus 

But it has to be done with care so that important settings are not trashed.... Some might have tweaked their setting in EEPROM and if not thinking carefull they could wipe their settings

maybe couple this request with this one: https://github.com/MarlinFirmware/Marlin/issues/1185

in that way you can safely save the settings before a wipe...

Natealus, 2015-04-06:
I wonder if there's a bit of code we can enter that knows when a rom flash is happening and after the main flash it clears the EEPROM. Then the firmware just loads the "hardcoded" defaults and auto saves to EEPROM. Is there a way to do such a script?

boelle, 2015-04-06:
and if user has settings saved in eeprom? do you just want to trash those?

Natealus, 2015-04-06:
So there are settings that are Only in the EEPROM and don't get stored in the config files anywhere? I must not use those features yet hehe. But if that's the case then we can make a selective wipe which leaves settings that do Not have a hardcoded value alone.

boelle, 2015-04-06:
thing like steps pr mm etc can be saved in eeprom... it was just one i
could mention on top of my head.... but there are many other settings

Natealus, 2015-04-06:
Well I know you can change and override with the epprom but Steps per MM is in the config file. If you pulled M502 it would pull that hardcoded value.

The only one that comes to the top of my head is the filament change routine which I haven't checked out yet.

thinkyhead, 2015-04-07:
Wiping EEPROM with zeros is not really necessary, and those zeros would be bad values if they were ever read back. Instead, just write a flag value to the first byte of the EEPROM that, if present, means the EEPROM should not to be read back. Then either disable `M501` or when the `M501` is used, do an `M502` instead.

As it happens, whenever the EEPROM version number changes, the values in the EEPROM are reset to defaults, and the defaults are used in the firmware. A message to this effect appears in the serial output.

Roxy-3DPrintBoard, 2015-04-07:
Just as an FYI:    Many people had difficulty getting their Auto Bed Leveling running.   In many cases, we traced the problem back to bad values in the EEPROM.    As a result of this, an extension was done to 'Invalidate the EEPROM'.   It is here:  http://3dprintboard.com/showthread.php?8201-M499-EEPROM-Invalidate-Command

It does something very similar to what ThinkyHead just mentioned.  It zaps the value in the EEPROM so the start up code thinks a different version of the firmware created the values stored in EEPROM.   It was an easy way to insure bad values were not being pulled out of the EEPROM and used in critical routines.

In theory, a person can just reset everything to default and store those numbers.  In practice, people just did not understand that and it resulted in a huge amount of effort being expended to support people as they migrated to Auto Bed Leveling.   

I'm definitely in favor of adding some kind of support to invalidate the EEPROM!

nophead, 2015-04-07:
Is it so much harder to type M502 M500 than just M499 that it warrants a
new code?

Roxy-3DPrintBoard, 2015-04-07:
> Is it so much harder to type M502 M500 than just M499 that it warrants a
new code?

No...  Obviously not!   But I can tell you first hand that people trying to get Auto Bed Leveling going did not realize they had done a M500 in the past and that it froze some values from taking affect from the Configuration.h file.   Having a single command that invalidates the EEPROM is much easier for people to grasp.   

And in fact, the current setup doesn't allow the EEPROM to be invalidated.  If a set of numbers is ever stored in the EEPROM, it will continue to be used unless explicit action is taken to change the version number of the information or turn the EEPROM functions off.  Both of these require rebuilding and flashing the firmware.    I think it can be argued that is not 'correct'.  I should have control of what is stored (and used) in the EEPROM without having to turn the EEPROM off.   

I understand why it wasn't done originally.  Doing a M502 and M500 pair pretty much addresses the  issues.  But from an 'ease of use' perspective, having an 'Invalidate EEPROM' command does help people.

scotty1024, 2015-04-07:
All I can say is I'd rather wiping my EEPROM took two steps rather than one. And when wiping it I'd rather go straight to factory defaults than zero values and hope they get updated as needed to better values.

thinkyhead, 2015-04-08:
I can see this as a developer feature more than an end-user feature. One of the things that is on the slate is to make a Developer (or Debug) Info Screen in the LCD menu, which a double-tap from the Info Screen would switch to. It would show extra info pertinent to debugging and development, and some extra menu items for doing diagnostics would also be added. This would make a convenient place to collect options like resetting the EEPROM, or even just testing it.

But, we could add a hash to the EEPROM, written in the last position, to check the integrity of the data. Currently it just looks at the stored version string versus the current version string and compares them. A checksum is easier said than done, perhaps. Also, with some clever macros in ConfigurationStore the processes of reading, writing, and resetting EEPROM values can be consolidated together, so that they are not scattered in 3 separate places. This kind of general cleanup needs to be done anyway before any rehashing, because the data relies on a very ad hoc structuring method – not a bad one necessarily, but definitely hard to read and maintain.

Roxy-3DPrintBoard, 2015-04-08:
> Still, with some clever macros in ConfigurationStore the processes of reading, writing, and resetting EEPROM values can be consolidated together, so that they are not scattered in 3 separate places. 

You mentioned something similar to this a while ago.    Arduino is open source.   At the very worst case...  (And I'm sure it doesn't come to this) we change Arduino so it automatically sets an EEPROM version based on build version, date, etc, etc.    The end user shouldn't even see these problems (in a perfect world).  Some how the system should invalidate the EEPROM automatically.

nophead, 2015-04-08:
Rather than invalidating it perhaps there should be an EEPROM enable setting stored in the EEPROM that determines if the other setting will be ignored despite being valid. Then you can turn EEPROM on and off at run time without losing the values stored in it.

thinkyhead, 2015-04-08:
EEPROM switch in the EEPROM is a good idea, easy enough to add.

Natealus, 2015-04-09:
Out of curiosity @thinkyhead while you're looking at the EEPROM would it be possible to add MANUAL_XYZ_HOME_POS values to the EEPROM so that one could adjust height easier from computer control? Mostly concerned with Z. As it is, I have to reflash the firmware if I want to adjust height without using the LCD. 

I still disable it because of the big impact on printing performance and I've verified (I don't know how far yet) that turning up Buffers in configuration_adv.h has allowed me to turn up the segments per second significantly without the LCD enabled. But doubling the block buffer from 16 to 32 while LCD enabled seemed to yield about the same results of a smooth print at 58 segments per second.

It must be something specific to the LCD subsystem code that causes these interruptions. Because so far I've gone from 165 to 200 segments and still print smoothly without LCD enabled and a 64 block buffer.

thinkyhead, 2015-04-09:
When adjusting height for a Z endstop that I don't want to move, I usually use the __Set Home Offsets__ option from the LCD menu. Those values can then be flashed.

Wurstnase, 2015-04-09:
@nophead Is M501/M502 not for that? So simply reading the default settings by M502 on startup without store them with M500. If you want to use the EEPROM-values you can load them with M501.

Natealus, 2015-04-11:
I'm saying that MANUAL_Z_HOME_POS isn't in the EEPROM and if you don't have the LCD enabled you have to do it from the firmware and reupload to the board. In the course the bed temp drops and I have to wait for it to climb back up again after the upload flash is finished. Am I missing an M code or something that adjusts the MANUAL_Z_HOME_POS without reflashing the firmware? I see the one to adjust the Delta endstops but that's not what I'm adjusting. I'm adjusting the overall height from the top to the bed if its too high and not printing the layer appropriately.

On a secondary note this is without auto bed leveling enabled cause I've been having some issues with ABL and needed to print some revised parts. So it has nothing to do with Z_PROBE_OFFSET.

thinkyhead, 2015-04-11:
@Natealus The `Z_HOME_POS` can come from `MANUAL_Z_HOME_POS` at compile-time, so I think what you are looking for is to be able to set and store the position of each endstop, especially Z. You are correct, they are set as constants and are not available to be changed at run-time. You can understand why these are fixed values. They are presumed to remain fixed and to be reference points for the position of the axes. Offsets can be applied to slicing, so it makes less sense to be able to change them. However, technically it can be done. (It already applies the `home_offset` at homing, essentially the same thing.) So, I suppose we need to build some consensus on why this should be used instead of other methods (such as hard-setting the current position with `G92` after homing, etc.).

Natealus, 2015-04-14:
Yeah on a delta...when you call a move to Z it moves all 3 of the towers equally. Not just the Z tower or the endstop on that tower. Z is the total height from contact on the endstops to the plate and if you are calibrating and changing things like radius or layer size, you would need to move the the nozzle closer to the bed or away depending on the layer size or if one changes the nozzle or in the Kraken's case...it also has several mm of adjustment in the heat breaks that can change that height significantly. 

So there's been many cases that I've made some changes and needed to change the Z height and had to recompile and upload the whole firmware...shutting the heaters down and in order to get accurate calibrations they need to be heated back up. Bed taking the longest by far. If I could change it with an M code or something then it would save Days of time reuploading and waiting for the temps to come back to calibration point in my case. 

Perhaps in other setups that's an odd thing to do lol, but I've only used a delta so far. Do other people have a more efficient way that I missed?

thinkyhead, 2015-04-14:
@Natealus So what is the effect of `M206 Z` on homing behavior? It should cause printing to start higher when set negative.

Natealus, 2015-04-15:
@thinkyhead I shall give it a try should technically work ya? :)

boelle, 2015-06-29:
will close this one... it adds more complexity than its worth and there are allready a sketch with ardriuno that can wipe eeprom

closed 2015-06-29 by boelle

thinkyhead, 2015-07-03:
@boelle And it's better for Marlin to wipe EEPROM by setting it to default values, not zeros. Currently you can do this with `M502` followed by `M500`.