MarlinFirmware/Marlin #1830: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER not working

lcfm1, 2015-04-06:
screen is not earned until I added the following parameters:

#define DOGLCD 
#define SDSUPPORT
#define U8GLIB_ST7920

AnHardt, 2015-04-06:
@lcfm1 
Can it be, that for some reason, you use an old 'Configuration_adv.h'?
Than 'Conditionals.h' would not be included.

May also help against #1832.

lcfm1, 2015-04-06:
No, I took the last archive, taking the delta  printer settings from examples in the subfolder and got this error

alexborro, 2015-04-06:
That may be the problem.. The DAMN specific files.

We should just include some specific file after the configuration.h and
configuration_adv.h to redefine a few specific labels.

Cheers

Alex

lcfm1, 2015-04-06:
http://pastebin.com/jfGBa8fA my conf.

AnHardt, 2015-04-06:
Would you please test to insert

    #error "Test error"

in line 46 of 'Conditionals.h'.

If you can still compile without errors, please insert the same line into

line 5 of 'Configuration_adv.h' right after #include "Conditionals.h" 

Than please report back.

lcfm1, 2015-04-06:
got error: Conditionals.h:46:6: error: #error "Test error"

AnHardt, 2015-04-06:
Ok.
Please remove the error line.
Please undefine 
    #define DOGLCD 
    #define SDSUPPORT
    #define U8GLIB_ST7920
in 'Configuration.h'

to see if your problem is gone.

lcfm1, 2015-04-06:
hmm... works. what it's mean?

lcfm1, 2015-04-06:
ok. I downloaded the latest build for today, filled in Arduino and nothing happened. screen clean. http://pastebin.com/fWTFBSET

lcfm1, 2015-04-06:
it did not work until I uncommented line #define REPRAP_DISCOUNT_SMART_CONTROLLER

AnHardt, 2015-04-06:
We can not define all dispays at once - they conflict with each other.
#define REPRAP_DISCOUNT_SMART_CONTROLLER
is the place to tell the compiler:
'Yes i have this display. Please switch it on for me.'

AnHardt, 2015-04-06:
In Conditionals.h the magic happens that if REPRAP_DISCOUNT_SMART_CONTROLLER is defined, automatically all options you mentioned in your first posting will be defined.

lcfm1, 2015-04-06:
AnHardt, I think, for so many screens - it's time to start a separate procedure, by analogy with motherboards

closed 2015-04-07 by thinkyhead