MarlinFirmware/Marlin #1832: #define LANGUAGE_INCLUDE GENERATE_LANGUAGE_INCLUDE(ru) error

lcfm1, 2015-04-06:
got error:

Marlin_main.cpp: In function 'float probe_pt(float, float, float, ProbeAction, int)':
language_ru.h:50:45: error: 'LCD_STR_THERMOMETER' was not declared in this scope

AnHardt, 2015-04-06:
@lcfm1 
Is this still a problem?

lcfm1, 2015-04-06:
i found problem. in line #define WELCOME_MSG                         MACHINE_NAME " Готов."

not enough LCD_WIDTH for WELCOME_MSG

"Mini Kossel Готов." = 18 symbols.

replace with #define WELCOME_MSG                         MACHINE_NAME " Готов" in line 16

KiteLab, 2015-04-06:
@lcfm1 
Thank you for this tip.
Indeed 'char lcd_status_message[LCD_WIDTH+1]' is a bit short since we can use UTF, what needs 2 places for one symbol.

SkyFlyer2, 2015-04-07:
Странно, я только вот на днях проверял, все там ок было.
Надо только не забыть кодировку русскую добавить, такой дефайн есть в обновленном configuration.h.

Речь идет о последней dev-версии, в ней все ок. Но в версии 1.0.2 будут ошибки!

@AnHardt
I've already checked a few days ago, Russian characters were all compiled and no errors seen.
I mean the latest dev-version.
But vesion 1.0.2 contain bugs, of course.

closed 2015-04-08 by thinkyhead