MarlinFirmware/Marlin #1833: PROBE_OFFSET_FROM_EXTRUDER error

lcfm1, 2015-04-06:
when i input param

  #define X_PROBE_OFFSET_FROM_EXTRUDER 10     // Probe on: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 24   // Probe on: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -6  // -below (always!)

i got error 

SanityCheck.h:147:10: error: #error "The given FRONT_PROBE_BED_POSITION can't be reached by the probe."
         #error "The given FRONT_PROBE_BED_POSITION can't be reached by the probe."

SanityCheck.h:174:10: error: #error The Y axis probing range is to small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS
         #error The Y axis probing range is to small to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS

TechMasterJoe, 2015-04-06:
can you post your config

lcfm1, 2015-04-06:
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif

// Define this to set a custom name for your generic Mendel,
#define CUSTOM_MENDEL_NAME "Mini Kossel"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
// #define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
[... 710 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

boelle, 2015-04-06:
could you edit that post and put the configs in a pastebin.com please?

lcfm1, 2015-04-06:
http://pastebin.com/jfGBa8fA

Natealus, 2015-04-06:
So your bed is is about 150mm circular diameter? One thing you can do to gain more probe offset is to change #define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10) to more like 20 or 30 so it virtualizes the probing space smaller to your print space, allowing for more offset. You can also turn the grid leveling points down from 9 to 6 or inbetween. Mess around with it stepping the variables until you can compile. I found a set of variables to do a -39 Y offset.

thinkyhead, 2015-04-07:
We may have to incorporate a separate `DELTA_MOVABLE_RADIUS` that specifies a wider radius beyond the size of the printable area.

Natealus, 2015-04-09:
I'm going to adjust the physical variable...make it actually closer to the hotend too. But that may help too.

boelle, 2015-04-15:
@lcfm1 did you try as @Natealus suggested?

closed 2015-05-09 by thinkyhead