MarlinFirmware/Marlin #1834: crazy idea: use hotend as z-probe

lcfm1, 2015-04-06:
I have a printer  Kossel by folgertech.

top of heating table - aluminum, the top is not electrically connected to the frame. if i connect one wire sensor to the table, and the second to hotend - can i get a z-probe sensor? i may have to use optocoupler to prevent burnout Arduino.

Another question: Z_PROBE_OFFSET_FROM_EXTRUDER = 0.2 does not specify (

boelle, 2015-04-06:
Not as crazy as you think

CNC mill's use this quite a lot... some have one aligator clip on the spindle and then a 2cm cube... when it hits the top of the block it knows its excatly 2 cm from the the item  to be milled

same for X and Y axis.... normal you leave this block bolted down in the corner and only come near it if you ask for it

For a 3D printer a bit of PCB could be used as contact plate

lcfm1, 2015-04-06:
boelle, I think about when hotend touch the table, could start heating,  thereby possibly burn the input of Arduino

AnHardt, 2015-04-06:
You could have a tiny little drop of plastic at the tip of your nozzle - what prevents conduction.

Ziggy2013, 2015-04-06:
Using the extruder as the probe is already being done - example Robo3d printer

lcfm1, 2015-04-06:
AnHardt, if I put plastic on the tip - How can I get the alarm input, when the nozzle touches the bed?) Then there is no electrical contact.

lcfm1, 2015-04-06:
Ziggy2013, in Robo3d almost all video what i found - people add trigger for z-level

AnHardt, 2015-04-06:
@lcfm1 
I don't now what your nozzle is doing. Mine is extruding plastic. Normally it's not clean.

So - yes, a crazy idea.

daid, 2015-04-06:
Won't work reliably, I did a lot of testing for this to possibly include this as leveling method in the Ultimaker2.

Ziggy2013, 2015-04-06:
@lcfm1 These must be old videos. For over a year the Robo3d has had auto level using the extruder and z nut switches.

Anyhow the point was using the extruder as a probe is not a new idea at all.  It is also one of the reasons the Z Probe offset value can be positive or negative.

thinkyhead, 2015-04-07:
All true. If the hot-end isn't hot, there's no guarantee there won't be a 2mm blob of plastic on the tip. I've seen a Prusa i3 that used a contact switch and a floating X gantry, so if the Z moves down when the hot-end is touching the bed, the gantry stays put and a contact switch is opened. But, it only works when the hot-end is completely clean.

scotty1024, 2015-04-07:
Lulzbot TAZ Mini uses tip of all metal RRD Hexagon hot end and four very wide washers at the corners of the bed. They also have a nozzle cleaner built in. You can find several videos of it's auto leveling in action on Youtube.

thinkyhead, 2015-04-07:
Oh yes, I also saw that @josefprusa was experimenting with little pushbutton switches at the bed corners. https://www.youtube.com/watch?v=QTpYXkOLWcQ

Grogyan, 2015-04-07:
Could to it capacitively.
Assuming a heated bed is used

scotty1024, 2015-04-07:
Direct electrical contact.

http://youtu.be/OWmOqBw5QG4

TechMasterJoe, 2015-04-07:
i might test out what luz bot is doing going to need to disable check routine to do it my print head can each about 15mm past the bed on all sides i will make the head cleaner for sure

thinkyhead, 2015-04-07:
@TechMasterJoe You shouldn't need to disable checks for movement limits, if that is what you're referring to. The `[XYZ]_MIN_POS` and `[XYZ]_MAX_POS` settings can be set beyond the bed, wherever your robot can reach. They represent the physical limits of the machine, w/r/t the endstop positions that you have set.

TechMasterJoe, 2015-04-08:
i agree but i offset home = 300 X 300mm my extended Y is 580mm so beds a long ways from my end stops.so i use home offsets but only on y my x is around 330mm or so and that endstop is lined up with the glass for home and i have marlin max soft limit enabled at 300mm

AnHardt, 2015-04-09:
Is this a firmware problem at all?

boelle, 2015-04-26:
This is more a hardware feature than firmware.... furthermore the platic on the hotend will isolate so it does give bad results.... so will be closing this one.

closed 2015-04-26 by boelle