MarlinFirmware/Marlin #1835: Auto Z_PROBE_OFFSET_FROM_EXTRUDER

lcfm1, 2015-04-06:
You can do this?

Theoretically, if I correctly input MANUAL_Z_HOME_POS, and manually expose position z-probe relative to hotend spending several measurements where the sensor will be triggered, i can get Z_PROBE_OFFSET

thinkyhead, 2015-04-07:
If you also have a Z endstop which is accurately set, then yes, you could automatically determine the distance from the Z probe to the Z position of the hot-end. It will also depend on having some point on the bed which you know to be at a good height.

jbrazio, 2016-03-10:
@thinkyhead Why is this marked as a [open] feature request ? This seems a support question.

thinkyhead, 2016-03-16:
@jbrazio My interpretation of this issue is that @lcfm1 is asking for us to write code implementing a new command which could automate the process of determining the distance from the nozzle tip to the probe. But in practical terms I don't think it's possible. It's also unnecessary, since a probe is expected to be at a fixed position, not movable.

closed 2016-03-16 by thinkyhead