MarlinFirmware/Marlin #1838: #define MESH_BED_LEVELING error

lcfm1, 2015-04-06:
Marlin_main.cpp: In function 'void process_commands()':
Marlin_main.cpp:4745:21: error: 'gcode_G30' was not declared in this scope
           gcode_G30();

epatel, 2015-04-06:
Did you also enable ENABLE_AUTO_BED_LEVELING? Both can't be enabled at the same time, currently.

lcfm1, 2015-04-06:
epatel, nothing about it says in the instructions https://github.com/epatel/Marlin/commit/4a78db29eb15d599f22c99d66cf79d1427061c64

epatel, 2015-04-06:
Line 10?

epatel, 2015-04-06:
Has been added to this PR https://github.com/MarlinFirmware/Marlin/pull/1774

lcfm1, 2015-04-06:
http://pastebin.com/fWTFBSET

when I select on-screen level beds, first g29, then there is nothing on the screen, the controller on the screen printer responds knock engine because endstops not work correctly.

epatel, 2015-04-06:
Yes, I tried earlier today and it seems like latest changes somewhere makes the display work differently. One need to press encoder button once for it to travel to first position.

I don't know what *"screen printer responds knock engine because endstops not work correctly"* mean.

lcfm1, 2015-04-06:
sorry for my english)

ok, i tried press encoder, on screen - z: and 0.00
then rotate the encoder, get gnashing engines.

epatel, 2015-04-06:
Z: 0.00? You defined `MESH_HOME_SEARCH_Z` to 0? should work mathematical, I have not tested though.

But you very close to the bed for this. I have designed it to work with Z endstop above the bed. Below the bed would be bad as the "endstop level" will be used for traveling between points.

epatel, 2015-04-06:
I will try to look into why the display works differently. Strange. Can't at the moment, doing a long running print >15 hours. :)

epatel, 2015-04-06:
Delta printer? oh, have no idea how that works, or if it even could with this mesh solution. Maybe @alexborro can give some pointers?

Wish I had one to test with though. Wife would complain I think...getting to much stuff in my man-cave ;)

lcfm1, 2015-04-06:
epatel, z:4.00

when i tried press encoder, it doesn't go to the first point, it making gnashing engines.

lcfm1, 2015-04-06:
if I press for the first time and in the future - it does not go down, he tries to go up for the endstops

epatel, 2015-04-06:
Well, I think this is something special with delta printers. I wish I could debug it. Will look at the delta code and see if it is something obvious, otherwise I think I need to get a delta printer to fix. Sorry.

lcfm1, 2015-04-06:
epatel, mark this issue as error

epatel, 2015-04-06:
Seems like I am not allowed to mark/label issue. But I added #error to PR mentioned above https://github.com/epatel/Marlin/commit/d27c82a4500229abdf9a3d478b981f0315cfee21

thinkyhead, 2015-04-07:
What is the feedrate that is used for these moves? Always make sure to have a reasonable Max Feedrate set for all your axes, then gnashing shouldn't happen with _normal_ movement.

lcfm1, 2015-04-07:
thinkyhead, http://pastebin.com/fWTFBSET

thinkyhead, 2015-04-07:
@lcfm1 Hey max, I'm not entirely sure, but it sounds almost like it homes but then it tries to raise the axis

 but are seeing:

- Homes then afterwards the motors try to move up
or
- As each carriage reaches the top it just continues to try to move up
?

lcfm1, 2015-04-07:
thinkyhead, after g29, they homes, endstops triggered and motors try to move up

lcfm1, 2015-04-07:
even if I manually in pronterface droop the carriage down at z = 0.1 and then push encoder on the screen, he tries to climb up. One gets the impression that the function of the z coordinate is inverted and not interrupt trigger stops

thinkyhead, 2015-04-07:
As an experiment, change your `Z_RAISE_BEFORE_HOMING` to 0 (which I think a max Z endstop should override anyway, but maybe not). That will probably have no effect. Then try changing your `Z_RAISE_AFTER_PROBING` to 0. And finally, try changing the `Z_RAISE_BEFORE_PROBING` to 0. See how each one changes the behavior. When homing to Z max (as a delta _almost_ must) the delta should not try to do any raises _after_ probing, except to retract a servo probe. And actually, I don't see why a delta would home automatically at the end of a `G29`. It should stay where it is, because homing disables the bed-leveling grid.

lcfm1, 2015-04-07:
Scott, it's can be similar like https://github.com/MarlinFirmware/Marlin/issues/1166
I can check it out only in the evening. 24:00 (GMT)

thinkyhead, 2015-04-07:
@lcfm1 I look forward to your results. We are trying to isolate the root of this behavior. Be sure to use the latest code.

lcfm1, 2015-04-07:
Z_RAISE_BEFORE_HOMING 0
after click on encoder - g28, then motors try to move up 2 seconds 

Z_RAISE_AFTER_PROBING
after click on encoder - g28, then motors try to move up 2 seconds

#define Z_RAISE_BEFORE_PROBING
after click on encoder - g28, then motors try to move up 2 seconds

set all to 0 - same

after first click on encoder(move to first probe) i send m114:
SENDING:M114
X:10.00 Y:10.00 Z:4.05 E:0.00 Count X: 9.99 Y:9.99 Z:4.01

but if i send g29 and after m114:
SENDING:M114
X:0.00 Y:0.00 Z:264.50 E:0.00 Count X: 446.22 Y:446.22 Z:446.22

thinkyhead, 2015-04-08:
Ok, I've posted the latest code with bed leveling and homing debugging messages added. To help debug the leveling and probing behavior, get the code at this branch:

https://github.com/thinkyhead/Marlin/archive/more_configurator.zip

Find the option at the top of `Marlin_main.cpp` named `DEBUG_LEVELING` and enable it. Then run homing and probing, if possible, and you will see output to the serial console. Paste the output from the serial console on Issue #1844.

lcfm1, 2015-04-08:
@thinkyhead, i am at work now. i live in (GMT+6) zone.

Before i go to work, I found the following:
The procedure is as follows:
1. carriage go home
2. Go to reset the coordinates x: 10 y:10 z: 0
3. After move to z:5

then found the next thing:
when turning the encoder is not moving the z-coordinate, and is moving towers z

try to looking for that in code. evening i try to make log.

thinkyhead, 2015-04-08:
@lcfm1 Just looking over the movement code for the LCD. There are 2 ways to move axes: Babystep or Move. The code for "Move" seems to have the delta transformation in place, as far as I can see.

epatel, 2015-04-08:
If HOMING is away from the bed I think it should be a minor fix, I think I could look at that (easy testable here). But I don't have any knowledge about DELTA movements...

lcfm1, 2015-04-08:
@epatel, you use change coordinates or g z1?

lcfm1, 2015-04-08:
@thinkyhead, can't compile. error:

temperature.cpp.o: In function `_temp_error(int, char const*, char const*)':
C:\Users\LCF\AppData\Local\Temp\build7512567997942369460.tmp/temperature.cpp:446: undefined reference to `IsStopped()'

lcfm1, 2015-04-08:
@thinkyhead, too many errors during compiling
my files:
configuration.h http://pastebin.com/ans3MeaZ
CONFIGURATION_ADV.H http://pastebin.com/kPhhRHNv

lcfm1, 2015-04-08:
@thinkyhead, can you compile with my settings?

thinkyhead, 2015-04-08:
Welcome to the active Development branch! Things should compile now, or soon…

thinkyhead, 2015-04-08:
Ok, I expect that zip to be better now.

lcfm1, 2015-04-08:
@thinkyhead, errors( http://pastebin.com/W5yChsxX

lcfm1, 2015-04-08:
@thinkyhead errors

http://pastebin.com/W5yChsxX
when i paste conf kossel mini

thinkyhead, 2015-04-08:
@lcfm1 That has been patched. Try the latest code.

lcfm1, 2015-04-08:
@thinkyhead try to compile and recieve errors
http://pastebin.com/vCqAvwMW

thinkyhead, 2015-04-08:
That's not the latest code. Line 1404 is now:

`SERIAL_ECHOPAIR("> ProbeAction:", (unsigned long)probe_action);`

and the code you just posted shows

`SERIAL_ECHOPAIR("> ProbeAction:", probe_action);`

lcfm1, 2015-04-08:
@thinkyhead, think I was able to compile. try after 4 hours

lcfm1, 2015-04-08:
@epatel, @thinkyhead code for mesh_level without debug print

lcfm1, 2015-04-08:
Connecting...
start
Printer is now online.
echo:Marlin1.0.2
echo: Last Updated: Apr  8 2015 19:08:50 | Author: (none, default config)
Compiled: Apr  8 2015
echo: Free Memory: 2827  PlannerBufferBytes: 1232
echo:V19 stored settings retrieved (378 bytes)
echo:Steps per unit:
echo:  M92 X44.44 Y44.44 Z44.44 E100.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z300.00 E300.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z1000 E1000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustement (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=delta_diagonal_rod, R=delta_radius, S=delta_segments_per_second
echo:  M665 L230.00 R141.00 S200.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Filament settings: Disabled
echo:SD init fail
echo:enqueing "G28"
echo:enqueing "G28"
Error:Printer halted. kill() called!

before kill printer move head to z-tower

epatel, 2015-04-08:
Mesh + Delta is not yet supported. Should have received an error just for that with latest I think.

epatel, 2015-04-08:
Also probing more than 7 point per axis is also not supported can also cause more problem. I will add a sanity check for that.

lcfm1, 2015-04-08:
@epatel, look at the code delta calibration - it works well on my printer.

epatel, 2015-04-08:
@lcfm1 code delta calibration, sorry don't know what that is or where to find it? You are using Mesh Bed Leveling + more than 7 points?

lcfm1, 2015-04-08:
@epatel  #define DELTA_CALIBRATION_MENU after display conf in configuration.h 

in ultralcd.cpp
{
    START_MENU();
    MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
    MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_X, PSTR("G0 F8000 X-77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Y, PSTR("G0 F8000 X77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Z, PSTR("G0 F8000 X0 Y90 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_CENTER, PSTR("G0 F8000 X0 Y0 Z0"));
    END_MENU();
}

lcfm1, 2015-04-08:
@epatel, or used 

    #ifdef DELTA
      calculate_delta(current_position);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], current_position[E_AXIS], manual_feedrate[axis]/60, active_extruder);
    #else
      plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], manual_feedrate[axis]/60, active_extruder);
    #endif

closed 2015-04-14 by thinkyhead