MarlinFirmware/Marlin #1840: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER reverse encoder rotation

lcfm1, 2015-04-07:
Switching on the menu goes down by rotation of the wheel to the left

TechMasterJoe, 2015-04-07:
ultralcd.cpp
switch + to - and - to +
look for this at bottom
```
//manage encoder rotation
  uint8_t enc=0;
  if (buttons & EN_A) enc |= B01;
  if (buttons & EN_B) enc |= B10;
  if (enc != lastEncoderBits) {
    switch(enc) {
      case encrot0:
        if (lastEncoderBits==encrot3) encoderDiff++; // change to encoderDiff--
        else if (lastEncoderBits==encrot1) encoderDiff--; // change to encoderDiff++
        break;
      case encrot1:
        if (lastEncoderBits==encrot0) encoderDiff++; // change to encoderDiff--
        else if (lastEncoderBits==encrot2) encoderDiff--;// change to encoderDiff++
        break;
      case encrot2:
        if (lastEncoderBits==encrot1) encoderDiff++; // change to encoderDiff--
        else if (lastEncoderBits==encrot3) encoderDiff--; // change to encoderDiff++
        break;
      case encrot3:
        if (lastEncoderBits==encrot2) encoderDiff++; // change to encoderDiff--
        else if (lastEncoderBits==encrot0) encoderDiff--;// change to encoderDiff++
        break;
    }
```

boelle, 2015-04-07:
closing this one as its not a bug.... but agreed it should default be the other way round... a matter of taste maybe

closed 2015-04-07 by boelle

thinkyhead, 2015-04-08:
The solution proposed by @TechMasterJoe will unfortunately cause edited values to go up when rotating the knob counter-clockwise, probably not what you want. A more proper solution has been discussed, and will be implemented at some point soon.

lcfm1, 2015-04-08:
@thinkyhead, it will need to open?