MarlinFirmware/Marlin #1841: Feature request: Save printer settings on sd card

lcfm1, 2015-04-07:
It would be good to printer settings saved on the memory card every time you run Auto Levels and calibration. In addition to the firmware had the opportunity to include this file in order to flashing every time you do not fill in the configuration file and not to Auto level

thinkyhead, 2015-04-07:
It's also been proposed to move to .cnf or .ini style files similiar to Repetier Host, which are just written and read literally, and which are also verbatim as the EEPROM settings. As long as it fits within the smallest EEPROMs.

lcfm1, 2015-04-07:
or dumping eeprom for test or something else

thinkyhead, 2015-04-07:
`M503 S0` is one of the more obscure features of Marlin.

TechMasterJoe, 2015-04-07:
not really i cross check my 501 and 503 every firmware upload after calling 502>500>reconnect>503
this is how i know for SURE all my old data was over written

boelle, 2015-04-07:
its been asked for here also: https://github.com/MarlinFirmware/Marlin/issues/1185

Roxy-3DPrintBoard, 2015-04-07:
I suggest we have a *BIG* discussion on this topic after the current bug fix phase.  The reason being this:   I don't think we want to go add different approaches in a Willy Nilly manner.  There are going to be 37 different ideas to address the issues.   How about we actually understand the real issues and then solve the problem just once?

I do see the value of saving the configuration information in different locations.   But how about saving the information in the second 1/2 of the EEPROM?  There might be room to have the first configuration stored in the first 1/2 of the EEPROM and a 2nd configuration stored in the 2nd 1/2 of the EEPROM.

Any way...  I'm really not trying to be difficult.   Instead...   I'm hoping for a well thought out discussion that leads to an elegant answer.

thinkyhead, 2015-04-08:
The `S0` argument to `M503` is one of the more obscure features of Marlin.

closed 2016-05-04 by jbrazio