MarlinFirmware/Marlin #1844: Autoleveling is a complete mess...

fotis400, 2015-04-07:
Just trying to make my printer work again after some mods (mechanical). I pulled some of the latest commits, and now I am trying to print.

My problem now is with autoleveling, G29. It performs completely wrong and unexpectedly...
G28 works correct.
G29 works again correct when send manually, after the G28.

Then I try to print the following file.
1. Heatbed warms up, all the axis home correctly.
2. Autolevel starts and gets to the first point correctly.
3. Same for the second and third point in the same row (I probe in 9x9 grid).
**4. After that it starts moving all the axis towards their homing direction VERY slowly. It also extrudes simultaneously with the same slow speed.**
5. When the first endstop triggers, it gets to the next normal point. It probes the next 3 points (the second row) normally.
**6. At the end of the second row again it starts moving towards homing, very slowly, again extruding...**
**7. After an endstop getting triggered, it goes straight down, like it should do for probing the next 3 points, but this time, X & Y are NOT moving at all**

After this kind of calibration, you understand that my printer has a very weird idea about reality, and printing is impossible. 

The file I try to print:
http://pastebin.com/NnQPkZqE

My config:
http://pastebin.com/ujBp4vq1

boelle, 2015-04-07:
@thinkyhead is working on it....

thinkyhead, 2015-04-08:
Get the code at this branch:

https://github.com/thinkyhead/Marlin/tree/more_configurator

Find the option at the top of `Marlin_main.cpp` named `DEBUG_LEVELING` and enable it. Then run probing, if possible, and you will see output to the serial console. Paste the output from the serial console here, to help debug the probing behavior.

lcfm1, 2015-04-08:
Compiled: Apr  8 2015
echo: Free Memory: 2499  PlannerBufferBytes: 1232
echo:V19 stored settings retrieved (378 bytes)
echo:Steps per unit:
echo:  M92 X44.44 Y44.44 Z44.44 E100.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z300.00 E300.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z1000 E1000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustement (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=delta_diagonal_rod, R=delta_radius, S=delta_segments_per_second
echo:  M665 L230.00 R141.00 S200.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Filament settings: Disabled
echo:Z-Probe Offset (mm):
echo:  M851 Z-3.50
echo:SD init fail
>>> g29
SENDING:G29
echo:Home X/Y before Z
>>> g29
SENDING:G29
G29 Auto Bed Leveling
9.00 4.57 0.29 -4.74 -12.70 -15.97 -15.99 -16.00 -16.01
0.18 -1.67 -6.02 -10.38 -14.35 -15.98 -16.00 -16.01 -16.03
-10.02 -7.91 -12.33 -16.01 -16.00 -15.99 -16.00 -16.02 -16.02
-13.00 -14.50 -16.00 -15.99 -16.00 -16.01 -16.00 -16.01 -16.00
-15.98 -15.99 -16.01 -16.00 -16.01 -16.00 -16.01 -16.00 -16.01
-15.99 -16.00 -16.02 -16.01 -16.00 -16.01 -16.00 -15.99 -16.00
-16.00 -15.99 -15.98 -15.99 -16.01 -16.00 -16.01 -16.00 -15.99
-15.97 -15.98 -16.01 -16.00 -15.99 -16.00 -15.99 -16.01 -16.00
-15.98 -16.02 -16.01 -16.00 -15.99 -16.00 -16.00 -16.00 -16.01
Disconnected.

fotis400, 2015-04-08:
Very interesting... And strange....

I sent a G28. All normal. I sent a G29 all normal.
Here is my output: http://pastebin.com/bzq1DKPP

I tried to print the above file which in fact starts with G28 and G29! Read the output. You will laugh (or cry...)

http://pastebin.com/PnfgRkXH

(It must be noted that all the times the probe worked, after probing it went up "in two steps" as in the above mentioned issue.)

thinkyhead, 2015-04-09:
Ok, so first thing is, at the beginning of `G28` it doesn't need to call `deploy_z_probe`. It should only do that if it's using the Z min endstop as the Z endstop. For a machine with a Z max endstop, it should skip this step.

lcfm1, 2015-04-09:
@thinkyhead the printer does not even arrives to the moment when the trigger should operate. he just tries air and goes to the next point. may have to verify that he has triggered, and only after proceed to the next point

lcfm1, 2015-04-09:
@thinkyhead please insert in these functions more debug print -  I can run this code on a printer for detailed debugging

thinkyhead, 2015-05-09:
Let's switch over to #2040 where the problem is being narrowed down.

closed 2015-05-09 by thinkyhead