MarlinFirmware/Marlin #1846: Cold extrusion prevention not working

fotis400, 2015-04-07:
Just tested it multiple times.

With my extruder cold, I send a command to extrude. The printer echoes that it prevented a cold extrude but it DOES extrude.

boelle, 2015-04-07:
Not good... @thinkyhead you think you can spot why this one is happening?

clefranc, 2015-04-07:
Potential bug confirmed with https://github.com/MarlinFirmware/Marlin/pull/1847

clefranc, 2015-04-08:
It's the new e delta calculation https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L508

You need to recalculate the delta after the `PREVENT_DANGEROUS_EXTRUDE` block https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L510

I've put this on line 526

`  de = target[E_AXIS] - position[E_AXIS];`

thinkyhead, 2015-04-08:
Fixed the delta e issue (in the above-linked commit).

closed 2015-04-08 by thinkyhead

fotis400, 2015-04-08:
tested and works ok.