MarlinFirmware/Marlin #1848: MSG_LEVEL_BED i didn't find in all language files, except en

lcfm1, 2015-04-07:


thinkyhead, 2015-04-07:
Feel free to contribute translations! English is the fallback for strings not yet translated.

lcfm1, 2015-04-07:
@thinkyhead, Who added all string in another language files?

AnHardt, 2015-04-07:
@lcfm1 
git blame gives the answer.

thinkyhead, 2015-04-08:
@lcfm1 Language contributions arrive whenever those who know the language make a PR and submit it. There are thousands of forks of Marlin, most not very dynamic, but there are a few very active contributors. Sometimes only one contribution from someone and then never heard from again.

boelle, 2015-04-08:
i will close this one as it not an issue with marlin and that people who wants support for something should submit an PR instead.....

closed 2015-04-08 by boelle