MarlinFirmware/Marlin #1851: Errors on delta printers: move coordinates from menu

lcfm1, 2015-04-08:
1. When I try to move the carriage from the menu on the home position, I start knocking engines. This happens due to the fact that to move in the x or y coordinate, it is necessary to move all three bogeys on the towers, but this does not happen, due to the fact that the carriage in the home position.

2. have not found moving along the axis z 10mm

solutions:

1. prohibit the movement of the carriage from the menu on the x or y until there is not enough space for carriages for this maneuver.

thinkyhead, 2015-04-08:
This overlaps with a general request elsewhere to restrict movement of the carriages at the towers, so that coordinates outside the physical limits will not be attempted, only the nearest-possible coordinate. I posted a sample implementation (not 100% correct) in that thread. [search]

lcfm1, 2015-04-08:
@thinkyhead, https://github.com/MarlinFirmware/Marlin/issues/1856#issuecomment-90795982

lcfm1, 2015-04-08:
@thinkyhead, you can calculate a triangle of carriage positions. And check for excess z_home_pos for delta printers

[image]

thinkyhead, 2015-04-08:
Refer to this code for the point where it would be simplest apply the limits, based on current Z position (of the nozzle). https://github.com/MarlinFirmware/Marlin/issues/928#issuecomment-88059303

boelle, 2015-04-08:
Is this really an firmware error?

thinkyhead, 2015-04-08:
By rights, the delta is really not ready to be moved until the head has returned to the bed. But, assuming the arms are low enough, XY movement should be permitted within the physical limits of the printer. The top-most usable region of the delta then becomes conical.

thinkyhead, 2015-04-14:
Two places the Repetier code checks for out-of-bounds conditions:

Starting at [`motion.cpp` line 1245](https://github.com/mrbi11/Repetier-Firmware/blob/work092/src/ArduinoAVR/Repetier/motion.cpp#L1245) the carriages are prevented from moving above the endstops, while the [`transformCartesianStepsToDeltaSteps` function at line 885](https://github.com/mrbi11/Repetier-Firmware/blob/work092/src/ArduinoAVR/Repetier/motion.cpp#L885) will return 0 in any case where the printer tries to move out of bounds. So these are the bits of code we would need to glean from.

closed 2016-05-04 by jbrazio