MarlinFirmware/Marlin #1854: Error: from menu when i press AUTO HOME and go up to level i see 2 Settings menu

lcfm1, 2015-04-08:
Steps:

press AUTO HOME
press go up level and it's look like:

Home
Settings
Settings
Clear

PS (I use Russian and so the translation of menu items may differ)

boelle, 2015-04-08:
Can anyone else verify this one?

thinkyhead, 2015-04-08:
```
Обзор
Настройки
Настройки
???
```
?

lcfm1, 2015-04-08:
@thinkyhead, yes

clefranc, 2015-04-08:
@boelle I confirm!
When doing an auto home from the LCD, or a G28 from host, the Prepare menu is replaced by the Tune menu while the homing sequence completes. After homing completion, if I turn the LCD knob, the Tune menu is marlingically replaced by the Prepare menu. The printer think it's printing when doing G28.

I'm looking at the code right now.

clefranc, 2015-04-08:
@thinkyhead Easy one:

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L387

Perhaps a new global variable like IS_PRINTING when moves are done with the LCD or host.

thinkyhead, 2015-04-09:
Hmm, so what do you think…

    if (movesplanned() > 1 || IS_SD_PRINTING) { . . .

…?

thinkyhead, 2015-05-18:
Is this still a problem? I also noticed that sometimes I have to do an extra click on the LCD. I'd like to try to get both at once, sometime soon.

thinkyhead, 2015-10-12:
This should be fixed in the latest RC (RC2). Please check and if the bug is fixed, close this issue.

closed 2015-11-06 by AnHardt