MarlinFirmware/Marlin #1856: Error: delta printer endstops not working

lcfm1, 2015-04-08:
when I try to change the coordinates from menu near homepos - it causes crackling engines

it also relates to 
https://github.com/MarlinFirmware/Marlin/issues/1838
https://github.com/MarlinFirmware/Marlin/issues/1851

lcfm1, 2015-04-08:
Relative movement of the carriages hotend like spheres intersecting towers - accordingly it is possible to calculate and to prohibit the carriage to move in order to avoid engine damage or belts

clefranc, 2015-04-08:
Somewhat related https://github.com/MarlinFirmware/Marlin/issues/1360

thinkyhead, 2015-04-14:
Two places the Repetier code checks for out-of-bounds conditions:

Starting at [`motion.cpp` line 1245](https://github.com/mrbi11/Repetier-Firmware/blob/work092/src/ArduinoAVR/Repetier/motion.cpp#L1245) the carriages are prevented from moving above the endstops, while the [`transformCartesianStepsToDeltaSteps` function at line 885](https://github.com/mrbi11/Repetier-Firmware/blob/work092/src/ArduinoAVR/Repetier/motion.cpp#L885) will return 0 in any case where the printer tries to move out of bounds. So these are the bits of code we would need to glean from.

thinkyhead, 2015-05-09:
I should get my delta working soon so I can start fixing this stuff…

closed 2016-04-23 by jbrazio