MarlinFirmware/Marlin #1858: Error: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER After pressing Stop, I can not do anything. how to proceed?

lcfm1, 2015-04-08:
Encoder responds sound when pressed, but nothing happens

SkyFlyer2, 2015-04-08:
Right, because this is the "STOP" button. As it should be. Then simply press reset on the main controller.

lcfm1, 2015-04-08:
@SkyFlyer2, reset button under printer. what i should do?

Then there must be some way to continue working with the buttons on the screen

AnHardt, 2015-04-08:
@lcfm1 
If you can't reach the reset button, try the power switch. If you have no power switch - pull the wall plug.

Did you hit the stop button by accident?

lcfm1, 2015-04-08:
no, i just try)

boelle, 2015-04-08:
This looks like @lcfm1 is just confused about his printer

Is this really a bug in marlin ????

thinkyhead, 2015-04-08:
Should be documented as: Full Stop requiring hardware reset. Used for most serious fault conditions.

closed 2015-04-08 by thinkyhead

boelle, 2015-04-08:
btw... if wanting to test hangout etc. i'm there rest of the day.... cam turned off thou

thinkyhead, 2015-04-08:
@boelle I'm off to bed soon, but next time…!

boelle, 2015-04-08:
np....