MarlinFirmware/Marlin #1859: Cold extrusion misbehavior

clefranc, 2015-04-08:
Doing more test from https://github.com/MarlinFirmware/Marlin/issues/1846, I got this when trying a 1mm cold extrusion:

`00:25:26.433 : N16 G91 *6
00:25:26.433 : N17 G1 E1 F600 *58
00:25:26.433 : N18 G90 *9
00:25:26.439 : echo: cold extrusion prevented
00:25:26.442 : echo: cold extrusion prevented
00:25:26.444 : echo: cold extrusion prevented
00:25:26.445 : echo: cold extrusion prevented
00:25:26.447 : echo: cold extrusion prevented
00:25:26.449 : echo: cold extrusion prevented
00:25:26.451 : echo: cold extrusion prevented
00:25:26.452 : echo: cold extrusion prevented
00:25:26.454 : echo: cold extrusion prevented
00:25:26.456 : echo: cold extrusion prevented
00:25:26.458 : echo: cold extrusion prevented
00:25:26.460 : echo: cold extrusion prevented
00:25:26.461 : echo: cold extrusion prevented
00:25:26.463 : echo: cold extrusion prevented
00:25:26.465 : echo: cold extrusion prevented
00:25:26.467 : echo: cold extrusion prevented
00:25:26.468 : echo: cold extrusion prevented
00:25:26.470 : echo: cold extrusion prevented
00:25:26.472 : echo: cold extrusion prevented
00:25:26.474 : echo: cold extrusion prevented
00:25:26.476 : echo:endstops hit:  X:0.01 Y:0.01 Z:0.01`

The odd part is the endstops hit at the end ???

On the second 1mm attempt, the last endstops line is gone !!! So perhaps it's normal behavior for endstops, but the multiple prevented messages is definitively not.

thinkyhead, 2015-04-08:
What happens if you do this…? Maybe `de` is _very close_ to zero but not zero. It's a stretch, I know.

    #ifdef PREVENT_DANGEROUS_EXTRUDE
      if (de < -0.0001 || de > 0.0001) { ...

thinkyhead, 2015-04-08:
The original code looked like this https://github.com/MarlinFirmware/Marlin/commit/13fbf42d95251eb3ccc36449a8056de8cd8d228a#diff-37f48ac4f0d16c80cfe24c1d399f6340L565

clefranc, 2015-04-08:
@thinkyhead Same result with 0.001 tolerance.

With the host sending a single G1 command I get:

1mm = 10 messages
10mm = 100 messages
100mm = 1000 messages

I doubled my  E steps/unit, same result.

clefranc, 2015-04-08:
@thinkyhead 
I should have looked at my i3, this is happening only on delta, here is the loop calling `plan_buffer_line` multiple time:

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L5503

Personnaly, I'll move the cold extrusion prevention code from the planner to the `prepare_move()` function. I think it's its main functionality, even the prepare move should be in planner.cpp.

I'm sure you'll find an elegant solution, as always...

thinkyhead, 2015-04-09:
How about this in `prepare_move`…?
```
void prepare_move() {
  clamp_to_software_endstops(destination);
  refresh_cmd_timeout();

  #ifdef PREVENT_DANGEROUS_EXTRUDE
    float de = destination[E_AXIS] - current_position[E_AXIS];
    if (de) {
      if (degHotend(active_extruder) < extrude_min_temp) {
        current_position[E_AXIS] = destination[E_AXIS]; // Behave as if the move really took place, but ignore E part
        SERIAL_ECHO_START;
        SERIAL_ECHOLNPGM(MSG_ERR_COLD_EXTRUDE_STOP);
      }
      #ifdef PREVENT_LENGTHY_EXTRUDE
        if (labs(de) > EXTRUDE_MAXLENGTH) {
          current_position[E_AXIS] = destination[E_AXIS]; // Behave as if the move really took place, but ignore E part
          SERIAL_ECHO_START;
          SERIAL_ECHOLNPGM(MSG_ERR_LONG_EXTRUDE_STOP);
        }
      #endif
    }
  #endif
  . . .
```

AnHardt, 2015-04-28:
@clefranc 
Is the error gone? Can this be closed?

thinkyhead, 2015-04-29:
I believe that this was fixed by #1876 which makes a check at a higher level so there will only be one attempt instead of many. At least, I hope so...

boelle, 2015-06-29:
we have not heard from @clefranc in 3 months so closing

closed 2015-06-29 by boelle