MarlinFirmware/Marlin #1860: Error: MSG_KILLED shows  only first 3 letters in ru language

lcfm1, 2015-04-08:


lcfm1, 2015-04-08:
it looks like a problem with utf-8 and variables size

thinkyhead, 2015-04-08:
I'm not sure where it's failing. Maybe it's getting the strlen wrong. According to what I read, it should looks something like this:

    int lcd_strlen_P(const char *s) {
      int j = 0;
      uint8_t c;
      while (c = pgm_read_byte(s)) {
        int len = 0;
        if (c < 0x80) {
          // Just increment below
        }
        else if (c < 0xC2) {
          // Invalid continuation, exit
          break;
        }
        else {
          // Length based on 1st byte
          uint8_t len = c < 0xE0 ? 2 : (c < 0xF0 ? 3 : (c < 0xF5 ? 4 : 1));
          while (--len) {
            c = pgm_read_byte(++s);
            if ((c & 0xC0) != 0x80) break;
          }
        }
        s++;
        j++;
      }
      return j;
    }

AnHardt, 2015-04-08:
@thinkyhead 
I think lcd_strlen() is ok. I will look into the recently changed routine for the status message.

AnHardt, 2015-04-08:
Yesterday fixed with #1839

lcfm1, 2015-04-08:
Confirm. latest build is working properly with this.

But if i setting language to ru and enable auto bed leveling - i get errors utf-8 decode while compiling

AnHardt, 2015-04-09:
@lcfm1 
It would be very kind if you could close your resolved issues - like this.

closed 2015-04-10 by thinkyhead