MarlinFirmware/Marlin #1870: Is volumetric extrusion (M200) broken?

clefranc, 2015-04-08:
I'm using volumetric extrusion since december 2014, was working great, downloaded many dev version and using it without problem. I know the Sclic3r setting and the M200 command.

Since the last few weeks, I've tried to use the volumetric extrusion without success (testing the latest dev). When enabled with M200 command, there is barely material extruded. I manage to extrude like Slic3r do by setting `M200 Dx.x` near 1mm. I'm using 1.75mm filament by de way. Perhaps I'm missing a new parameter?

Can someone confirm that volumetric extrusion is functional (or not)? Thanks

thinkyhead, 2015-04-09:
All the code looks to be in place. I'd be interested to see what kind of results are working versus those that don't. The formula for volumetric multiplier is currently:
```
float calculate_volumetric_multiplier(float diameter) {
  if (!volumetric_enabled || !diameter) return 1.0;
  float d2 = diameter * 0.5;
  return 1.0 / (M_PI * d2 * d2);
}
```
Passing 1.75 gives a result of 0.41575
Passing 3.0 gives 0.14147

Then your slicer is supposed to compensate by using cubic millimeters as the E axis unit instead of linear mm. Is the slicer producing the right numbers too?

AnHardt, 2015-04-09:
There are two very simmilar functions.
    float calculate_volumetric_multiplier() calculates the value.
    void calculate_volumetric_multipliers() is the menu function.

I at least can see the use of the menu function in ConfigurationStore.cpp

@thinkyhead Dou you want to look into that?

thinkyhead, 2015-04-09:
Close. While `calculate_volumetric_multipliers()` _is_ used as a post-callback for menu functions that alter the `filament_diameter` values, it is _also_ used anywhere that the `filament_diameter` values might change, such as after `M200`. It just recalculates all the `volumetric_multiplier` values based on the current `filament_diameter` values.

boelle, 2015-04-15:
should this be upgraded to verified?

clefranc, 2015-04-15:
@boelle I'm closing it, I've used the latest devs without problem. Surely a misconfig on my end.

closed 2015-04-15 by clefranc

boelle, 2015-04-15:
things happen :-D