MarlinFirmware/Marlin #1871: static void _lcd_level_bed_homing() in ultralcd.cpp Do not use code for delta Pinter move

lcfm1, 2015-04-08:
should use the condition:

    #ifdef DELTA
      calculate_delta(current_position);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], current_position[E_AXIS], manual_feedrate[axis]/60, active_extruder);
    #else
      plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], manual_feedrate[axis]/60, active_extruder);
    #endif

thinkyhead, 2015-04-09:
I see that `_lcd_level_bed` also uses `plan_buffer_line` without taking `DELTA` into account. Is that also wrong?

lcfm1, 2015-04-09:
yes. and mesh_level too

thinkyhead, 2015-04-09:
Mesh Leveling is not allowed to be enabled with DELTA.

lcfm1, 2015-04-09:
@thinkyhead, I know about it, I said to the author in another topic.  _lcd_level_bed_homing -  it does not appear

thinkyhead, 2015-04-09:
#1879

thinkyhead, 2015-04-09:
Checking the code, it looks like you must go to __Level Bed__ (Настр. уровня кровати) first. It homes, then goes to the `_lcd_level_bed_homing` menu (but it's not a menu). After moving to some position, it then opens the `_lcd_level_bed` menu (also not a menu). Then when it's done, it goes to `lcd_return_to_status` menu. I'm not sure why these are implemented like this. I would expect them to show a blank screen while running.

epatel, 2015-04-09:
`_lcd_level_bed_homing` is a temporary screen while homing, like a state machine. When ready it goes to the next, `_lcd_level_bed`, which is in use while probing all points. Move to position, allow Z movement and wait for encoder press.

This is how I did it, trying to keep it clean and out of the way of everything else. Maybe there are better ways to implement this but there is a myriad of flags and stuff that I don't know about, so I just tried to be careful. I have no knowledge about how the `DELTA` printers setup works so I disabled it until someone fixed that (and got it tested, I can't test it unless some send me a delta printer)

thinkyhead, 2015-04-10:
@epatel I think it makes sense to add some kind of feature so the screen-to-screen system is more clean and can be used in just the way that your functions use them, but we might need to add a function or two, and wrap it into a simple class. Particularly, it would be nice at each of these screens, if there is some wait happening, to display some translatable text as a standard screen option. They might just call teh status screen display function, with the status line set. Or, they might call a special function to display up to 4 lines of custom text. Stuff to be worked out…!

epatel99, 2015-04-10:
@thinkyhead I did try a little to add a "homing" message in the intermediate state. It actually did fall through when I implemented it, i.e. today one seem to need to press the encoder one time extra in the beginning because something has changed in the menu system. I will look at fixing that when I have time.

thinkyhead, 2015-04-10:
@epatel I think you probably just need to set `lcdDrawUpdate = 2;` to clear the display and cause it to redraw. But, instead of adding a faux `MENU_ITEM` we could add a function `lcd_implementation_textpage` that just draws some text, up to 3 lines, and call it in the same way that `lcd_implementation_status_screen` is called from `lcd_status_screen`.

thinkyhead, 2015-04-10:
@epatel Wish my delta was in working condition. It needs some carriages based around "shower rollers" and a sturdy base, yet to be designed. Built on 15mm OpenBeam extrusions.

lcfm1, 2015-04-11:
@thinkyhead you can makelike that http://www.ebay.com/itm/3x-Kossel-Mini-Linear-Carriages-Replaces-MGN12H-Delrin-Recirculating-/161665515939?pt=LH_DefaultDomain_0&hash=item25a40401a3

closed 2015-04-14 by thinkyhead