MarlinFirmware/Marlin #1872: #define MANUAL_BED_LEVELING does not show the menu _lcd_level_bed_homing

lcfm1, 2015-04-08:
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER it does not show the menu

i think because     
in ultralcd.cpp

in line 1385 :
    #ifdef ULTIPANEL
      if (currentMenu != lcd_status_screen &&
        #if defined(MANUAL_BED_LEVELING)
          currentMenu != _lcd_level_bed && 
          currentMenu != _lcd_level_bed_homing && 
        #endif  // MANUAL_BED_LEVELING
          millis() > timeoutToStatus) {
        lcd_return_to_status();
        lcdDrawUpdate = 2;
      }
    #endif //ULTIPANEL

but i dont declare ULTIPANEL

AnHardt, 2015-04-09:
@lcfm1 
Indirectly you defined ULTIPANEL
in Conditionals.h line 45 to 52.

AnHardt, 2015-04-09:
@lcfm1 
_lcd_level_bed_homing is called in the sequence of lcd_level_bed, what you will find uner the name MSG_LEVEL_BED, when MANUAL_BED_LEVELING is defined, at the bottom of lcd_prepare_menu().

It will home before the actual leveling begins.

The lines 1385++ say: 'Don't return to the status-screen while in this or that menu.'

epatel, 2015-04-09:
Yes, the added thing for `MANUAL_BED_LEVELING` in the lines described above is so the timeout do not interrupt the process of probing the grid.

closed 2015-04-10 by thinkyhead