MarlinFirmware/Marlin #1873: General LCD support does not compile

ghpicard, 2015-04-09:
When trying to compile Marlin with an LCD only panel (16x2), no encoder, no SD Card, which would correspond to only #define ULTRA_LCD not commented, I get the following error:

ultralcd.cpp: In function ‘void lcd_goto_menu(void (*)(), uint32_t, bool)’:
ultralcd.cpp:254: error: ‘encoderPosition’ was not declared in this scope
ultralcd.cpp:255: error: ‘lcd_quick_feedback’ was not declared in this scope

AnHardt, 2015-04-09:
#1877 will fix this.

ghpicard, 2015-04-10:
Compilation is now OK. Tomorrow will upload the code to the printer and check how it works before closing the issue. Thanks!

ghpicard, 2015-04-10:
Just checked it in the printer (Reprap Prusa i3 with an Arduino Mega 2560 and Ramps 1.4) and it works fine.
Bug closed! :)

closed 2015-04-10 by ghpicard