MarlinFirmware/Marlin #1881: Request: level_bed_homing and mesh_level_bed make the tree structure as the rest of the menu items

lcfm1, 2015-04-09:


AnHardt, 2015-04-09:
For what?
The first action you have to do is to tune z at the first probing point. Exactly this will happen. Everything until there is automatic and not useful when not in manual probing the bed. Ok, Homing is useful - but this already has a menu.

thinkyhead, 2015-04-10:
Is this request related to the way that the manual and/or mesh bed leveling use the `menu_goto` functions to invoke functions that aren't really menus? If that is the case, I do wish it was done more elegantly. I don't know of any other code that "abuses" the menu system in that way.

epatel99, 2015-04-10:
@thinkyhead Sorry for "abusing" the menu system, I just didn't want to rewrite it, which would probably been harder to get approved I guess. :)

thinkyhead, 2015-04-10:
@epatel Yeah the menus being called repeatedly is an interesting feature, and maybe that can be handled in a better way too. The whole page-to-page system works and it isn't bad, but as this particular part doesn't need to be a speed-demon, we can afford to make it more class-based. Anyway, I think the menu system just needs to be more explicit, and provide hooks for step-by-step tasks, then uses like this will just make more sense.

closed 2016-05-04 by jbrazio