MarlinFirmware/Marlin #1885: Possible Servo problem, with workaround

c-born, 2015-04-10:
While trying to find the cause of some seemingly random Z-Probe servo motor jitter, I hooked a digital scope up to the servo signal at the motor to check for noise etc.
I was surprised to find that after a move, the signal was occasionally left in the high state, rather than always low as expected.
It appears that what is happening is that with a PROBE_SERVO_DEACTIVATION_DELAY set to the default 300mS, and the pulse period being 20mS, depending on the pulse length and small random timing variations the servo can be detached while the signal is high, and it remains in this state.
I don't know if that is a bug in the servo library or by design, not having delved into it in any depth, perhaps it is supposed to be turned off before being detached?

In any case, when I changed PROBE_SERVO_DEACTIVATION_DELAY to 310mS, putting the turn-off midway between the periodic pulses when the output will always be off, the signal was always left in the off state after a pulse train.

I don't know if this warrants a look at the servo library or its use, or just a change to the default off delay value and a comment in the code.

SkyFlyer2, 2015-04-10:
I set delay to 500ms, with this jittering are not present.

Roxy-3DPrintBoard, 2015-04-10:
Do the RC Servo's care if the signal to them is high?   Is that not OK ?

Roxy-3DPrintBoard, 2015-04-10:
This servo twitching issue has been going on for a long long time and caused a lot of support issues.   If it can be resolved by just making the control pin going to the servo be zero when the servo is deactivated, we should call it a bug and fix it.    

But is this really an issue?  Can it be fixed by setting the control line to zero?

thinkyhead, 2015-04-10:
It would be nice to have some reliable way to ensure the servo is off in all cases. If setting the control pin to LOW works, then we should add that.

Ziggy2013, 2015-04-12:
Please review how the arduino servo library actually works before trying to manipulate controls pins. The servo library detach() function is meant to close the servo down.

As the standard servo library uses timer interrupts, the servo needs to be given enough time to move and settle before the detach() is executed.

thinkyhead, 2015-04-12:
So, really just a matter of picking a working `PROBE_SERVO_DEACTIVATION_DELAY` value?

Roxy-3DPrintBoard, 2015-04-12:
> So, really just a matter of picking a working PROBE_SERVO_DEACTIVATION_DELAY value?

I think it is more than that.  I've only spent 10 or 15 minutes looking at it.  But the detach() code pulled from Servo.cpp is below.  If there is only one servo on the timer, it will detach without updating the pins.

>void Servo::detach()
>{
>  servos[this->servoIndex].Pin.isActive = false;
>  timer16_Sequence_t timer = SERVO_INDEX_TO_TIMER(servoIndex);
>  if(isTimerActive(timer) == false) {
>    finISR(timer);
> }
>

Ziggy2013, 2015-04-12:
There is some debate about the pin handling in detach().  It's worth a read

http://forum.arduino.cc/index.php?topic=312869.30

Roxy-3DPrintBoard, 2015-04-12:
Wow!  That is a monster thread!   My vote is we do two things.   First, we wait at the start of the detach() function until the pin is low before we continue.   And then, we write a zero to the pin (just to be safe) before we leave the detach() function.   

The downside is we would burn a few milliseconds of processing time every time detach() is called, but two good things happen.  First, the pulse train controlling the servo is 'regular' with no short pulses.  The second good thing is the control signal to the servo is always zero if we are not trying to position it.  That will be much more consistent behavior from a debugging perspective.

c-born, 2015-04-12:
@Ziggy2013 Crikey, I just read that thread you linked to, and I was amazed to read so many childish and immature responses coming from moderators and seniors in a group! Some appear to have no knowledge of the actual operation of a servo, but don't let that stop them from making strong but incorrect assertions about the code to control them. Other comments were right on the mark and contained code for a working fix, but were ignored. 

I think we can conclude that there is a "bug" or inconsistency in the servo library code which can lead to erratic behaviour in some servos, depending on their specific implementation in an area that is not specified or standardized.

For our immediate concern this can be worked around by making sure that the PROBE_SERVO_DEACTIVATION_DELAY  is not an integer multiple of the 20mS servo pulse period, to ensure we don't try and detach() in the middle of an active pulse.

I suggest adding a comment to this effect in the code, and changing the default value given in the code to say 290mS or 310mS rather than 300mS.

thinkyhead, 2015-04-13:
@c-born We could even add a little test code to `SanityCheck.h` to address the issue...
```
#if PROBE_SERVO_DEACTIVATION_DELAY > 0
  #define PROBE_20 (PROBE_SERVO_DEACTIVATION_DELAY % 20)
  #if PROBE__20 < 5 || PROBE_20 > 15
    #error PROBE_SERVO_DEACTIVATION_DELAY needs to be set farther from a multiple of 20ms
  #endif
  #undef PROBE_20
#endif
```

Ziggy2013, 2015-04-13:
I 100% agree that many of the comments in that Arduino thread were just plain immature.

IMO I think there are in fact two servo issues which potentially impact Marlin users

- If 
#define PROBE_SERVO_DEACTIVATION_DELAY 300  

is uncommented then the time allowed MUST be greater than the time required for the servo to move and settle. Otherwise the detach() could leave then servo in an undefined state.

- Second

The servo must always be given enough time to move and settle BEFORE the Z endstop is tested. If the PROBE_SERVO_DEACTIVATION_DELAY  is defined, then that will achieve the result required. Otherwise there should be an explicit delay() after the servo.write() which allows the servo to settle.

Take these two issues together and the desired result can be achieved if there is ALWAYS enough delay after the servo.write() is executed for the servo to settle. In other words the delay() after the servo.write() is not optional.

c-born, 2015-04-13:
@thinkyhead Your SanityCheck code should work to warn of a potential problem. I think some explanatory notes should also be included to help the user to understand the problem. There is no reason the pulse period has to be 20mS, and some future library change could alter it (or it may already be externally configurable, I haven't looked at the library code). In that case the fault could recur, and adding notes now would save someone having to work through it all again in the future.

c-born, 2015-04-13:
@Ziggy2013 Some background on different servo implementations could have helped people understand why different problems appear on different hardware, and proposed fixes may help in one case and not in another. I'm no expert on these things, but  as I understand it there are a couple of basic servos types.

1. The original servos used a simple monostable (or one-shot) with time set by a potentiometer connected to the servo shaft. With the pot. at midpoint the on-time was 1.5mS, with a typical range of 1mS-2mS. It was edge triggered on the input, with the difference between the input pulse and the monostable pulse being used as an error signal to drive the motor forwards or backwards. Integrating this signal meant that the correction would be stronger with a greater error, and with a faster pulse-train, giving a PI control loop.
Note that this system can be quite insensitive to noise, or variations in the period of a single input pulse, as correction is due to the sum of the errors in the repeated pulse stream.

2. Newer "digital" servos have an inbuilt uC which reads the input pulse width and then internally drives the motor/potentiometer to the required position. It can do this based on a single pulse, and the internal gain can be set for optimal control, regardless of incoming pulse rate. However this can leave them open to noise problems, and a single short or long pulse at the end of a train can result in a wildly inaccurate position being set.

With no defined standard, manufacturers have to set their own, so some may treat a constant high signal differently to others, pulse widths may be internally limited to different values, etc.

With these things in mind, trying to build a "one-size-fits-all" servo library becomes something of a challenge.

Roxy-3DPrintBoard, 2015-04-13:
> With these things in mind, trying to build a "one-size-fits-all" servo library becomes something of a challenge.

Yes, but right now there are race conditions and indeterminate behavior.  If we add one line of code to wait for the control signal to be LOW before detaching the pulse train will not have an abnormal, shorter pulse at the end.  And the control signal will always be parked in the LOW state. 

And just to be extra careful, it wouldn't hurt to set the control signal to a LOW state once again before we leave the detach() function.  Almost for sure this would be unnecessary, but it would force consistent behavior.

Roxy-3DPrintBoard, 2015-04-13:
> We could even add a little test code to SanityCheck.h to address the issue...
>
>#if PROBE_SERVO_DEACTIVATION_DELAY > 0
>  #define PROBE_20 (PROBE_SERVO_DEACTIVATION_DELAY % 20)
> ...

I suspect this is going to have some issues.   Any deviations in timing caused by other higher priority interrupts firing will shift where in the pulse train the detach() function gets called.  For example, If serial communications is transferring a lot of data that will consume the processor to get that work done and to get those characters stored away in the buffer.  This will cause the the timing to shift.

Ziggy2013, 2015-04-13:
So it seems the potential solution is

- Make the  #define PROBE_SERVO_DEACTIVATION_DELAY mandatory with a value which allows the servo to move and settle AND

- add logic around the the detach() to ensure the control pulses terminate properly and the pin is parked in the low state.

Comments??

Roxy-3DPrintBoard, 2015-04-13:
> So it seems the potential solution is
>
> Make the #define PROBE_SERVO_DEACTIVATION_DELAY mandatory with a value which allows the servo to move and settle AND

Perhaps...  But some people are going to want to have very small numbers.  And it is RepRap!  You do want them to be able to do crazy things.   And people that use a metal nozzle that touches a metal print bed have no need for a servo even though they want to use Auto Bed Leveling.

> add logic around the the detach() to ensure the control pulses terminate properly and the pin is parked in the low state.

Agreed!  I've got my fingers crossed that will get rid of all the servo twitching problems.

> The detach() logic really should be modified in the Arduino library. But given the level of understanding in that earlier Arduino forum thread, that would be an uphill battle.

From reading that monster thread, it almost sounds like detach() doesn't get used by the main stream Arduino people.   And their opinion is if you are going to detach() it is your responsibility to make sure everything is in a clean state prior to calling the function.   I think a couple extra lines of code in our detach() function is much easier to maintain and get the right results.

c-born, 2015-04-14:
Ok, I decided I better take a look at the servo library code. Not bad, but I don't know why somebody decided to diverge so much from Upstream with a bunch of cosmetic formatting changes that do nothing to the logic but make comparisons much more difficult. And as for using the same name "servos" for the external array of Servo class structures as for the static internal array of servo_t structures in the library, why would anyone do this?!

If we want to diverge from the upstream servo library, we can simplify things a lot and reduce code and data requirements somewhat, but for me that defeats the purpose of using a common library.

Anyway, a simple fix for our current problem is to change line 81 of Servo.cpp in the handle_interrupts function from
```C++
    if (SERVO_INDEX(timer,Channel[timer]) < ServoCount && SERVO(timer,Channel[timer]).Pin.isActive)
```
to
```C++
if (SERVO_INDEX(timer,Channel[timer]) < ServoCount)
```
This ensures that the final pulse is the correct length, and doesn't expose us to any of the potential race states that changing IO outside the ISR code would do.

In our case the ISR continues to run even if there are no active servo pins, as finISR() is an empty function, so it makes no difference to timing. If anyone had concerns about repeatedly writing the output low in the ISR (and there is no reason for concern in our case, we aren't trying to use the pin for something else) we could always allocate a "pending" flag from the bitfield containing isActive and use that so as to only set the output low once, but there is no need for it.

This change should not cause a problem for Upstream, either.

As a separate change to bring us back closer in line with Upstream, as well as saving some data space, I would suggest getting rid of:
```C++
    #if defined(ENABLE_AUTO_BED_LEVELING) && PROBE_SERVO_DEACTIVATION_DELAY > 0
      int pin;                           // store the hardware pin of the servo
    #endif
```
from Servo.h, and 
```C++
  #if defined(ENABLE_AUTO_BED_LEVELING) && (PROBE_SERVO_DEACTIVATION_DELAY > 0)
    if (pin > 0) this->pin = pin; else pin = this->pin;
  #endif
```
from Servo.cpp, replacing it with 
```C++
	  if(pin == 0)
		  pin = servos[this->servoIndex].Pin.nbr;
```
The pin storage is already there, so why duplicate it and incur the extra storage space?
I don't see why Upstream wouldn't accept this into the library, although they may argue about it for some time!

thinkyhead, 2015-04-14:
@c-born Any code included within the Marlin folder is specifically there to be altered and adapted, so feel free to make any necessary changes that improve compatibility with Marlin itself, whether or not they diverge from the original code, or whether Servos library will ever accept it in their main branch. This has been done before.

For example, you'll notice that –like the Servo.cpp code– the SdFile and other Sd library code is also somewhat older, but has been adapted for Marlin. Specifically, it was given long filename support in a completely original way before the original library added such support (in a current beta version with different interfaces).

When it makes more sense to make a library into an external dependency (especially if we can get better performance and smaller code size) then the complete specific compatible version(s) of the library that the code was written for must still be included – in the ArduinoAddons folder. Code there is guaranteed to be only unaltered versions of libraries.

Ziggy2013, 2015-04-14:
Ok this is aimed at solving the issue where the detach() can leave the servo in an undefined state. However it does not solve the issue of the servo having enough time to settle before the Z endstop is tested.

This is why I keep pointing out that the delay after the servo.write() is mandatory so the servo has time to move and settle.

thinkyhead, 2015-04-14:
@Ziggy2013 So, for example, you would change this code…
```
#if SERVO_LEVELING
  servo[servo_endstops[Z_AXIS]].attach(0);
#endif
servo[servo_endstops[Z_AXIS]].write(servo_endstop_angles[Z_AXIS * 2]);
#if SERVO_LEVELING
  delay(PROBE_SERVO_DEACTIVATION_DELAY);
  servo[servo_endstops[Z_AXIS]].detach();
#endif
```
…to this…
```
#if SERVO_LEVELING
  servo[servo_endstops[Z_AXIS]].attach(0);
#endif
servo[servo_endstops[Z_AXIS]].write(servo_endstop_angles[Z_AXIS * 2]);
#if PROBE_SERVO_DEACTIVATION_DELAY > 0
  delay(PROBE_SERVO_DEACTIVATION_DELAY);
#endif
#if SERVO_LEVELING
  servo[servo_endstops[Z_AXIS]].detach();
#endif
```
?

Also, notice that `SERVO_LEVELING` is defined (based on old code) like so:

    #define SERVO_LEVELING (defined(ENABLE_AUTO_BED_LEVELING) && PROBE_SERVO_DEACTIVATION_DELAY > 0)

But, does this mean that `attach()` and `detach()` should only be called in the case where the delay has been set? What do you suppose was the original meaning of this logic?

c-born, 2015-04-14:
@Ziggy2013 After I made the code changes above I did some testing with various delay times, mainly to check that the code worked as expected and fixed the last pulse variation, but I also checked how many pulses were required for my servo to achieve full travel. In my case it was there in 4 pulses, so a delay of 300mS which gives 14 pulses is plenty of time. I expect this will vary with the hardware setup, but I think the current warning in the code should be enough to make people aware of this?

However, I think you are referring to a different delay, not the one before we deactivate the probe servo, but one between initiating the probe deployment and taking a reading from the Z-probe sensor? I have only used configurations where the Z-probe limit is in the untriggered state when the probe is retracted, so this isn't an issue, the limit can be read at any time. However I can see there may be hardware designs where this isn't the case. I would think a different delay time should be used  in that case, as it is a different function.

@thinkyhead The attach(0) and detach() should only be called when we are going to disable the servo, which only makes sense if we have a PROBE_SERVO_DEACTIVATION_DELAY >0, otherwise we won't get a single pulse out. The original code you show appears correct. I think that if there is a need to delay between staring to deploy the probe and taking a reading then that should be something separate from the PROBE_SERVO_DEACTIVATION_DELAY.

thinkyhead, 2015-04-14:
Well, there is supposed to be a notable delay. The servo should only be deployed when there is some space between the bed and the probe. Then it takes time to move the Z axis down to the bed. So where is the additional delay required?

Roxy-3DPrintBoard, 2015-04-14:
> In our case the ISR continues to run even if there are no active servo pins, as finISR() is an empty function, so it makes no difference to timing. If anyone had concerns about repeatedly writing [...]

Given where we are, your idea of fixing the irregular last pulse in the ISR is good.   But I don't like the fact the ISR is continuing to run even after the servo is detached()'ed.  That seems sloppy.   It would be a very low priority 'bug' but I think it can be argued that should get cleaned up and fixed???   It almost looks like we just need to #define WIRING to make that behave correctly?

Ziggy2013, 2015-04-14:
@c-born 

"However, I think you are referring to a different delay, not the one before we deactivate the probe servo, but one between initiating the probe deployment and taking a reading from the Z-probe sensor? I have only used configurations where the Z-probe limit is in the untriggered state when the probe is retracted, so this isn't an issue, the limit can be read at any time. However I can see there may be hardware designs where this isn't the case. I would think a different delay time should be used in that case, as it is a different function. "

Correct - this could and should be done via different delay value.  

Thanks for understanding why this is important for some probe designs.

nitewing76, 2015-07-08:
I have no problems with jitter in 1.0.2,  I only have problems in Dev and integration. I have tried different servos, on different pins...with and without stepper motors attached to RAMPS 1.4.  I also used a random number generator for (301-1021) for PROBE_SERVO_DEACTIVATION_DELAY. The jitter oscillations are so bad that I cannot perform G28/29/30. 

Can someone please help me remember how to configure 1.0.2 so the z-axis will raise before probe retraction?

@thinkyhead
@Roxy-3DPrintBoard
I see your names more than anyone else....THANK YOU for all you do for this community!!!

thinkyhead, 2015-07-09:
@nitewing76 I wish I knew more about the quirks of servos (and had one on hand to test). If adding more delay doesn't help, I'm at a loss to understand what else might be going on. But maybe a comparison with 1.0.2 code will shed some light on the issue.

nitewing76, 2015-07-09:
I do not know much about programming code...the mechanical engineering is my area of expertise.  If I do too much, I may melt down the reactors at the local nuclear power plant. lol

I did notice a few differences in servo.cpp and servo.h. So I copied them from 1.0.2, into the Dev folder and uploaded it. There was no change in the jitter of the servo, which eliminates these two files as being the problem. Correct?

Would knowing the frequency of the jitter help identify the problem?  As a jack-of-all trades (only a master of keeping my ex-wife made at me)....I do not have an o-scope, but I think that I might know an audio trick to estimate the oscillations.

thinkyhead, 2015-07-09:
@nitewing76 It sounds like the code might have been introduced in the last month or so, then. The debug_leveling branch is about that old now.

nitewing76, 2015-07-09:
I just tried Mesh Bed Leveling and the servo jittered like crazy.  But there is still no jitter with ENABLE_AUTO_BED_LEVELING (Grid, nor 3-point ABL). Hopefully this will help narrow it down even more.

thinkyhead, 2015-07-09:
> I just tried Mesh Bed Leveling and the servo jittered like crazy

With the 1.0.2 branch?

nitewing76, 2015-07-09:
I am sorry, I should have been more specific. No, I was referring to your debug_leveling branch. I was attempting to create a sort-of matrix of where the problem does and does not exist. Which now this allows the problem to researched within the same branch.

I will compare what I can and report back.

thinkyhead, 2015-07-09:
> your debug_leveling branch

Ah, ok, good. So with the Mesh (manual) bed leveling are you deploying the probe manually with `M401` or does the `G29` routine do that for you? (It looks like it doesn't deploy/stow on its own.)

thinkyhead, 2015-07-10:
Ok, this branch can be used to get debug output based on leveling code as of July 9…
https://github.com/thinkyhead/Marlin/tree/debug_G29

AnHardt, 2015-07-12:
@thinkyhead 
 With mesh (manual) bed leveling no probe is involved. Only Z is adjusted manually.

nitewing76, 2015-07-13:
I am sorry for the delay...the jittering was occurring immediately after uploading the sketchs.

I made the changes to servo.h and .cpp as recommended by Anhardt and tried the degug_29, but while the frequency of the jitter was slightly different and only jittered slightly before any motors were moved....the servo arm did start jittering like crazy after the X-axis was moved and continued after M84.

After try these. I reconfirmed that there is no jitter under any circumstances while using 1.0.2.

AnHardt, 2015-07-13:
I can not confirm there is no jitter on the servos with 1.0.2. at all. I can confirm, there is no jitter on the servos with 1.0.2 and PROBE_SERVO_DEACTIVATION_DELAY active in the cases i tested.
For the Development branch i see exactly the same.

Other observations on my test system:
Jitter is more when the servo is at ~90° than 0° or 180°.
Jitter is more when moving z than x or y (prusa i3 x,y with 17 teeth pulleys, z with M5 )
No stepper activity - no jitter.
No jitter when the servo is detached.
I see a bit stronger jitter with the Development  branch.

@nitewing76 
I guess you have not activated  PROBE_SERVO_DEACTIVATION_DELAY on the degug_29 branch.
#2425 only saves 2 byte per servo and can not help with your problem.
#2426 is ought to improve in ___rare___ random occurrences where the servo-pin is left high when it is detached. It's unlikely to improve your problem.
#2427 is mainly a logic cleanup what saves some bytes and improves the general servo-endstop-feature (not z-probe), where the detach was forgotten until now. It's unlikely this will make a improvement with you problem.

nitewing76, 2015-07-13:
I have not used 1.0.2 without the PROBE_SERVO_DEACTIVATION_DELAY. I said there was no jitter at all, but I should have said, "The only time there is jitter is immediately after reset and only for a time equal to the PROBE_SERVO_DEACTIVATION_DELAY."

I did not analytically compare the jitter at various angles, only to that there is jitter at all angles. I do concur that there is more jitter when the Z-axis moved, compared to X and Y. 

With your changes to servo.h and .cpp...I do get a small amount of jitter (about once per second, for 3-4 seconds), then nothing until after a axis is move. But M84 does not stop the massive jitter.

I also concur the jitter is stronger in the Dev branch.

The PROBE_SERVO_DEACTIVATION_DELAY is always activated.
I am attempting to implement #2426, but it has been trying to compile for over 5 mins.

The question still remains...since there is no jitter in 1.0.2 (except after reset, which is proper), then what is the difference between 1.0.2 and other branches? I tried to determine this, but I only became even more confused. However I am learning things, which is always good!!

I did try uploading the Development branch (July 8) to a bare Arduino Mega (no jitter). Then I uploaded with to a bare RAMPS 1.4...with Pololu's installed there was jitter, without the Pololu's there was no jitter.  So could the difference be in the programmed behavior of the Pololu drivers, not just the servo's?  Because even if the pic does detach properly, if the drivers were programmed differently after 1.0.2 then could EMI from one trace to another cause these problems? (the trace of pin 11 at times runs parallel to and/or directly over the various traces of the Pololu's)

I hope some of my ramblings will help you experts have an epiphany. lol  I have been digging through here for the previous 15 months or so....believe me I do not intend that as sarcasm, but honest wishful thinking!!!

AnHardt, 2015-07-13:
My thoughts went more in the direction that a heavily loaded stepper-interrupt could disturb the servo-interrupt.
If you tell me it makes a difference if the  Pololu's are in or not, other possibility's arise.
Do you have an extra power supply for the servo, or is it fed by the Arduino?

nitewing76, 2015-07-13:
The servos are powered by the same ATX PSU as the printer...which is branded as Dell, but is manufactured by LiteON. In the past I used a Motorola 5V/500mA wall-wart with the same results. 

Both the ATX PSU (12V and 5V) and wall-wart have smoothing capacitor ladders (0.01uF-->100uF, by multiples of x10). Recently I added 1000uF to the 12V/5A rail of the RAMPS, with no change (except I no longer hear a high pitched noise when the heater active).

thinkyhead, 2015-07-15:
Servo changes are being merged - the updated servo delay feature first. Please test and see how it's working.

alexandrezia, 2015-07-18:
Just a quick feedback.
I'm on this commit from 06/Jul : 4f58d72e4cf658efbe899d31e901bb0b400207cd
If I move forward to 17/Jul : 3c803c71f9bc3b0c738d87dbde41d58bbb32513d
My ZProbe servo keeps powered and locked in a certain position, do not move for Home and BAL

thinkyhead, 2015-07-18:
@alexandrezia If you undo the changes from this commit https://github.com/MarlinFirmware/Marlin/pull/2425/files does it fix the problem for you? Also, it should be noted that the servo delay settings have changed, so you need to uncomment the line `//#define DEACTIVATE_SERVOS_AFTER_MOVE`.

alexandre-zia-movile, 2015-07-18:
Hi, I know about the change in DEACTIVATE_SERVOS_AFTER_MOVE and I've set this properly.
I did revert the commit as you suggested and now my servo is free (not constantly energized as before), But now the servo does not move when homming and BALing, it keeps retracted.
Reverted to commit 4f58d72 and all back to normal.

thinkyhead, 2015-07-18:
@alexandre-zia-movile Ok, good to know. I'll try doing a direct comparison of July 6 to the current code and get a definitive list of the relevant changes.

nitewing76, 2015-07-19:
I downloaded the Dev branch (at 23:30 CST), uncommented about the servos (including the new DEACTIVATE_SERVOS_AFTER_MOVE), and ofc updated the usual config files. Still no change.

However, I did rewire the servo connector to work on pin 32 (and the adjacent +/-). The jitter was MUCH less!!

thinkyhead, 2015-07-19:
@alexandre-zia-movile @nitewing76 Hopefully #2470 addresses the issue. It patches the servo code to be more like the code from 06/Jul. I will merge it soon for you to test.

nitewing76, 2015-07-20:
Congratulations and THANK YOU!  The servo works as perfectly as Release 1.0.2!!!  All jitter on RAMPS 1.4 is within the range of the SERVO_DEACTIVATION_DELAY 300.

The next step would be eliminate the jitter entirely and add deceleration to the servo line completely. To do this, I believe the board would need to re-routed. For the deceleration...this topic is a topic for another day. 

I say these because it seems that unless the SERVO_DEACTIVATION_DELAY is timed perfectly, it could affect the accuracy of z-probes.  But then again...maybe the 2nd bump mitigates these factors.

Once again, THANK YOU for your time and your constant dedication to Marlin!!! From my perspective...this problem has been solved. :)

thinkyhead, 2015-07-23:
Good news, @nitewing76 … I hope we can address the jitter soon, and I would be interested in finding out how "deceleration" might apply. Right now we just set the servo angle in one line. Are you suggesting we set the servo angle in more steps, slowing down at the end?

nitewing76, 2015-07-24:
I do not know of the exact answer of "how" or if the time invested to do this would improve accuracy of bed leveling. If someone knows they have a bed that has a true plane, a reliable servo and switch with low deviation...then maybe they could spend a few minutes testing to see if 1-3 degrees of servo inaccuracy (caused by jitter or very cheap servo) makes a difference or if this inaccuracy is mitigated by the re-bump. If a user sets their servo to a non-perpendicular n-degrees, but the rebump moved the servo arm to a true perpendicular position (or at least an angle that is the same at every probe point)....the deceleration may not matter unless it is proven to significantly reduce the mechanical stress on the carriage.

Or maybe to set the servo angle, pause a 10ms, the reset the servo angle again and reset it between each probe point could be an easier alternative?

Or maybe it should be up to each user to calculate a more finite SERVO_DEACTIVATION_DELAY?

AnHardt, 2015-09-25:
Servo code seems to be ok now.

closed 2015-09-25 by AnHardt