MarlinFirmware/Marlin #1888: Strange deceleration issue

boelle, 2015-04-10:
Hi

I know i run of an old copy but its what i had that works 

https://github.com/buserror/Marlin/tree/master/Marlin

NOTE: My motherboard is set to 91 (Open Motion Controller from TVRRUG)

I'm printing spares for my printer and have a funny issue when the printer is doing non print moves. 

The thing is that its random.. ie it does not happen on the same spot of the axis each time, and it can happen on both X and Y or a combination... 

the only thing i have spotted is that it happens when the printer moves from one spot to the other and right when it slows down its seems like it gets stuck. but then it makes a big "CLONCK" sound and moves to the right spot.

Print are not affected in any way what so ever... 

You could argue that it does not matter... but this clonk also sends a shock wave through the printer and it cant be healthy for the bearings and printed parts

made a small video of it: https://www.youtube.com/watch?v=PN0esgTXL5c

any ideas what could be wrong?

I plan to start testing the latest code as soon i get all my printed parts replaced.... but i guess the answer to my Q could be relavant also in the latest code

BTW.... the github link is a modded version that has support for TB6560AHQ stepper drivers...

boelle, 2015-04-10:
btw... i removed the option to comment the video... lets keep it here...

and remember that the firmware i use are an 2-3 year old copy

Wurstnase, 2015-04-10:
What is the frequency of your board?

boelle, 2015-04-10:
20 Mhz.. if you mean the CPU clock?

Wurstnase, 2015-04-10:
Ok. Just thought about it.
Well, the only part which is missing is the step_loops_nominal. You could copy it in your firmware.
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L378
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L776

Are the TB6560AHQ something which activate #define CONFIG_STEPPERS_TOSHIBA?

AnHardt, 2015-04-10:
@boelle 
Show us the Extruder. Retraction with high Jerk?

boelle, 2015-04-10:
Extruder? the problem is not the extruder so why show that?

@Wurstnase yes... #define CONFIG_STEPPERS_TOSHIBA	1 enables support for the toshiba steppers, depending on the capacitor used with the steppers the pulse width needs to be between 10 µs and 60 µs

ntoff, 2015-04-10:
Kind of looks like what mine does when it moves  outwards around the perimeter to start the next loop, but magnified 10x. Mine's nowhere hear that severe.

Is it moving up in Z before the travel, then down again, and THEN doing that weird movement? Could it be the filament path getting stuck somehow? As it moves, it puts tension on the filament which suddenly gives? Mine did that when I had a makeshift filament path made out of binder clips. I thought it was skipping steps until I looked closely and it was the filament doing a click thing as it moved around in the binder clip.

Which slicer are you using and could you maybe http://www.pastebin.com a part of the gcode that appears around that area?

boelle, 2015-04-10:
i have a Z lift of 0.25 mm before all printing moves

so yes i raises Z before and lowers again after this happens....

made a screen cap from pronterface that shows where it happen in my current print
[image]

Gcode:

using kisslicer.... 

PS.... pastebin is a LOOOONG time getting the Gcode.... so will fix link later

boelle, 2015-04-10:
links to G code and firmware config.h are there.... let me know if there are other files i can put there that  would help

boelle, 2015-04-10:
made a video: http://youtu.be/IdO40QCp72o

youtube is doing processing so it will be live in about 10 mins or so, maybe less

clefranc, 2015-04-10:
@boelle 
I'm not an expert in G codes, but looking at your file, line 745524 (the last longest segment you highlighted) is:

`G1 X54.5 Y-52.44 Z72.999 E0 F15000`

The previous feedrate per minute was F240, so is it possible that acceleration from 240 to 15000 jam your steppers before it get at the end of the segment?

boelle, 2015-04-10:
i did not highlight any special section...

but yes.... that feedrate looks crazy.... kisslicer is set to 250 mm/s as the max, and in firmware its set to 500.... 

very odd

clefranc, 2015-04-10:
That F15000 is generated by your slicer configuration, the firmware have nothing to do with it.

Edit you G code file, replace all F15000 to F1000 and see if it solve your issue.

boelle, 2015-04-10:
i know.... but firmware should cap it to 500..

and the odd thing is that the slicer is set to 250 as the maximum.... 

but yes.... the only thing to test is to replace and print  again....

Wurstnase, 2015-04-10:
Your maximum speed is 125mm/s. The Toshiba-part only allow 10.000 steps/s. At 80steps/s you are at 125mm/s.

boelle, 2015-04-10:
So if my math is correct 500mm/s is 320steps/s ? well within specs

kisslicer is set to half of that.... 

or am i missing something? (i admit i might not be the sharpest knife in the drawer)

Wurstnase, 2015-04-10:
500mm/s with 80steps/mm is 40.000steps/s.
And
https://github.com/buserror/Marlin/blob/master/Marlin/Configuration_adv.h#L90
Toshiba is max 10.000steps/s.

boelle, 2015-04-10:
oki, just my math that stinks then...

boelle, 2015-04-10:
but will correct the firmware values then and give it another test tomorrow.... just a few layers should be enough to tell

thanks so far :-D

ZetaPhoenix, 2015-04-10:
What acceleration values are you using? Jerk?

boelle, 2015-04-10:
https://www.dropbox.com/s/q91ikl2vw0rlo63/Configuration.h?dl=0

#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000} 

#define DEFAULT_ACCELERATION          500    // X, Y, Z and E max acceleration in mm/s^2 for printing moves 
#define DEFAULT_RETRACT_ACCELERATION  3000   // X, Y, Z and E max acceleration in mm/s^2 for r retracts

// 
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)

boelle, 2015-04-10:
but tomorrow.... time for a bit of sleep

Wurstnase, 2015-04-10:
500 is ok. For standard cartesian printer is this the max-acceleration.

boelle, 2015-04-11:
So i set the feedrate in firmware to 124 and it still happens even at the first layer when it completes the first of the spool brackets and wants to go on to the next one

boelle, 2015-04-11:
trying with a object sliced with 124 as the max too

boelle, 2015-04-11:
did not help either.... but as i need this part i will let it go on... 5 hours estimated

boelle, 2015-04-11:
it will happen a lot on this one since there are a lot of screw holes.... but initial its not as bad as before, so i think i will just have to lower it a bit more.... maybe 20% ?

boelle, 2015-04-15:
So i started out with 125, then at 118 it was very minimal... tried to say 125 - 20% and its almost gone.... at least so little now that it will not harm anything...

closed 2015-04-15 by boelle

ZetaPhoenix, 2015-04-15:
@boelle

Your DEFAULT_MAX_ACCELERATION is very high. I think mine is 2000 for X and
Y.  Also, try turning your XYJERK down.

boelle, 2015-04-15:
will try, and thanks :-D

nophead, 2015-04-15:
The acceleration in Marlin is definitely broken at high speeds when it starts double / quad stepping. I have a logic analyser hooked up to the the E step signal and the speed jumps by a factor of 2. I will open a new issue when I understand what is going on.

Wurstnase, 2015-04-15:
This shouldn't, because the steps are halfed when reaching double stepping. I've looked alot in that part of code and made also a python code to look at it and I can't find any bigger issues.

boelle, 2015-04-15:
i could also reopen this one? 

but i have found out that these toshiba steppers are a pain in the *beeep* to deal with... So i might give up on them.... unless they really are great but just need that special handling

nophead, 2015-04-15:
It isn't just Toshiba drivers. I am using A4988 and get this problem on the E motor because it has the highest steps per mm.

@Wurstnase I see it switch to double pulses but the period stays roughly the same, so the speed instantaneously doubles.

boelle, 2015-04-15:
being the novice still... are there any advantages with the toshiba steppers ? 

other than its easy to bolt a heatsink on them :-D

nophead, 2015-04-15:
I think the "advantage" is they are available through hole for people who want to hand solder their own boards and they can handle more current.

Wurstnase, 2015-04-15:
@nophead When does this appear? Only at the moment when going from single to double to quadstepping. Or the hole time while in double/quad stepping?

nophead, 2015-04-15:
It seems to happen when it gets to the plateaux speed. The acceleration rate is not fast enough, so it doesn't hit the target speed and so jumps to it at the end of the acceleration period. I am not sure if it is simply a coincidence that it gets to about half the target speed and that is just within the single step range but the plateaux speed needs double step.

Wurstnase, 2015-04-15:
I have that problem while I made the port to Due.

This happens when the processor is to slow. Only reducing the double/quad-step frequency can help then. On the plateaux the calculation for the timer does never run, because this is OCR1A_nominal. And there you have no problem with the speed of the timer-calculation.

nophead, 2015-04-15:
Why did you have the problem with the Due then? Surely that is a lot faster.

Wurstnase, 2015-04-15:
Because I use 1/64 µstepper driver. With the standard division (84MHz/2/step_rate) for the timer the maximum single step rate is ~46k. With 320steps/mm this is just 100mm/s on a coreXY. With the lookuptable and some modifications in the timer-calc (32bit, dword_near etc) it's now at ~100kHz. I run it at 90kHz realy stable.

nophead, 2015-04-15:
It seems to work properly with lower accelerations. Somewhere above 3500 is goes wrong when the target speed is also high.