MarlinFirmware/Marlin #1889: How git can help us

alexborro, 2015-04-11:
I strongly suggest we start some kind of management on Marlin development. 

Check this out: http://nvie.com/posts/a-successful-git-branching-model/

Cheers 

Alex.

thinkyhead, 2015-04-11:
At the very least a master and development branch. But I think – not sure – only Owner users can do the merging between these branches. …?…

boelle, 2015-04-11:
you are right.... and i left the owner group some months ago.... 

not the smartest thing to do, but always easy to be smart afterwards

@ErikZalm & @daid is it possible to get an invite so i can get back in owners group?

oysteinkrog, 2015-04-12:
1. A minor thing; I think it would be nice to use "develop" instead of "Development" (as a general naming pattern).
2. At the moment the history is messy, IMO all PR branches should be _rebased_ before merging!
A PR branch should never ever have a merge in it.
3. There are scripts to help facilitate various git workflows, but personally I've always found them too restricting.

thinkyhead, 2015-04-13:
What are the commands to rebase a branch that has been worked on and has commits already pushed up to the branch in my fork? I can never remember how that goes, as much as I use git every day…

oysteinkrog, 2015-04-13:
Well usually a simple git remote --update && git rebase marlin/Development will do (where marlin is the name of the remote pointing to this main repository).
If you want you can also do an interactive rebase which makes it possible to squash etc.
I'm not sure that squashing to a single commit is always desired and it might also be a bit much to ask from everyone.

thinkyhead, 2015-04-13:
@oysteinkrog My procedure has been to use my upstream connection to MarlinFirmware/Marlin. I added this to my BASH `.profile`:
```
gitup() {
  cd /Volumes/Projects/MAKER/Firmware/Marlin-thinkyhead
  git fetch upstream
  git checkout Development
  git merge upstream/Development
  if [ $# == 1 ]; then
    git checkout $1
    git merge Development
  fi
}
```
Then I can just use `gitup` from the repo to get the latest commits, then sync them with my `origin` fork to keep that up to date. I used to use `got rebase` more frequently, but it was a real hassle. I found myself redoing the same merges over and over. So maybe I was doing it wrong…?

oysteinkrog, 2015-04-13:
Hmm, I think just substituting merge with rebase there should not cause much extra trouble.
The big difference is if you get a conflict, with a merge there will be one big conflict that you resolve and then commit. With rebase you will get a conflict "on" the commit that causes it, then you resolve it and use git rebase --continue.

gregrebholz, 2015-04-13:
I'm not a git wiz but in general rebasing is only viable on git repos that are 100% private.  Once you've pushed your changes to your fork of a project on github (for a PR) you have no way of knowing who else may have cloned your repo, and a rebase of it will break their commit history badly.  Once a PR has been submitted the requester has to keep it "fresh" via @thinkyhead 's process (creating new commits to merge advances in Development into their own PR) until it is accepted or rejected.  The only way I know of to keep that to a minimum is to make decisions on PRs quickly.

oysteinkrog, 2015-04-13:
That is really not correct, it is generally better to rebase PR branches (and force push the updates to your fork/repo), otherwise reviewing a PR becomes much more difficult, and the resulting history is very messy.
By rebasing the PR branch (and squashing fixes etc that come up during the review process) all that is avoided.
It is true that force pushing is something one should avoid for repos/branches that others rely, but a PR branch is not something others should rely on (outside of the scope of the PR), and as such it is an exception to the rule.

thinkyhead, 2015-04-14:
As long as commits haven't been merged into your own master branch they can always be squashed, even if they have been pushed to Github – but you have to start a new branch, use `git rebase` to fix it up, then push the new branch to Github. For example...
```
git checkout messy_branch ;# published branch with lots of commits and merges
git checkout -b cleanup_branch ;# a new unpublished copy
git rebase -i HEAD~11 ;# use the number of commits on this branch
git add Marlin/my_merged_file.cpp Marlin/another_fixed_file.h
git rebase --continue
git push origin
```
You can use some git magic in BASH to list all the unique commits for the current branch:
```
git rev-list HEAD ^Development --topo-order ;# commit hashes not in Development
git log Development..HEAD ;# commits not in local Development
expr `git rev-list HEAD ^Development | wc -l` ;# get the number of commits
```
Then this line puts it all together…
```
git rebase -i HEAD~`git rev-list HEAD ^Development | wc -l | xargs -I{} expr {}` ;# rebase to the 1st new commit
```

boelle, 2015-04-15:
@ErikZalm @daid any of you have time to give me and maybe @thinkyhead an invite to owners group so we are able to create branches when that time comes?

boelle, 2015-04-15:
thanks to @daid

boelle, 2015-04-15:
I cleaned up the readme.md a bit and added space for a donation link

while thinking of it would it not be the most ideal situation to use our domain http://www.marlinfirmware.org/ for documentation? and then only use git for the code?

thinkyhead, 2015-04-16:
@boelle True, there is some duplicated effort in having documentation as markdown within the repository now that there is a wiki. It should be easier to get help with editing the wiki than editing the markdown.

closed 2016-04-23 by jbrazio