MarlinFirmware/Marlin #1894: Z_PROBE_OFFSET_FROM_EXTRUDER not working

fotis400, 2015-04-11:
Still fighting with autolevel.

It seems to me that Z_PROBE_OFFSET_FROM_EXTRUDER is not working at all. Neither M851 has any effect.

Whatever I do, the head height after both G28 and G29 is always dependent on Z_MAX_POS.

In other words G01 Z0 will position the head always on the same height from the table, regardless my probe settings.

thinkyhead, 2015-04-12:
The only place that the `zprobe_zoffset` is applied is in `G28`, whenever Z is homed. I notice that it isn't applied in `homeaxis(AxisEnum axis)` and it isn't applied in `axis_is_at_home(int axis)` (which is a setter function, not a getter function, if the name is confusing – need to change that). I would think that if the robot _knows_ that it is using a Z probe to find the home position, then the code to add the probe offset (for Z) should be in `homeaxis` or `axis_is_at_home`. Then, presumably, it will always be correct for Z.

But anyway, I have posted test code that you can use to help debug the homing and leveling issues. I just added logging at the point where it adds `Z_PROBE_OFFSET_FROM_EXTRUDER` (actually `zprobe_zoffset`) to the current position. Another thing is to of course make sure your `home_offsets` are cleared, which you can do with `M206 X0 Y0 Z0`. Please give the test code a run and post the output log here, when you have the time…

https://github.com/thinkyhead/Marlin/archive/debug_leveling.zip

fotis400, 2015-04-12:
here you are

http://pastebin.com/xUBWmDTg

MagoKimbra, 2015-04-12:
In an my old code marlin, I'm find in the void set_bed_level_equation_lsq (...) this line:
// Put the bed at 0 so we do not go below it.
current_position [Z_AXIS] = zprobe_zoffset; // In the lsq we reach here after raising the extruder to the two loop structure

In the new code of this line there is no trace ... In addition there is no function that corrects Z of zprobe_zoffset ....
Could this be the problem. Right now I do not have the printer on hand to try, they are out for a long time to work ....

Exusme for my bad english....

thinkyhead, 2015-04-13:
@MagoKimbra You are welcome to mess with that line of code. It was removed because it was apparently superfluous, and it was assumed that the inverse Z correction to the current position is desirable. I'm not sure that it was guaranteed that the probe would be touching the bed at this point, either, since there is now probe retraction that may happen ahead of that line. But possibly the probe retraction has no effect, and maybe that line was important. You can find a lot of debate on the question, starting with this comment… https://github.com/MarlinFirmware/Marlin/issues/1682#issuecomment-86174949

MagoKimbra, 2015-04-13:
Yes ok, but it is the only section where is Z point correct of z probe offset! In gcode_g28 this line exist.

--- Messaggio originale ---

Inviata: 13 Aprile 2015 07:30
A: "MarlinFirmware/Marlin" <Marlin@noreply.github.com>
Cc: "MagoKimbra" <magokimbra@hotmail.com>
Oggetto: Re: [Marlin] Z_PROBE_OFFSET_FROM_EXTRUDER not working (#1894)

https://github.com/MarlinFirmware/Marlin/issues/1894#issuecomment-92217576

thinkyhead, 2015-04-13:
@MagoKimbra So try it and see what happens.

GustavoGoulart, 2015-04-16:
Hi there,
I recently bought a inductive sensor for my Prusa i3 (that is running with Marlin 1.0, arduino and ramps 1.4)  and I´ve been fighting with this issue for 4 days now, researching every forum about RepRap and Marlin. I thought I was alone with this problem, because everyone told me it was easy to do.

Today I spent some hours reading some debates here at Github about this issue, I´m no programmer (once was...) But I´m here just to comment that the issue maybe had started before, because if I have the 1.0 version and got the same problem. Unless, offcourse, the guys who sold me the kit, have given me a hibrid version of Marlin, altered by themselves.

By the way, is there any temporary solution for this issue? Somthing to just get printing without going back to regular z endstop?

Thanks.

thinkyhead, 2015-04-18:
@GustavoGoulart Yes, it does seem that the probing logic has always been a little convoluted and spaghetti-like. Give this fork a try and see if it works better: https://github.com/alexborro/Marlin-BedAutoLev

If it does work any better, I will use it as a guide to fix up the latest code.

thinkyhead, 2015-05-09:
We're getting close to clearing up leveling issues. Check out #2040 for the latest developments. Basically, I've narrowed down the culprit to just one of a few recent changes. I may roll the code back since – although not perfect – the code I'm testing with may be a little more correct.

boelle, 2015-08-03:
moved to: https://github.com/MarlinFirmware/MarlinDev/issues/30

closed 2015-08-03 by boelle