MarlinFirmware/Marlin #1900: Babystepping does not work with DRV8825 stepper drivers

Ziggy2013, 2015-04-13:
In stepper.cpp the babystepping step pulse duration is set using a useless piece of floating point code that just wastes some time. 

 //perform step 
    WRITE(Y_STEP_PIN, !INVERT_Y_STEP_PIN); 
    #ifdef DUAL_Y_CARRIAGE
      WRITE(Y2_STEP_PIN, !INVERT_Y_STEP_PIN);
    #endif
    {
    float x=1./float(axis+1)/float(axis+2); //wait a tiny bit // absolutely useless
    }
    WRITE(Y_STEP_PIN, INVERT_Y_STEP_PIN);
    #ifdef DUAL_Y_CARRIAGE
      WRITE(Y2_STEP_PIN, INVERT_Y_STEP_PIN);
    #endif

This seems to work ok with A4988 stepper drivers which require a minimum 1us step pulse duration.  However it does not work with DRV8825 drivers which require a minimum 1.9us step pulse.

Suggested solution (which works) is to replace the useless floating point code in the X, Y and Z babystepping functions with

delayMicroseconds(2);

ntoff, 2015-04-13:
don't tell me that :( I just put some DRV's in my Mendel!

Wurstnase, 2015-04-13:
The useless floating point was removed some weeks ago already.

Ziggy2013, 2015-04-13:
Yes I see the latest version of stepper.cpp the useless code has been replaced with

_delay_us(1U); \

However 1us delay is not enough for DRV8825 steppers - check the spec sheet

http://www.ti.com/lit/ds/slvsa73f/slvsa73f.pdf

min 1.9us high (or low) step pulse is required.

ntoff, 2015-04-13:
Does this affect how the DRV's work in normal printing mode? I've not noticed any weird issues but I'm using a rather old version of Marlin. Is this *only* affecting baby stepping?

Ziggy2013, 2015-04-13:
AFAIK only babystepping but it appears there are many other changes to the latest stepper.cpp

If you are using 1/32 stepping on the DRV8825 there "may" be an issue with the stepper handling performance - I have not tested this at all.

Wurstnase, 2015-04-13:
It could be, because in the past first all motors WRITE(HIGH) and after that they WRITE(LOW). Now they WRITE(HIGH) and (LOW) for each motor independent. So the delay could be too short.

ntoff, 2015-04-13:
Well I built a second printer JUST to test out dev versions of Marlin (so I don't have to molest my Mendel) so I can throw some DRV's in that thing to test whatever code you want to throw at it.

@Ziggy2013 I'm not 100% sure how the DRV's are configured vs the A4988's but Marlin's steps / mm leads me to believe they are indeed 1/32 stepping. I've successfully printed a few things with them in there so at the moment they don't seem to be causing any issues. I haven't really printed anything above 30 - 45mm / s though.

Wurstnase, 2015-04-13:
I think there could be some problems when reaching double stepping. This is approx at 70mm/s depending on your pulleys.

ntoff, 2015-04-14:
I've never even attempted to print over 60mm/s as it just seems too violent, but I do have my travel speed set to 150mm/s.

I'm using GT2 belts with 20 tooth pulleys on the Mendel and the test printer has T5 belts with 10 tooth pulleys.

Wurstnase, 2015-04-14:
Travel speed or printing speed makes no different for the DRV. So if travel works with 150mm/s, printing will also work.

Ziggy2013, 2015-04-14:
If there turns out to be DRV8825 print/travel speed issues, IMO a separate issue needs to be raised.

The babystepping delay is obviously (refer spec sheet) not enough for the DRV8825 driver.  The step pulse duration should be increased to at least 2us.

thinkyhead, 2015-04-15:
@Ziggy2013 Should we change the delay to 2µs in all cases, or should we add a configuration option that will make this 2µs only for DRV8825 drivers?

Ziggy2013, 2015-04-15:
@thinkyhead IMO a configuration option is not necessary.

It's possible that some printers will mix A4988 and DRV8825 drivers, so best to keep this simple and make the step pulse long enough for both driver types (ie 2us).

BTW it's not clear to me why  _delay_us (xU) is being used in stepper.cpp. There is a standard Arduino function delayMicroseconds(x) which could be used instead. Maybe I should look at how the standard function is implemented at the assembler code level.

Wurstnase, 2015-04-15:
Because of

> When an interrupt is received during the execution of the delayMicroseconds(), the timing of delayMicroseconds() will be wrong.

ntoff, 2015-04-15:
I actually have DRV's on X and Y with A4988 drivers for the extruder and Z so yeah I'm a mixed case. 

I just a tiny bit of testing and I'm using Marlin 1.0.2 on the Mendel and I can get up to 350mm/s on the DRV8825's without it skipping any steps or doing anything weird but momentum keeps it going at the end so it skips a step or two simply by the weight of the cradle overcoming the holding force of the stepper.

So whatever timing Marlin 1.0.2 uses for the actual travel moves seems to work just fine for (at least for my) DRV8825's.

I'll throw a couple of spare DRV's into the test printer right now and test it out with the latest dev version. @thinkyhead let me know if you want to use it to test any potential fixes

nophead, 2015-04-15:
I think there is a bug with high step rates where the deceleration doesn't
work properly. That is why it skip steps at the end I think. It was an open
issue for DRV but I think it got closed without resolution.

Ziggy2013, 2015-04-15:
@Wurstnase 

Because of

    When an interrupt is received during the execution of the delayMicroseconds(), the timing of delayMicroseconds() will be wrong

And if you disable interrupts to accurately delay by 2us, then the timing of something else could well be wrong. The DRV8825's require a minimum of a 1.9us step pulse - a longer pulse is not an issue.

Wurstnase, 2015-04-15:
> According to the babystepping code as implemented there will be no interrupts during execution of the code.

Sure, it should not. The babystep is executed in the temperature-ISR. I don't know if that could be a problem. There is no cli or sei in it. So I think, it could be interrupted.

boelle, 2015-04-15:
can we upgrade this to verfied or?

Ziggy2013, 2015-04-15:
@boelle  Yes. Upgrade to verified

Ziggy2013, 2015-04-15:
@Wurstnase 

    According to the babystepping code as implemented there will be no interrupts during execution of the code.

Sure, it should not. The babystep is executed in the temperature-ISR. I don't know if that could be a problem. There is no cli or sei in it. So I think, it could be interrupted.

Typically processors disable interrupts in hardware when an interrupt is triggered and re enabled when the interrupt routine returns - this is done in hardware to avoid race conditions. The ATMEGA 2560 also does this ref http://www.atmel.com/Images/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf  section 7.8. So there is no need for an explicit disable/enable interrupts in Arduino ISRs

thinkyhead, 2015-04-16:
#1923 should deal with any babystep issues. During actual printing there doesn't appear to be a problem for DRV8825, is that right?

Ziggy2013, 2015-04-17:
AFAIK there is no issue with DRV8825's during printing - but this needs testing to verify

ntoff, 2015-04-18:
I've done a few prints on the Mendel with v1.0.2 and a couple on the test printer with the latest (at the time, so before this issue was fixed) dev version and neither of them had any issues while printing. Both using DRV8825 only on X and Y, with A4988's on the E and Z.

thinkyhead, 2015-04-18:
The new code will add one extra second of print time for every 1 million steps, which should not be too noticeable. But please report if it actually slows things down.

Fixx, 2015-05-05:
I'm using DRV8825 and I can print & babystep(just tried Z) fine.

AnHardt, 2015-07-18:
No further negative reports on this. Closing.

closed 2015-07-18 by AnHardt