MarlinFirmware/Marlin #1905: Organization & Governance

msutas, 2015-04-13:
I love Marlin, it is a big conrtribution to Reprap. I am not much involved in the development of it but when I find a solution for something I need, I propose the addition of it to the Marlin. I do not follow the issues closely. I wanted to point out the biggest issue that I have seen so far.

This software is evolving but nobody knows where it is going. New features are being added and corrections are being made continuously. But all these changes creates new bugs and problems.

Several times I have seen threads with the subject coders or testers needed and complaints about not being able to cleaning up the bugs list. However, the problem is not the lack of coders or testers but it is the lack of governance and organization.

If it continues like this, in a couple of years Marlin would end up being a useless bunch of code. Maybe I am too pessimistic. 

Please do not misinterpret me. All collaborators have valuable contributions to the software. I do respect and thank all the people involved in this project. But I believe another milestone is needed at the moment. That is defining a core development team which is open to public with clearly defined roles and responsibilities who defines the future development path of the software while openly taking feedback of the users and managing the development accoring to the targets.

I believe this change should be initiated by the project collaborators and I am ready here if I could be of any help in the organization and doing the secretarial work of this formation such as defining the roles, responsibilities, procedures and organizing meetings.

Roxy-3DPrintBoard, 2015-04-13:
> If it continues like this, in a couple of years Marlin would end up being a useless bunch of code. Maybe I am too pessimistic.

I disagree.  Even now, the Marlin code has been separated into a development branch and a release branch.   How did that happen if there is no Organization and Governance?

My guess is that if features keep getting added at a high rate of speed we may end up with multiple versions of Marlin.  Some tailored to this and some focusing on that.   But that only happens if there are enough people interested in a particular branch to push it forward and maintain it.

msutas, 2015-04-13:
@Roxy-3DPrintBoard ,That does not mean the development path is being planned and the development is headed to a direction with analysis. The development branch is evolving without control and the development branch will end up to be a release branch. I believe Marlin is getting fat day by day. If there were also a branch such as experimental, I would share your thoughts.

I still believe there should be a core management team who are making the decisions open to the public with the feedback from the users. Don't think about this team as a team of elite people with the priviledge of making decisions for lifetime. Every role and responsibility should be determined and the members should be changed periodically say once a year.

Roxy-3DPrintBoard, 2015-04-13:
Well...  I understand the benefits of what you are suggesting.  And in fact, if this was being developed by a commercial company (for profit) they obviously would do something close to what you are suggesting.  They would decide what features were needed and weigh those in terms of development cost and support issues.   There would be much less chaos.

But instead we have an Open Source project where people are donating their time and effort.  So...  in that model, people can do what ever crazy things they want.   And if those things have merit, they can be pulled into the main branch.   Or for that matter, any other branch of the tree if the owner of that branch sees merit.   

If Marlin gets too big and fat, I'm pretty sure somebody is going to say "My branch of the tree is lean and mean."   The problem will fix itself.

emartinez167, 2015-04-13:
I agree with Roxy. If anyone does not like Marlin the way it is, feel free
to create your own branch and do whatever you like with it. This is Open
Source.

scotty1024, 2015-04-13:
This is all well and good in theory but let me share with the most important thing I learned from my Eagle Scout project: you want people to work on what *YOU* want them to work on you better feed 'em or pay 'em. Because otherwise they're generally just going to work on what they want to work on, and they may or may not finish even that.

Personally I'd help fund a Marlin Bug Stomp Event with pizza, beer and t-shirts at some 3D conference with enough days to actually be able to stomp some of the more pressing bugs.

msutas, 2015-04-13:
@emartinez167 @Roxy-3DPrintBoard You are misunderstanding me, I am not against the open source logic. I believe the development should be more systematic. Actually the collaborators are somewhat trying to do this, if you check the labels on the issue list, most of them marked as on hold. While they are trying to finalize the bugfix round, their efforts are staying on the individual level and if there are decisions about the development path of the software it is not shared to the public.

@scotty1024 This is human nature but people also like to be a part of a succesful team devoted at reaching a common goal. Sometimes a reorganization may lead to success.

thinkyhead, 2015-04-14:
"Agile" & "Scrum" development processes are modeled somewhat on the open source approach, where software is allowed to develop somewhat more organically instead of having a top-down control structure. We have been making more use of Milestones, loose goals, and –most recently– this current Bugfix Phase to limit the development process for a while and prepare for a tagged release.

The code is being moved in a certain direction through developer initiative and open collaboration. At the moment the focus (for me) is very much on preliminaries: cleaning up and organizing the code, fixing bugs in the myriad of features and improvements that have been added since the end of last year, and getting to the point where it can be tagged 1.0.3. Following that milestone, the code is in need of better encapsulation – so in the next phase code units will be encapsulated into simple singleton classes with static data members.

Now what I do agree with is, it would be helpful to have funding for development! :wink: But to be serious, no one has much time to devote to Marlin, and I am personally devoting a lot of time to take – primarily along with @boelle – a custodial role in Marlin, and to bring together developers who are interested in moving the project forward and dealing with the various deficits in the code-base. For developers like myself to be able to truly devote time and energy to Marlin without undue pain, some funding to cover food and rent would be most welcome.

As for the process leading to bugs in the Development branch, I don't see that as problematic. That is the point of the Development branch: to have a channel for code in development to be tested, so bugs will be found. A lot of brave testers have decided to try the Development branch instead of sticking with 1.0.2 (not that 1.0.2 is bug-free) and their reports have been extremely helpful.

boelle, 2015-04-14:
I agree all the way here, thou i have a hammering headace right now (painkillers not in effect yet)

And when we put up the website i suggested some sort of donation link, maybe put up 2, one for those who wants to donate towards cost of running the site, and one for those who want to donate towards the development itself

of course we can always just have one link and split it up internal.

I have no need for funding for the work i do,

TechMasterJoe, 2015-04-16:
i think what thinkyhead, boelle, and all the other offical dev coders need to do is a simple 1~2week  indiegogo. just say truth simply ask for what you think is fair allow 1~100$ donations asking for help covering food and rent so you can focus on fixing bugs and taking marlin to the next step.

if you did this I know i chip in say 20~40$ or so it's the least i can do marlin is on all 12 of my printers lol
1 perk you can toss in is a simple 3d printed version of the marlin logo for donations over say 30$ simple fast to make. i can see you guys making more then a few grand if you push it to all the printer mfgs that are building printers that run marlin. craps on every airwolf printer that they sell for 4,200$ for marlins sake I'm sure they can help a tad

emartinez167, 2015-04-16:
I will be more than happy to contribute!

thinkyhead, 2015-04-16:
@TechMasterJoe That's sort of the direction I was thinking. And as I'm about to move from Seattle to Portland in June, the timing would be good to set that up as a Summer sabbatical project. I have a working 3D printer —in fact running a calibration print with the latest Marlin code right now— and would gratefully produce Marlin logos in glowing PLA (or other PLA colors I have lying around :grinning:) for anyone willing to contribute to the worthwhile attention this project needs.

boelle, 2015-04-16:
> logos in glowing PLA (or other PLA colors I have lying around [image:

thinky.... you have an paypal or something that people can donate too?

boelle, 2015-04-16:
@thinkyhead or do you think that a joint link of some sort is better?

thinkyhead, 2015-04-16:
@boelle I have done that sort of thing before, just asked for donations by PayPal. For example... http://www.thinkyhead.com/tabletmagic/contribute . But I believe that a proper fundraising campaign can do some good things to also draw attention to the project and bring some anticipation for future updates.

boelle, 2015-04-16:
I agree,,, just thought of putting something up until we get that nailed

thinkyhead, 2015-04-16:
Hmm… Well, I can put up a similar page for Marlin as I did for TabletMagic, at least make a place where donations can happen.

msutas, 2015-04-16:
Organizing a fund raising campaign each year would be a quick way to reach targets in addition to keeping a donation link. I am ready to support.

The important thing here is again organization. People must see where the funding goes. This is a free open source project and should stay as it is. If this project turns into a commercial business, that would not be fair for the people who did support the project both financially and intellectually.

boelle, 2015-04-16:
it will never go commercial

the money will as said go towards food/rent money for the active developers

and the website..

nophead, 2015-04-16:
Paying people to do work, how is that not commercial? I am sure the tax man will class it as income.

boelle, 2015-04-16:
Then again not many do work for free right? And whatever people tell the tax man about the extra income will be up to each individual's own consciousness

I think in general this is needed just to attract coders to the project.... the outcome of course should/will still remain open source and free.

nophead, 2015-04-16:
>Then again not many do work for free right?
Yes they do, that is how open source works in the main.

boelle, 2015-04-16:
That is also how it should be

But the marlin firmware needs more than one person working on it, and that one person can only do so much, the bugs are just piling up 

I think the idea behind donating is not to give people a full time job, but just enough to say thank you for the effort and to attract people...

Roxy-3DPrintBoard, 2015-04-16:
> But the marlin firmware needs more than one person working on it, and that one person can only do so much, the bugs are just piling up.

I have code ready to go...  But I've been waiting for things to stabilize.  

The honest truth is I would rather watch a college football game than an NFL game.   The college teams are not quite as good, but they are playing because they want to play.

msutas, 2015-04-16:
There is a thin line between being commercial and rewarding a coder. Again
defining the process of managing the funds is very important.

Maybe rather than hiring coders.  Rewards may be assigned to the issues and
whoever makes the acceptable solution to the issue receives the reward.
This may attract some coders to deal with the project.

nophead, 2015-04-16:
Why not use https://www.bountysource.com. OpenScad uses it and it seems to work.

msutas, 2015-04-16:
That seems promising. It is much more manageable than collecting funds and
managing them individually.

Roxy-3DPrintBoard, 2015-04-16:
> Why not use https://www.bountysource.com. OpenScad uses it and it seems to work.

I was just looking at some of the bounties put up for OpenScad issues.  It looks like they have been very successful getting people to work on the various hot topics.   Also, a lot of small companies are shipping Marlin firmware with the printers they make.  This would give them a direct path to getting the issues they care about fixed and/or developed.

scotty1024, 2015-04-16:
Open source isn't about people "working" for free. Open source is about the source being free so people can modify it if they feel the need/desire to do so. RMS fully supports people being paid to do open source development.

thinkyhead, 2015-04-18:
Quite so, Scotty! Open source is not martyrdom. I have loto of talent and experience but very little equity (read:none), so in the interest of pain-reduction I ask for donations for some of my open source work. Sometimes I get paid to fix bugs in open source software for an organization and then contribute those fixes back to the community. As an open source GPL project, no individual could possibly require people to pay for Marlin as a product and lock it behind a paywall. I humbly ask people to support my independent efforts to move projects that I find worthwhile forward, because I am poor but I like things like heat and food.

The old models of investment - meaning money, time, effort, material - are giving way to philanthropically funded models, fueled by the Internet. I for one would like to make full use of the good will that projects like Marlin rightly earn, to make it possible to contribute in all the ways that I can. I don't aim to ride on the coattails of Marlin for some kind of free lunch, but to give my fullest and best attention and to heighten its quality. Time I spend on Marlin actually does _cost_ something for me, burning calories, heating fuel, and caffeine, whilst also not having those hours available for personally profitable ventures. And believe me, I have put in hundreds of hours on Marlin in the last few months. Not because I have a bank of RepRaps lining my wall (I don't), but because I really love this technology and the nexus it creates.

boelle, 2015-04-18:
I cant even with my best will see the problem with donations... they are not required at all

If people dont like to donate then just dont donate instead of this "mud fighting" back and forth

There is absolutely nothing wrong with a small thank you for the work you put in to it

nophead, 2015-04-18:
No but I don't think a "small thank you" will affect how many coders you get working on it. Writing software and fixing bugs takes a long time. If people were to be motivated to do it with money you would need to pay the going rate for a programmer, which would be large amounts of money.

Roxy-3DPrintBoard, 2015-04-18:
> No but I don't think a "small thank you" will affect how many coders you get working on it. 

Actually...  I do think it will have an effect and it is going to be negative.   I agree ThinkyHead has been the biggest contributor here lately.  But I contributed code in the past and have more code ready to go.   What makes sense to me is to wait until things get more defined and I see the 'lesser' contributors getting 'Thanked' before I contribute any thing else.   There is no reason to donate my work if other people are being 'Thanked' for their work.  

Can you see how this can snowball into a negative effect?   At least the https://www.bountysource.com system would encourage the development in specific issues.

JustAnother1, 2015-04-18:
It doesn't make sense to give small amounts of money to developers. Either give them a full payed job or no money at all.

A different Idea that might work:

Give away hardware or commercial software licenses.

Sometimes the developers don't have the chance to test their code as they lack the needed hardware, or diagnostic tools like oscilloscopes or software tools like simulators. And it is probably harder to write good code for a delta printer if you don't have one sitting on your desk. Especially for those expensive things there could be some scheme of mailing it to the developer currently working on something.

Giving the needed hardware to those that want to develop but cant would help get more developers working on Marlin.

A printer vendor could donate a printer to Marlin to get better support for his printer. The small donation of people interested in the support of the printers would be enough for the shipping costs (if the developer is not at a too remote location) and I think it would be a huge boost of motivation if you get a printer even if only borrowed for a few weeks/months.

Others might already own expensive tools needed to find a bug, but they don't have the printer/the electronics that has the bug,...

But then this would need a lot of organization. Detecting the bugs where additional tools could help, managing the tools and printers that can be shipped, find people that donate things that can be shipped to the developers,  find people that have tools they would use if the problematic hardware would be shipped to them, managing the donations for the shipping costs, customs clearance,..

scotty1024, 2015-04-18:
A project like this could provide an early unit to get it added to Marlin:

https://www.kickstarter.com/projects/wr3d/the-diamond-hotend-single-nozzle-multi-color-3d-pr?ref=discovery

Roxy-3DPrintBoard, 2015-04-18:
> A project like this could provide an early unit to get it added to Marlin:
>
> https://www.kickstarter.com/projects/wr3d/the-diamond-hotend-single-nozzle-multi-color-3d-pr?ref=discovery

Having some hardware like this donated would motivate me to get it working.

scotty1024, 2015-04-18:
This project used to have it's "Linus" whom was passionate and provided passion and direction for others.

Now instead of passion we discuss cash. And as the old saw says about that: now that we've determined you are a prostitute, let's discuss price?

The grandma story I prefer is baking the loaf of bread. Many steps in making the loaf, not all of them equally fun as eating the loaf.

Why would people want to join in making the loaf? Some might want to learn how to do some particular task. Debugging for example is something many have little skill for. Working in a project with others can help people build skills. But you need a structure in which people can work together: hangouts, 3D printer fairs etc...

As shown by the attempt at a hangout this weekend, these aren't free. You need someone that enjoys organization and cat herding to set them up.

And whatever is done? One of the big things that is missing is a sense of community and we're all in this together each doing our little bit. This community is missing that. We need our moments during the year where we sit together and eat the hot bread from the oven together. Share the stories of making it and each member realizes it is a worthy use of their time to continue in this community.

thinkyhead, 2015-04-20:
In the interest of doing the kind of work I love, I would also accept donations of time, space, food, and equipment in lieu of cash, since it is only for the sake of those things that cash is even mentioned. Invite me to stay at your villa for a few months so I don't have to worry about all this rent rigamarole. I can also cook fabulous vegan food and provide therapeutic massage, if my sparkling wit and musical talent should prove extraneous.

TechMasterJoe, 2015-04-20:
how about allowing each person to take donations and name projects and code pushes they have been a part of leaving it up to the dev / coder to build the page and find his own hosting (not like this takes any effort)
then let people that want to donate pick who they can or want to donate to
and have it as a posted rule that devs can't tell anyone what they receive to prevent jealousy amongst them.
it makes no deference to me how it's done
the truth is that the "just take anything in" standard has distorted marlin making it bloated and full of holes and only a few in marlin's history has stuck out the fight to bring it back.
thinkyhead has every right as an American to ask for donations for the work he has done
the same for every GPL coder on earth let alone the ones apart of the marlin project

but in all seriousness this is not the type of talk to have here i think Marlin needs it's own Forum setup in a clean quick way with it's own file hosting to do all this dev work.

P.S. thinkyhead i have a few printers i will more then happy part with if it helps nail the bugs down
but you can't have the monsters in the shop they are not finished yet lol
things i can part with range from old cupcakes to printrbot simple Metal with all the latest mods

Details on the monsters (no drooling aloud yet i need to get them running and post vids first)
7 foot tall delta 80% done needs real steppers/drivers or servos and belts but on hold.

94in long 28in deep 72in tall fully enclosed Core-XY with 4 0.8mm nozzles each is fed by 2X 1.75mm filaments for a total of 12 feeds LOL
using leadshine servos NEMA 34's drive 1in wide belts
bed rides on 4 25mm 5mm pitch acme screws but the screws are fixed, 1 servo belt drives the acme nuts around the screws this reduced rotating mass my 40% and let me pull 4000mm/s acceleration on z and over 10000mm/s on x and y with the core XY layout. the print bed is almost 100lbs without glass really want a 32 bit ver of marlin someday soon

thinkyhead, 2015-04-24:
@TechMasterJoe Good points on how developers may manage their own financials. I have pretty much decided to make any indie solicitation efforts independently. Not too proud to beg!

And, thanks for the hardware offer, I may take you up after I finish my move to Portland, sometime in June. I need no monsters, only one or two additional varieties to test with. (I have one solid cartesian with HD44780 controller, plus one half-finished delta that I can use to test carriage moves – just not effector moves.)

I understand the frustration at allowing "anything" into Marlin, and how the code has ended up a lot like spaghetti in some places as a result. I think more top-down design will come, but we still have a few more months of bottom-up code cleanup to get the code cleaned up, plus give myself a chance to study those neglected areas as it's being cleaned. I would very much like to see the different segments of Marlin encapsulated into simple singleton classes, and things like Extruders turned into formal structures / classes too. We just have to start on those sections and go for it… As the codebase becomes more modular, it should become easier to manage. At the moment, even just adding documentation comments to the code will help a lot!

thinkyhead, 2015-10-12:
I'm running two campaigns to help fund my development efforts.
Please pass on these links to all your _nouveau riche_ friends and family.

[image]
http://patreon.com/thinkyhead

[image]
https://www.gofundme.com/zmh9h9vc

closed 2016-05-04 by jbrazio