MarlinFirmware/Marlin #1908: DISABLE_Z not working

marekzehra, 2015-04-14:
Z stepper is not turned off after movement. I just pulled the latest versions of Development and Stable branch. Both have the same problem. 
M84 works fine for disabling Z steppers.
DISABLE_X and DISABLE_Y works fine. 

Configuration.h:
```
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(marekzehra)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 115200

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif

[... 746 lines cut ...]
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H
```

thinkyhead, 2015-04-14:
I think this might be addressed by this change: https://github.com/thinkyhead/Marlin/commit/3f02da646c88ec0a8a332f59cede0d1901e1e251. I will merge it and you can give it a test

Ziggy2013, 2015-04-17:
There is an issue with the steppers when they are disabled between moves.  When a servo is disabled (ie the A4988 driver is disabled) the microstep position is lost.  When the driver is re enabled the servo moves to the nearest full step position before resuming micro stepping.

Scan this thread and watch the video and you will see the effect of disabling a stepper.

http://community.robo3d.com/index.php?threads/z-axis-artifacts-known-issues-and-fixes.2576/

IMO this stepper disable feature should be removed - or at least there should be a warning about the effects.

NarimaanV, 2015-04-17:
Well we definitely shouldn't remove the ability to disable steppers, otherwise we wouldn't be able to safely move an axis by hand without risking hurting the driver board from the current steppers generate when you turn them by hand. Also there are those who experience overheated drivers that cool down when disabled even for a second at a time. I agree there should possibly be a warning or even look into modifying the code if there's a way to store the microstep position before its disabled, but definitely can't get rid of disabling steppers.

Ziggy2013, 2015-04-17:
Of course steppers need to be disabled to move by hand.  

But this configuration.h parameter is about disabling steppers between moves (eg while printing)

If you have overheated drivers then IMO disabling is a more of a "band aid" type solution.  The real reason for the overloading should be addressed.

nophead, 2015-04-17:
The drivers don't actually loose the microstep position when you disable
them. When they are re-enabled they output the same pattern as when they
were disabled. However, if your motors have a high detent torque, greater
than the static friction of your axis, they may rotate to the nearest
detent when the power is off. They do snap back to where they should be
when the driver is enabled (or you would have objects the wrong height) but
it will do the layer in the slightly the wrong place while the power is off.

I always keep them enabled but my Z axis doesn't twitch when I enable and
disable them.

Ziggy2013, 2015-04-18:
@nophead I thought the enable/not enable also held the internal microstep state and just turned the A4099 FETs on/off (as per the A4988 datasheet) .  However the behavior of the stepper as shown in the video posted in the thread above is different.  

On the Ramps 1.4/A4998 driver boards the A4988 sleep and reset pins are simply joined together - I suspect (but have not tested) that this may be the reason.

nophead, 2015-04-18:
The explanation for the video is simply that an unloaded motor will snap to the detent positions when the power is off. RAMPS disables the motor with the enable pin. Reset is tied to sleep because sleep has a pullup on the driver board, so they are both permanently high.

I agree it is daft to disable the motors between layers but it isn't due to the driver losing position. It is the detent force of the motor casing it to turn with the power off. If the axis doesn't loose steps it must return to the correct place when the driver is enabled again.

Ziggy2013, 2015-04-18:
Regardless of the reason, the Z can settle slightly in the wrong position if the stepper is disabled between moves. I saw this happening on my printer Z axis which is why I did some investigation.

 We both agree it is daft - so why should the "disable steppers between moves" option remain?

nophead, 2015-04-18:
Because it would work if you have enough friction to stop the detent torque moving the lead screw and allows people to use under rated drivers or motors that get too hot when permanently powered. Presumably if it is removed some people's machines will stop working if the upgrade Marlin.

My leadscrews don't visibly move when I enable and disable the motors but an unload motor moves to the nearest of the 50 detents, which can be up to nearly 2 full steps away. It always moves back to the right place because the driver has 64 states, which it does retain, and 64 x 50 is the number of micro positions for a 200 step motor with x 16 drivers.

I don't personally care if it is removed as I don't have any machines that need it.

Ziggy2013, 2015-04-19:
@nophead I think there is a software related issue with disabling as well - see my earlier edit. 

It seems strange to me that if a stepper is disabled you need to rely on enough friction/stiction to stop any movement. 

However if this disable "feature" is to remain there should at least be some warning in configuration.h about the consequences.

nophead, 2015-04-19:
>It seems strange to me that if a stepper is disabled you need to rely on enough friction/stiction to stop any movement. 

Well if you play with a disconnected motor you will find you can't set it to an arbitrary position. When you turn it it feels notchy and it always snaps to the nearest detent. If you count them there are 50 for a 200 step motor. The detent torque is sometimes quoted on the datasheet and varies considerably. 

Yes if the firmware loses the position when Z is disabled between layers that is probably a recent regression.

Ziggy2013, 2015-04-19:
I was aware of the stepper detent. My comment re the friction/stiction was referring to the printer application. In other words, if a steeper driver is disabled on a printer then the stepper will move to a detent if there is not enough friction/stiction to hold it precisely in place. It all depends on the printer mechanics which I find is a strange situation and IMO undesirable.

Re the firmware losing position - the particular version of firmware I referenced before is well over a year old. When I checked the dev version also has the same definition and loss of position in Marlin.h

So I still question the value (or the sense) of allowing stepper disables between moves.

nophead, 2015-04-19:
If is really has been broken for that long and nobody noticed perhaps it can simply be removed. What effect does clearing the axis_known_position flag have?

Ziggy2013, 2015-04-19:
One effect of the axis_known_position flag being set false is the firmware thinks the axis no longer has been homed and the Z=0 reference is lost.  The effect in the latest dev version is unclear to me as there have been so many major changes to the way Marlin_main.cpp, homing and auto leveling works.

I suspect the reason no one is noticing is that auto leveling is now being used more widely.  In this situation the Z Stepper is constantly moving to compensate for the bed slope so the Z stepper driver never actually disables during printing.  IMO another good reason to remove the driver disable "feature".

CONSULitAS, 2015-04-23:
@nophead Hi Chris,

thanks a lot!

I sometimes had a knocking/cracking sound on my K8200 with slightly optimized Z axis in mid of a layer print. Had no clue what it was.

Yesterday I have seen the Z axis turning some degrees while knocking, cause I watched the situation. It is very funny:

The axis turns to the next layer, the layer starts to print and sometimes after a few seconds the axis knocks and turns a tiny bit. Now I know what causes this annoying statistical Z banding.

I would suppose, that my axis has enough friction to hold the position most times, but vibration and other influences manage sometimes to shake it loose enough for turning to the next full step.

Thanks again. Will now disable DISABLE_Z forever. :-)))

thinkyhead, 2015-04-25:
Do we still have a feature request, then? To put a stronger warning in the documentation and comments?

Ziggy2013, 2015-04-25:
Yes.  If the "feature" must remain, then a stronger warning is definitely warranted IMO.

thinkyhead, 2015-04-25:
#1968

closed 2015-04-25 by thinkyhead