MarlinFirmware/Marlin #1909: How to add Chinese language

diver197, 2015-04-14:
How to add Chinese language，I want to know how to do the dogm_font_data file, it is best to have the detailed steps, thank you!

AnHardt, 2015-04-14:
@diver197 
Chinese - that's difficult.
Currently we are limited to 256 symbols per font. That's, by far, not enough to represent the complete Chinese charset.  Maybe you can make a selection of symbols you need. The language_xx.h file will then look like 'language_kana.h'.
We use a menu system with 5 lines, on a display with 64 pixel height. That means we have 12 pixel for a line. To have some space in between the lines we can't use more then 10 pixel height for the symbols.
We use fixed width fonts. To fit 22 Symbols on the 128 pixel wide screen, the symbols can't be wider than 5 pixel. Maybe you can work with half symbols - two places in the charset will than build one wide symbol.
The latest fonts i have made with 'Fony.exe' - but 'fontforge' will probably work to. 
So the easyest way to go is:
* Get 'Fony.exe'
* Copy one of the existing *.fon files and work with this.
* Change the pixels. Don't change width or height. 
* Export as *.bdf
* Use 'bdf2u8g.exe' to produce the *.h file. Examples for the existing fonts are in 'make_fonts.bat'
* Edit the produced *.h file to match our needs. See hints in 'README.fonts' or the other 'dogm_font_data_*.h' files.

Than produce a matching language_xx.h file.

Good luck.

diver197, 2015-04-19:
Using the above methods can in earlier versions, now the latest I have no way to display text after use, according to blank

AnHardt, 2015-04-19:
@diver197 
I hope i understand your problem. You cold be a bit more descriptive about what you did.

My explanation was incomplete.
In Your language_xx.h file:

     #define MAPPER_NON
     #define DISPLAY_CHARSET_NEWNAME

In 'dogm_lcd_implementation.h' before line 81 (  #else // fall-back) inset a new block to load your font.

    #elif defined( DISPLAY_CHARSET_NEWNAME )
      #include "dogm_font_data_yourfont.h"
      #define FONT_MENU_NAME YOURFONTNAME

diver197, 2015-04-20:
I use is 12864 lce, add the relevant file, switch to the Chinese, no any display

AnHardt, 2015-04-20:
Would you please be so kind to show me your 'relevant' fileS?

Configuration.h
dogm_lcd_implementation.h
Your font dogm_font_data_yourfont.h
Your language file language_xx.h

Please use a pastebin or even better - give me a link to your fork of marlin.

diver197, 2015-04-21:
http://pastebin.com/PCkDq4sN
http://pastebin.com/rnqQ1XMC
http://pastebin.com/KH2dfTW9
http://pastebin.com/WtwsdAb7

AnHardt, 2015-04-21:
@diver197 
Got it. I have a more or less working example for you. Please see: https://github.com/AnHardt/Marlin/tree/Chinese

I totally forgot about the fallback test in 'language_eh.h'. I guess that was the problem.

If you do not see a picture at all - The second half of your 'Configuration.h' is lost. No definitions of languages or displays at all.
'language_en.h' is not the ideal base for a new translation. All the #ifndef's are not needed.
Your font is a bit to large - especially the European part.
[image]

[image]

All the differences to https://github.com/MarlinFirmware/Marlin/tree/Development
you can see at: https://github.com/AnHardt/Marlin/pull/3

diver197, 2015-04-22:
Thank you, big font issues have been identified, is now being revised

diver197, 2015-04-23:
How to adjust the menu at the top of the space, there are 3 pixels, I want to show to the top
[image]

[image]

AnHardt, 2015-04-23:
@diver197 
Drawing the white bar and positioning the cursor for the text is done in file 'dogm_lcd_implementation.h' in function lcd_implementation_mark_as_selected''. Line 369 draws the box. Line 375 positions the cursor. 
DOG_CHAR_WIDTH and DOG_CHAR_HEIGHT are defined around line 95.
START_ROW is 0 and will be removed some day.

From line 343 on the position for the status-line on the status screen is set.

Maybe you can also play with the baseline of your font.
LCD_HEIGHT defined in 'Conditionals.h' sets the number of lines in the menu!

AnHardt, 2015-04-23:
@diver197 
You may also want to play with:
USE_SMALL_INFOFONT line 46. 
Applying FONT_STATUSMENU to line 335.
Shifting SD-card symbol (from line 253 ) one pixel up.
Shifting progress bar (from line 261) a bit up.

AnHardt, 2015-04-23:
And when playing with USE_SMALL_INFOFONT obviously with FONT_MENU in line 344 to get the Chinese font for the status line.

diver197, 2015-04-23:
Thank you very much! I am doing now is 1.2 m * 1.2 m * 1.5 m machines

AnHardt, 2015-04-23:
@diver197 
Added a quick and dirty hack to make some vertical room for your font.
See https://github.com/AnHardt/Marlin/pull/3 - second commit.
The horizontal problems are still unsolved.
As I said all is depending on a certain font size.

[image]

[image]

AnHardt, 2015-04-23:
@diver197 
If you promise to send us back your finished 'language_cn.h' file and your font sources (*.fon or *.bdf)
i'll have a look on the remaining horizontal problems.

diver197, 2015-04-23:
2015-04-23 

152053

diver197, 2015-04-23:
Have you received the attachment, if there is no email 152053 @163.com

AnHardt, 2015-04-23:
@diver197 
No. I have not. Try github(at)kitelab(dot)de

diver197, 2015-04-24:
I don't know how to upload it

AnHardt, 2015-04-24:
@diver197 
So send a mail with attachment to github@kitelab.de

AnHardt, 2015-04-27:
@diver197 
Please Test if #1977 works for you.

AnHardt, 2015-04-28:
#1977 is merged.

closed 2015-04-28 by AnHardt